emsesp/EMS-ESP #422: Missing values for Junkers ISM1

FredericMa, 2020-07-06:
I have the ISM1 solar module from Junkers.
In the setup I have/had from Norbert, I had 2 extra parameters that indicate if the warm water tank is fully heated and if the collectors are switched off (because the tank is fully heated).
Can these 2 extra parameters be added? I can imagine that these 2 parameters are not generic enough and cannot be applied to other solar systems?

Looking at the code of Norbert, it looks like it is about the [msgID_868_Solar](https://github.com/norberts1/hometop_HT3/blob/26234f81573566c75f84842212109b0b176e6abc/HT3/sw/lib/ht_discode.py#L2312) function.

proddy, 2020-07-06:
I made the change in v2 (the web version) as enhancing the devices is a lot easier now in v2. The only problem is you'll need to upgrade and there are still issues with the Web UI on the ESP8266 because of lack of memory. But it works on an ESP32!

FredericMa, 2020-07-06:
There is an ESP32 on its way but I have no spare at the moment. I expect it to arrive within the next 2 weeks.

I saw in commit 58b93fb62dd6c267c00b0b82575e34956120df2d that you read the 2 values from the SM100Status message, which totally makes sense since I referenced to that but it also got me thinking since I have an ISM1.
And it looks like the values are also present in the message with ID 259 (0x0103): [link to Norberts code](https://github.com/norberts1/hometop_HT3/blob/26234f81573566c75f84842212109b0b176e6abc/HT3/sw/lib/ht_discode.py#L2628)

So at least 2 systems will be able to use these values ^^

proddy, 2020-07-06:
good catch, I'll add it there too. The ESP8266 code seems to be working so if you want to try it out give it a shot. But once you move over, there's no way going back to v1.9.x

FredericMa, 2020-07-06:
No problem! I'm already on v2 since we covered the issues/enhancements of my junkers setup.

FredericMa, 2020-07-08:
I can confirm that this works as expected.
Thanks!

closed 2020-07-08 by FredericMa