emsesp/EMS-ESP #425: (v2) can not set mqtt home base

SpaceTeddy, 2020-07-17:
Hi Paul, 
did you remove the mqtt home base setting in telnet console?

chris

proddy, 2020-07-17:
Yes, this was after a short discussion in https://github.com/proddy/EMS-ESP/issues/421#issuecomment-654316186 . I looked at other systems and figured it wasn't really useful and confused some people. The root of all topics will be the hostname (ems-esp as default) like Home Assistant, Tasmota, Espurna, Homie etc..

If you really need it back I can re-add it.

proddy, 2020-07-17:
I'll also add a breaking changes section to the README

closed 2020-07-17 by proddy

SpaceTeddy, 2020-07-17:
no, not really needed. I can change it in my HA system. 
BTW: I started to evaluate the latest version from yesterday, and my RC20RF remote is not working with this v2_web 2.0.0b4 version. Did you made any changes here?

system:
Sieger BK15
es73 Thermostat
RC20RF/RFM10