emsesp/EMS-ESP #426: (v2) Multi-language support (starting with German/DE)

proddy, 2020-07-17:
Support multi-locale for Telnet console commands, MQTT, SysLog messages and the Web. Since there is not enough memory to hold all the translations it will have to be different builds (DE, NL, UK). String literals will be stored in translation files and built at compile time.

@MichaelDvP @bbqkees what you vote for this feature?

bbqkees, 2020-07-17:
Most Gateways go to German speaking countries so it makes perfect sense to provide the interface in their own language. I get some questions about it regularly.
If there is not enough memory to store everything, we should at least provide a language selector at the first install which will then download and install the new firmware. 
So we could just have the UK build as default with a selector to install another language.

MichaelDvP, 2020-07-17:
Yes, that would be nice. What about a changeable language-file in the spiffs? There is a lot of space left and you have only one software.

proddy, 2020-07-17:
That's a good idea. Store the localizations in SPIFFS and download to PROGMEM when EMS-ESP boots.

proddy, 2020-07-25:
In order of importance
 - Web UI
 - MQTT topics and payload
 - Console commands (serial and telnet)
 - Logging information (to console & syslog)

For the web I had a look at react-intl, i18next-react, polyglot and this one called lingui-js but they'll all quite big libraries and there is just not enough progmem to store all the translations to have it real-time on the ESP8266. react-intl is the right one here (14k zipped).

proddy, 2020-09-23:
First localized version of the UI is looking promising.

proddy, 2021-01-17:
with the extra 160K on the ESP32 I can easily store the strings in SPIFFS and load them dynamically