emsesp/EMS-ESP #429: Missing telegram types for boiler Junkers CerapurSolar-comfort 24/75

joanwa, 2020-07-20:
I have a Junkers CerapurSolar boiler recognized as:
```
Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.06) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E
 Subscribed MQTT topics: thermostat_cmd cmd_temp cmd_mode cmd_temp cmd_mode cmd_temp cmd_mode cmd_temp cmd_mode
```
with two more missing telegram types:
```
000+15:39:19.677 D 42: [emsesp] No telegram type handler found for ID 0xBF (src 0x10, dest 0x00)
000+00:08:32.915 D 275: [emsesp] No telegram type handler found for ID 0x267 (src 0x10, dest 0x00)
```
not really sure what it is, but what be happy to assist in encoding!

proddy, 2020-07-20:
Both these are from the thermostat (device ID 0x10). 

I'm not sure what 0xBF is. I can't find any details on it and I usually use Norbert's guide for reference (https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf) when it comes to Heatronics.

0x267 is an odd one. How many heating circuits are you using?

joanwa, 2020-07-20:
just one heating circuit, but also using a mixing and a solar module MM100 & MS200. The Bosch Gateway is showing this:

[image]
[image]

norberts1, 2020-07-28:
@joanwa, @proddy : some information on that telegrams above, not yet in my documents but it will be as soon as possible.

[image]
[image]
[image]

proddy, 2020-07-31:
thanks @norberts1 for describing those extra telegrams.

@joanwa is this information important to add to EMS-ESP? It's error status codes and floor drying status. I can add, but not sure if it's useful to report out in the WebUI or via MQTT.

Actually those screenshots you sent from the Bosch gateway look interesting. Is it just a mobile app? I'd love to be able to replicate that and show the same information. Do you think you could help collect some information on this

joanwa, 2020-07-31:
@proddy those telegrams not actually needed. I'm still after the missing upper boiler and solar heat exchanger temperatures. Just posted all not recognized telegrams I've come across for the sake of completeness in pursuit of the missing temperatures.

Yes, thats the Bosch Easy Remote app. This app is just horrible. But if you want to replicate to show the same information, happy to help! They also do have a web login & interface from bosch, but I couldn't figure out how to get this to work. And believe it or not, local connection from the App to that gateway is only a feature for a year or so. Before everything had to go via Bosch servers.

proddy, 2020-07-31:
great. If there is information which was useful to see in the Bosch app but missing now in EMS-ESP then lets try and add it.

proddy, 2020-09-05:
Closing this. We determined the missing type_id's and not really worth picking up

closed 2020-09-05 by proddy

joanwa, 2020-09-07:
> great. If there is information which was useful to see in the Bosch app but missing now in EMS-ESP then lets try and add it.

I'm right now trying to figure out where the historical data for temperatures and energies are available. It's shown in both, app and controller. If I need help, will open a new issue.

[image]
[image]
[image]
[image]