emsesp/EMS-ESP #431: (v2) automatic migration from 1.9.x to 2.0

proddy, 2020-07-23:
Detect the 1.9.x json setting files in SPIFFS, convert over to 2.0 in LittleFS.

proddy, 2020-07-31:
This turns out to be harder than expected. EMS-ESP 1.9 uses the SPIFFS filesystem. There is no easy way to detect which type of filesystem is present on an ESP. My original idea was to mount SPIFFFs, detect the old config files, load them and then write them back to LittleFS which is the new filesystem ib v2. Problem here is that as soon as you mount a FS and it fails, it will format the complete filesystem. So basically each time you start EMS-ESP v2 all the settings will be erased. Sucks.

Then I thought about building a web UI to import the backup file from 1.9. In this scenario, the user would have gone into the AP and already at the settings screen so might as well just complete the rest of the key settings like wifi SSID, password and MQTT IP with a few clicks.

Conclusion, for the few users on 1.9 its not worth it. I'll just document the steps in the wiki.

proddy, 2020-08-28:
Ok, I found a way to make this work. Will push in a change

proddy, 2020-08-28:
@bbqkees can you help test? When uploading the firmware over an older v.1.9 firmware it should automatically port over the settings.

bbqkees, 2020-09-04:
I ran a number of tests yesterday.

If you select the V2 8266 bin file from the releases page and upload it via the 1.9.5 web interface the update to V2 works fine.
The update itself works on both the Lolin Wemos D1 Mini V3.1 and the clone Wemos D1 Mini V2 version.
(I used those clones on the very first Gateways until about a year ago).
I tried it powered via USB and on a 1.6 Gateway bus powered.
(So its safe to update a Gateway. Just make sure you haven't downloaded the ESP32 bin file instead of the ESP8266 file.)

But indeed with an update via the web, settings are not migrated from 1.9.5 to V2.
So it starts clean.

I don't think this is a major problem, its just basically setting WiFi and MQTT and of you go.
But of course it would be better if the settings are actually kept.

proddy, 2020-09-05:
made some changes, tested it too and it seems to work (at least for me). Also via the 1.9 web 'upload' page.

proddy, 2020-09-12:
There have been reports of the settings not being persisted after a reboot.

proddy, 2020-09-12:
> There have been reports of the settings not being persisted after a reboot.

ran a few tests again, erased flash, uploaded 1.9.5, set settings, tested it, upgraded to 2.0.1. Not issues.

commit 6b4e91aa (2020-08-28): auto update settings 1.9->2.0. Closes #431
closed 2020-09-12 by proddy