emsesp/EMS-ESP #432: (V2) incorrect hostname in console when using static IP adress

lmdecroos, 2020-07-23:
When using a static IP address, then the hostname is correct in the webgui and syslog. 
But in the console the hostname is ESP-xxxxxx wheroff xxxxxx are the last 6 digits of the MAC address
MAC address: DC:4F:22:5E:96:9C
Hostname: ESP-5E969C

to reproduce:
go to wifi settings in the webgui
select static IP and provide the network settings
change the hostname to a name not starting with ems
restart the gateway
open telnet session
go to system
su
show

result the displayed hostname will be ESP-xxxxxx wheroff xxxxxx are the last 6 digits of the MAC address

tested on B5, B6 and B7

proddy, 2020-07-23:
Thanks for reporting. I couldn't reproduce with B6 but will try again.

proddy, 2020-07-25:
Ok, I reproduced this. The wifi hostname is not set when using a static IP so in `system` & `show` it will show the default one the ESP core gives you which is based on the MAC address. The console prompt is correct.

closed 2020-07-25 by proddy