emsesp/EMS-ESP #434: Support writing to Buderus RC200 thermostat

fupe, 2020-07-24:
Hi,
is it possible add write temperature to Buderus rc200 thermostat?
my configuration
EMS-ESP version 1.9.6b0

These 3 were recognized by EMS-ESP:
 Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si/27i (DeviceID: 0x08, ProductID: 123, Version: 06.08)
 Controller: BC25 Base Controller (DeviceID: 0x09, ProductID: 125, Version: 03.03)
 Thermostat: Buderus RC200/Bosch CW100/Junkers CW100 (DeviceID: 0x18, ProductID: 157, Version: 41.08)

and part of log

```
info

EMS-ESP system status:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled
  Saved known device IDs: none

EMS Bus status:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=559, # CRC errors=0
  Tx: Last poll=4.660 seconds ago, # successful write requests=0

Boiler data:
  Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si/27i (DeviceID: 0x08, ProductID: 123, Version: 06.08)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water circulation pump type: 3-way pump
  Warm Water circulation pump freq: 2x3min
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desinfection temperature: 70 C
  Warm Water circulation active: off
  Warm Water set temperature: 10 C
  Warm Water current temperature: 58.9 C
  Warm water temperature (intern): ? C
  Warm water temperature (extern): 58.9 C
  Warm Water current temperature (extern): 58.9 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 1203 times
  Warm Water active time: 15 days 10 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 30.5 C
  Max boiler temperature: ? C
  Return temperature: ? C
  Gas: off
  Warm water mode: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 68 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Outside temperature: --3276.-8 C
  Exhaust temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 4026 times
[... 113 lines cut ...]
(01:28:36.500) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 05 01 31 00 00 00 00 00 40 80 00 02 4D 80 00 00 00 FF 30 48 00 CB 00 00 00 (#data=25)
(01:28:36.875) Boiler -> All, UBAMonitorSlow(0x19), telegram: 08 00 19 00 80 00 80 00 80 00 00 00 00 00 00 0F BA 01 73 F4 00 00 00 01 1D 3C 00 0B 07 80 00 (#data=27)
(01:28:37.114) Boiler -> All, UBAMaintenanceStatusMessage(0x1C), telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (#data=25)
(01:28:37.356) Boiler -> All, MC10Status(0x2A), telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 67 00 00 80 00 00 80 00 80 00 80 00 04 (#data=21)
Publishing MQTT data for: boiler
(01:28:40.312) Thermostat -> All, Type(0xA2), telegram: 18 00 A2 00 00 00 00 00 00 (#data=5)
(01:28:45.841) Boiler -> All, MC10Status(0x2A), telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 67 00 00 80 00 00 80 00 80 00 80 00 04 (#data=21)
(01:28:46.063) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 08 00 34 00 0A 02 4D 02 4D 20 00 00 03 00 00 56 B8 00 04 B3 00 80 00 (#data=19)
(01:28:46.335) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 05 01 31 00 00 00 00 00 40 80 00 02 4D 80 00 00 00 FF 30 48 00 CB 00 00 00 (#data=25)
Publishing MQTT data for: boiler
(01:28:53.044) Thermostat -> All, RCPLUSStatusMessage_HC1(0x01A5), telegram: 18 00 FF 08 01 A5 01 ED 01 01 03 01 ED 00 89 (#data=9)
(01:28:54.310) Thermostat -> Boiler, UBAFlags(0x35), telegram: 18 08 35 00 01 01 (#data=2)
(01:28:55.854) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 05 01 31 00 00 00 00 00 40 80 00 02 4D 80 00 00 00 FF 30 48 00 CB 00 00 00 (#data=25)
(01:28:56.159) Boiler -> All, MC10Status(0x2A), telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 67 00 00 80 00 00 80 00 80 00 80 00 04 (#data=21)
(01:28:56.380) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 08 00 34 00 0A 02 4D 02 4D 20 00 00 03 00 00 56 B8 00 04 B3 00 80 00 (#data=19)
Publishing MQTT data for: thermostat boiler
lo(01:29:05.841) Boiler -> All, UBAMonitorWWMessage(0x34), telegram: 08 00 34 00 0A 02 4D 02 4D 20 00 00 03 00 00 56 B8 00 04 B3 00 80 00 (#data=19)
g(01:29:06.157) Boiler -> All, UBAMonitorFast(0x18), telegram: 08 00 18 00 05 01 31 00 00 00 00 00 40 80 00 02 4D 80 00 00 00 FF 30 48 00 CB 00 00 00 (#data=25)
(01:29:06.377) Boiler -> All, MC10Status(0x2A), telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 67 00 00 80 00 00 80 00 80 00 80 00 04 (#data=21)
 n

System Logging set to None
```

thanks Martin

proddy, 2020-07-25:
Hi @fupe , although I have tested it the latest version 2.0 should support reading and writing to your RC200 as it uses the same telegram set as the RC100 and RC300. Version 2.0 is still in development, but pretty stable. You can either download, build and upload yourself or grab the pre-built binary. Note however that it's not yet backward compatible with 1.9.x yet so you will need to re-enter all your settings (wifi, mqtt, etc) manually via the web or the new telnet console. You'll find all the info at https://github.com/proddy/EMS-ESP/blob/v2/README.md

fupe, 2020-07-31:
Hi Proddy,
thanks for reply.
I download and setup v2.0.0b10
NOw i can see all three devices in web and putty terminal
in tx_mode 1 (I try mode 1 2 3)
Boiler Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i 0x08 123 06.08
Controller BC25 0x09 125 03.03
Thermostat RC200/CW100 0x18 157 41.08

ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08
ProductID:123, Version:06.08)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34
0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9
 Subscribed MQTT topics: ems-esp/boiler_cmd ems-esp/boiler_cmd_wwactivated
ems-esp/boiler_cmd_wwonetime ems-esp/boiler_cmd_wwcirculation
ems-esp/boiler_cmd_wwtemp

Thermostat: RC200/CW100 (DeviceID:0x18, ProductID:157, Version:41.08)
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9
0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E

Controller: BC25 (DeviceID:0x09, ProductID:125, Version:03.03)

But, I can switch to set thermostat.
ems-esp:/$ thermostat
Command not found

  boiler is ok

ems-esp:/$ boiler
ems-esp:/boiler$

on mqtt explorer i can see boiler_data but not thermostat data.
[image: image.png]

here is part of log
000+00:21:49.699 N 398: [emsesp] Boiler(0x08) -> All(0x00),
UBAMonitorFast(0x18), data: 05 02 27 00 00 00 00 00 40 80 00 02 14 80 00 00
00 FF 30 48 00 CB 00 00 00
000+00:21:49.699 D 399: [emsdevice] Received UBAMonitorFast
000+00:21:49.994 N 400: [emsesp] Boiler(0x08) -> All(0x00),
MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80
00 80 00 04
000+00:21:49.994 D 401: [emsdevice] Received MC10Status
000+00:21:50.239 N 402: [emsesp] Boiler(0x08) -> All(0x00),
UBAMonitorWW(0x34), data: 0A 02 14 02 14 00 00 00 03 00 00 57 6D 00 04 C0
00 80 00
000+00:21:50.239 D 403: [emsdevice] Received UBAMonitorWW
000+00:21:50.343 D 404: [mqtt] Publishing topic ems-esp/boiler_data (#166,
attempt #1, pid 1)
000+00:21:59.673 N 405: [emsesp] Boiler(0x08) -> All(0x00),
UBAMonitorWW(0x34), data: 0A 02 14 02 14 00 00 00 03 00 00 57 6D 00 04 C0
00 80 00
000+00:21:59.673 D 406: [emsdevice] Received UBAMonitorWW
000+00:21:59.955 N 407: [emsesp] Boiler(0x08) -> All(0x00),
UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00
000+00:22:00.024 D 408: [emsdevice] Fetching values for device ID 0x08
000+00:22:00.024 D 409: [telegram] Tx read request to device 0x08 for type
ID 0x19
000+00:22:00.024 D 410: [telegram] Tx read request to device 0x08 for type
ID 0x33
000+00:22:00.024 D 411: [telegram] Tx read request to device 0x08 for type
ID 0x16
000+00:22:00.024 D 412: [emsdevice] Fetching values for device ID 0x09
000+00:22:00.024 D 413: [emsdevice] Fetching values for device ID 0x18
000+00:22:00.024 D 414: [telegram] Tx read request to device 0x18 for type
ID 0x06
[... 126 lines cut ...]
00 80 00
000+00:22:39.897 D 484: [emsdevice] Received UBAMonitorWW
000+00:22:40.129 N 485: [emsesp] Boiler(0x08) -> All(0x00),
UBAMonitorFast(0x18), data: 05 02 26 00 00 00 00 00 40 80 00 02 14 80 00 00
00 FF 30 48 00 CB 00 00 00
000+00:22:40.129 D 486: [emsdevice] Received UBAMonitorFast
000+00:22:40.267 D 487: [mqtt] Publishing topic ems-esp/boiler_data (#172,
attempt #1, pid 1)
000+00:22:49.566 N 488: [emsesp] Boiler(0x08) -> All(0x00),
UBAMonitorFast(0x18), data: 05 02 26 00 00 00 00 00 40 80 00 02 14 80 00 00
00 FF 30 48 00 CB 00 00 00
000+00:22:49.566 D 489: [emsdevice] Received UBAMonitorFast
000+00:22:49.807 N 490: [emsesp] Boiler(0x08) -> All(0x00),
MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80
00 80 00 04
000+00:22:49.807 D 491: [emsdevice] Received MC10Status
000+00:22:50.016 N 492: [emsesp] Boiler(0x08) -> All(0x00),
UBAMonitorWW(0x34), data: 0A 02 14 02 14 00 00 00 03 00 00 57 6D 00 04 C0
00 80 00
000+00:22:50.016 D 493: [emsdevice] Received UBAMonitorWW
000+00:22:50.328 D 494: [mqtt] Publishing topic ems-esp/boiler_data (#173,
attempt #1, pid 1)

thanks for help.
Martin

proddy, 2020-07-31:
great!

MichaelDvP, 2020-08-02:
@proddy seems you closed to fast and overlooked that there are 2 issues left.
1. the thermostat is the only thermostat, but uses device_id 0x18. It is not recognized as master. 
2. he can't get into thermostat context because there is no master-thermostat, but to set the master he has to enter the context.
 
(now we have the case mentioned [here](https://github.com/proddy/EMS-ESP/issues/362#issuecomment-629652880))

Quick solution is allow all thermostats as master, not only 0x10 and 0x17. This will make the system of @fupe useable.
But if someone uses more RC200 for different hc we need to have hc2 type-id matched to 0x19, hc3 typ_id to 0x1A, etc. and only the fist one of these controllers is used for DHW-settings.

proddy, 2020-08-02:
oh dear, I overlooked that. I'll have a think about the options.

MichaelDvP, 2020-08-03:
A possible patch for the problem with multible hc and RC200 at 0x18.. is:
- Modify `emsdevices.h` to
```
    inline uint8_t device_id(uint8_t hc = 0) const {
        if (((device_id_ & 0x7F) >= 0x18) && ((device_id_ & 0x7F) <= 0x1B)) {
            return ((device_id_ & 0x80) + 0x18 + hc);
        }
        return device_id_;
    }
``` 
and add optional hc to `write_command` (also in `emsdevices.cpp` with call of `device_id(hc)`)
```
    void write_command(const uint16_t type_id, const uint8_t offset, uint8_t * message_data, const uint8_t message_length, const uint16_t validate_typeid, const uint8_t hc = 0);
    void write_command(const uint16_t type_id, const uint8_t offset, const uint8_t value, const uint16_t validate_typeid, const uint8_t hc = 0);
    void write_command(const uint16_t type_id, const uint8_t offset, const uint8_t value, const uint8_t hc = 0);
```
and for RC100-300 in thermostat add the hc. 

Then the first of the 0x18.. thermostats is registered as master with 4 hc, the others are not writeable, but writing to a hc will automatically address the right controller.

proddy, 2020-08-03:
that would work. I'll add it to the new branch.

fupe, 2020-08-04:
Hi proddy,
I am now 1000km from home. I will return on sunday and test your path and
report. Thanks
Martin.

closed 2020-08-11 by proddy