emsesp/EMS-ESP #435: (v2) can`t logon after upgrade from B6 to B7

sq4bja, 2020-07-25:
Hi,
Since I upgrade from B6 to B7 I can`t logon from WWW admin login. All my passwords are incorrect. Telnet password works OK.
Tom

proddy, 2020-07-25:
did you change the passwords?

sq4bja, 2020-07-25:
Yes. I set new password on previous version, after upgrade from alfa to beta.
Tom

proddy, 2020-07-25:
ok, I'll try and reproduce. To reset the system use the `format` command

proddy, 2020-07-25:
I added a console command called `show users` under system. Make sure you're admin (with `su`) and check that the usernames and passwords are correct.

sq4bja, 2020-07-25:
Ok, but current release is still b7 from 11 hours ago. If i try to upgrade from espota, I must put ems-esp-neo password. Current SU password doesn`t work. So i try to use admin login ans ems-esp-neo to web access - still the same "invalid credential".

proddy, 2020-07-25:
the build is broken, working on it.

sq4bja, 2020-07-26:
After format command still i can`t login via web page.
And another bug - after format command i try to put wifi settings again. After I send "set wifi ssid my_setwork" ems-esp do the reset. After reconnect it works as AP 30 seconds and it try to reset again. It is too difficult to set correct password for wifi network while it reset again...
And now i can`t find mqtt settings. Only "system show mqtt".

proddy, 2020-07-26:
- good spot on the 'set wifi' commands. It shouldn't immediately reset the WiFi (btw it only reconnects, not reboots the ESP but I can imagine the feel similar). I'm working on a fix for this.
- you can only change MQTT now via the Web. It was too much code to put in both Web and Console for the ESP8266.
- I'm confused why you can't log in. Try a different browser or incognito mode. Perhaps the web token is cached?

proddy, 2020-07-26:
I tested on ESP8266 and also seeing a lot of issues with the web (not loading, crashing). Seems to only happen on the ESP8266 (not the ESP32) so I suspect its memory related. I'm looking into it.

sq4bja, 2020-07-26:
Ok. I`m testing B8 - On Dashboard -> EMS DEVICES i`m getting:
`Problem fetching: JSON.parse: unexpected character at line 1 column 1 of the JSON data.`
Tom

proddy, 2020-07-26:
does this happen when you click on the table? which device did you select?

sq4bja, 2020-07-26:
[image]

proddy, 2020-07-26:
Thanks. Screenshots tell so much more. When you go into telnet and type 'show' or 'show devices' what does it come back with?

Also is your login problem fixed now?

sq4bja, 2020-07-26:
show devices:

```
ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.08)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9
 Subscribed MQTT topics: ems-esp/boiler_cmd ems-esp/boiler_cmd_wwactivated ems-esp/boiler_cmd_wwonetime ems-esp/boiler_cmd_wwcirculation ems-esp/boiler_cmd_wwtemp

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E
 Subscribed MQTT topics: ems-esp/thermostat_cmd ems-esp/thermostat_cmd_temp ems-esp/thermostat_cmd_mode

Controller: BC25 (DeviceID:0x09, ProductID:125, Version:03.03)
```
After FORMAT command everything works OK. Now, I have admin admin login.
Tom

proddy, 2020-07-27:
do you still get that error on the web in the EMS Devices page? If so you'll need to help me debug. Easiest way is to use the browsers' debugger. Like in Google Chrome, open the Developer Tools (either F12 or developer tools from the menu), go to Network, go to EMS Devices page, wait for it load and then check if you have an element called `allDevices`. Inspect the content for any malformed JSON.

[image]

proddy, 2020-07-28:
@sq4bja did you find out what was breaking?

sq4bja, 2020-07-28:
Ok, sorry for delay, I was out of home and i can`t access from WAN to EMS-ESP.
Below is the debug network inspector of above page:
[image]

sq4bja, 2020-07-28:
Sorry, but now I can`t reproduce this error. Everything is working correct. But sometimes I can`t get webpage at all. Nothing helps - F5, CTRL+F5, restart command doing by telnet session.

proddy, 2020-07-28:
okay, thanks for reporting back. If you use telnet and web at the same time it will become unresponsive because of all the memory it uses. I'm looking into that. I'll close this issue for now (related to login from b6 to b7). Feel free to reopen or create any new issues for weird things you find. thanks!

closed 2020-07-28 by proddy