emsesp/EMS-ESP #437: Condens 9000i with CW400

chaosjug, 2020-07-26:
Helllo,

I have a new Condens 9000i with CW400.

EMS-ESP is detecting the following:
```
These 4 were recognized by EMS-ESP:
 Boiler: Buderus Logamax plus/GB192/Bosch Condens GC9000 (DeviceID: 0x08, ProductID: 208, Version: 01.04)
 unknown?: unknown? (DeviceID: 0x09, ProductID: 224, Version: 35.05)
 Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 18.06)
 unknown?: unknown? (DeviceID: 0x16, ProductID: 220, Version: 01.05)
```
Not sure what the unknown devices are.

I have a few not recognized values in Boiler data:
Max boiler temperature: ? C
Exhaust temperature: ? C

The other values shown by info seem to be correct.

The CW 400 is showing more information that would be great to have, e.g. power and gas consumption.

Any hints how to find the data in the telegrams?

proddy, 2020-07-26:
Hi @chaosjug 

Thanks for reporting your findings. The deviceID 0x09 is the controller built into the boiler. It's already added in version 2.0. The deviceID 0x16 we've seen before (see https://github.com/proddy/EMS-ESP/issues/397#issuecomment-643592594) with the 9000i and still not sure what it is. In any case it's safe to ignore for now.

Version 2 also cleans up the reporting a little so you'll only see and receive the information that you have streaming in from the EMS bus. However this newer firmware version is still in development and recently had some nasty issues with the web on an ESP8266 so you may want to wait a few days before trying it out if you're using an ESP8266

Adding more info from the CW400 should be possible. Like everything Junkers/Heatronics related I refer to Norbert's bible on the telegrams at https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf. If you find it there we can implement it.

chaosjug, 2020-08-02:
Hi @proddy 

I'm using a Gateway from BBQKees which is using the ESP8266 as far as I know, so I'll wait before I update to 2.0.

If I understand the telegrams correctly the two missing temperatures are actually not available in my system.

Norbert's bible was renamed today and can be found here: https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf. Unfortunately there is nothing in there about power or gas consumption. Maybe this is estimated by the CW 400 based on the burner data. The CW 400 does seem to store some data. E.g. you can view a chart of the outdoor temperature of the past few days as well.

Is there anything I can help with the deviceID 0x16? I haven't seen any telegrams from or to it so far.

proddy, 2020-08-04:
The 0x16 is some type of unit built into the boiler, but doesn't seem to do anything so I would just ignore it. You said the CW400 is showing you gas and power consumption? Can you share screenshots on what is being displayed? The data is coming from the boiler. We could probably trace which calls the CW400 is making to the Boiler to get this data using the `watch` command. In any case you'll need to move to v2 soon.

chaosjug, 2020-08-04:
Here is a screenshot of the consumption and also of the temperature graph. For both I didn't find anything in Norbert's bible [image]
[image]

norberts1, 2020-08-05:
@chaosjug that outdoor-temperatur bytes you can find in messageID:25 (88 00 19 00 <outdoor-temp>...) at bytes 4/5. see attached picture. That history shown above is only created by the CW400 itself.
That information and related message 'Energieverbrauch' isn't yet documented. You are the lucky man to have that newest boiler 9000i to deliver this information/messages.
@chaosjug @proddy
I'll think it would be the best to get out some raw-data to analyze. I'll decode this then as good as I can.
Currently its very hot outside, so is the best to get and log data if there is heating required. Hopefully that messages are send more then once per hour.

[image]

chaosjug, 2020-08-16:
@norberts1 how would I best capture raw data? I've played around with telnet and logging, but that didn't seem very efficient when searching for new messages. Mostly I've seen messages which are already in your document. I think only one rather short one which wasn't.

norberts1, 2020-08-17:
@chaosjug it's a bit difficult to get out that data and to log it, because at least more then one hour log-data is required.
So it seems a bit more convenient if we can log that heater poweron startup-sequence. At this early time the **optional** displayed information (Systeminformationen, Energieverbrauch) are requested by the CW400 from the boiler-electronic.
My CW400 does this request for that 'Systeminformationen' and get that related data 'Brennerstarts', 'Brennerlaufzeit' etc. but that messages for that display 'Energieverbrauch' isn't findable, cause my boiler-electronic can't deliver that data and that menu-item isn't available.
So, what's required:
1. running that EMS-ESP modul with usb-power and not powered by the heater-bus.
2. preparation of that EMS-ESP modul for raw-data logging with filter on device-ID 10.
3. Switching of that heater-system to get that CW400 - modul unpowered (off-state for at least 5-10 minutes regarding on that gold-cab or capacitor in the CW400).
4. switch on your heater-system again and hopefully that EMS-ESP will log that data as early as possible.
5. checking that menu-item of 'Energieverbrauch' for availability (perhaps an additional request is generated).
Log-data for at least 3 - 5 minutes should be enough.
This procedure above I didn't yet checked on my system but it should be possible.

@proddy is it possible to run the EMS-ESP usb-powered and prepared for raw data logging during that heater-system powered of ?
Can we get that early data-communication with that type of logging?

proddy, 2020-08-17:
@norberts1 yes EMS-ESP can run powered via the USB. To capture the logs the easiest way would be to put a watch on the DeviceID (using `watch raw 10`) and enable syslog (on a RPi or Linux box) and restart the CW400. BTW you can also change the BUS ID from it's defauly 0x0B if that causes conflicts.

chaosjug, 2020-08-18:
@proddy not sure if I can follow you. The `watch raw 10` is a telnet command right? But how is this related to syslog?

proddy, 2020-08-30:
@chaosjug sorry for the late response. forgot about this one. So the syslog will just capture any logging and send it to the syslog server. When you use the `watch` command it will write as NOTICE messages so if you have SysLog configured to watch ERR, INFO or DEBUG they will be picked up.

[image]

chaosjug, 2020-08-30:
It took me a while to get this set up, so I'll document the steps here:
Enable logging and set IP in general settings.
On the raspberrypi I put the following in /etc/rsyslog.d/10-remote.conf
```

$ModLoad imudp  
$UDPServerRun 514

$template RemoteStore, "/var/log/remote/%HOSTNAME%.log

:source, isequal, "ems-esp" -?RemoteStore
& ~
```
That way the messages from ems-esp don't pollute my syslog.

@proddy is there a way to continue logging when telnet is closed? Or do I need to keep the telnet up and running?
By the way, is the v2 stable enough now?

proddy, 2020-08-30:
you don't need the telnet session open, syslog will always work. v2 I stable enough I think. I spent 6 months re-writing the code with a lot of help from @MichaelDvP. There will be issu

closed 2020-08-30 by proddy