emsesp/EMS-ESP #438: HomeAssistant mqtt autodiscovery temperature parsing error

joanwa, 2020-07-28:
Already wrote in Gitter, just adding as issue here to track. Basically the home assistant autodiscovery features publishes configs into

- `homeassistant/climate/ems-esp/hc1`
- `homeassistant/sensor/ems-esp/external`

and in my setup a state only into
`homeassistant/climate/ems-esp/hc1`

with climate sample config:
```
{
"name":"hc1",
"uniq_id":"hc1",
"~":"homeassistant/climate/ems-esp/hc1",
"mode_cmd_t":"~/cmd_mode",
"mode_stat_t":"~/state",
"temp_cmd_t":"~/cmd_temp",
"temp_stat_t":"~/state",
"curr_temp_t":"~/state",
"mode_stat_tpl":"{{value_json.hc1.mode}}",
"temp_stat_tpl":"{{value_json.hc1.seltemp}}",
"curr_temp_tpl":"{{value_json.hc1.currtemp}}",
"min_temp":"5",
"max_temp":"40",
"temp_step":"0.5",
"modes":["off","heat","auto"]
}
```
and state:
```
{
"hc1":
{"seltemp":20,
"daytemp":20,
"nighttemp":13,
"targetflowtemp":0,
"mode":"auto",
"modetype":"summer"}
}
```

and sensor state:
```
{
"dev_cla":"temperature",
"name":"ems-esp-sensor1",
"uniq_id":"ems-esp-sensor1",
"~":"homeassistant/sensor/ems-esp/external",
"stat_t":"~/state",
"unit_of_meas":"°C",
"val_tpl":"{{value_json.sensor1.temp}}"
}
```

home assistant logs report the following:
```
Log Details (ERROR)

Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:414
Integration: MQTT (documentation, issues)
First occurred: July 25, 2020, 12:35:47 AM (6112 occurrences)
Last logged: 10:12:36 PM 

Could not parse temperature from
```
That `Could not parse temperature from` seems a bit truncated, but even full logs on hass don't show more.

proddy, 2020-07-28:
yes, I see the `currtemp` is missing from the payload so HA can't render the thermostat component. Check from the console with a `show` command to whether you see "Current room temperature: xxx". If not then we know where the problem is.

joanwa, 2020-07-29:
yeap, exactly. I don't have a good solution proposal. My heating system is external temperature triggered, therefore it doesn't use a room temperature sensor. 
My previous setup with the FW200 instead of CW400 published `currtemp`, but that was more or less useless since it uses the inbuilt sensor inside the FW200, which is attached to the gas burner. So it's in the heating room, not in a room where there's a realistic and useable temperature. The CW400 doesn't have this inbuilt sensor anymore.

proddy, 2020-07-29:
I agree, no easy solution. Here's a few ideas
 - we modify the EMS-ESP code to show a "currtemp" of 0 (or any other value) if it doesn't exist. It'll make the widget in Home Assistant look silly though, so not recommended.
 - don't use HA for the MQTT format in EMS-ESP but go back to using `nested` and build the HA component yourself (for display and adjusting)
 - use a different HA component (like this [one](https://github.com/nervetattoo/simple-thermostat) I used in EMS-ESP 1.9). Then just suppress showing the current temperature. You will need to add the climate.yaml file back. I think this is the easiest solution.

norberts1, 2020-07-31:
@joanwa that CW400 has one inbuilt sensor like that Fxyz. But the reason for that decoding-problems is:
Junkers/Bosch have used different MessageID's (telegram-types) for that series of moduls. 
That they is true for all of that 'new' EMS2 like moduls like MSx00 and MMx00 and what else.
But that new modul's can understand also that older MessageID's partly.
@proddy 
For that reason it is required to decode/assign that values dependant on the modul-type (Fxyz or Cxyz).
For the value "currtemp" you can find that value (cuurent room-temperature) in the MessageID 677 at byte-position 6/7.
Anyway, that CW400 then must be mounted in the room (as I have done) and not in the heater itself.
See attached extraction from my documentation:
[image]

joanwa, 2020-08-04:
@norberts1  is that CW400 room temperature sensor only enabled if in the external housing? My CW400, which is mounted onto the boiler doesn't show any room temperature on the CW400 display itself... The FW200 did show the room temperature when mounted onto the boiler. I know this values is useless in this setup, but anyways...

norberts1, 2020-08-04:
@joanwa I had never mounted a thermostat in my boiler. They where always mounted external. So I can't realy answer your question. I know that there are 3 pins in the boiler, 2 for 2-wire bus and the third for detecting of buildin thermostat.
But my CW400 (external mounted, in my livingroom) shows that temperature. You must dig into that 'info'-menu and select that 'heizung' or 'boiler' or what ever is shown in englisch. See attached picture.
But anyway, you should find that information in the received telegramm 677_0_0 (01A5) see above.

[image]

joanwa, 2020-08-05:
@norberts1  is you heater set to "Außentemperaturgeführt" or controlled by inner room temperature?

norberts1, 2020-08-05:
@joanwa this is configureable as many other parts too (see manual). I configured it as: outdoor-controlled with based-fixpoint. But the inner room temperature is still send with that telegram. 
My configurations see attached picture:
[image]

proddy, 2020-09-05:
closing this. The problem is in HA that the standard thermostat component doesn't work without having the current temperature.

joanwa, 2020-09-07:
yeap. will come back to this once I figured out in what makes the room temperature disappear. Apart from my controller sitting on top the heater instead of wall mounted in some room, my setup looks the same, but doesn't show the  room temperature.

joanwa, 2020-09-11:
> @joanwa this is configureable as many other parts too (see manual). I configured it as: outdoor-controlled with based-fixpoint. But the inner room temperature is still send with that telegram.
> My configurations see attached picture:
> [image]

@norberts1 I just found this:

> Der CW400 hat sowohl einen Innen- als auch Außtentemperaturfühler. Wenn er [...] im Gerät eingeclipst ist dann ist der IT-Fühler deaktiviert.

https://www.haustechnikdialog.de/Forum/t/209844/Einstellungen-fuer-Regelung-CW400-an-Junkers-ZSB-14-5C?page=2

that explains why you see the values and I don't. Unfortunately this also implies that everyone who has the controller boiler-mounted can't use HA autodiscovery in the current form :(

proddy, 2020-11-18:
@joanwa so there is still an enhancement to EMS-ESP that if the thermostat is not using current room temperature, then the HA climate component should use the outdoor/outside temperature? As you know right now the climate component is not created but the data is still present in other entitites.

proddy, 2020-11-19:
duplicate of 582

closed 2020-11-19 by proddy