emsesp/EMS-ESP #441: (V2) version 2 software causes communication issues between RFM20 and RC20RF

lmdecroos, 2020-07-30:
Since is started testing 2.0 i have A01 816 , A01 832 and  A22 errors on my boiler and RC20RF.
I started testing with b5 and have seen these errors also with b6 and b7.
I have no TX errors when using mode 40 in the esp gateway but communication errors on my second heat circuit controlled by my RC20.
When I switch of ( removing batteries) of my RC20RF and disable the HC2 on my RC35 master thermostat then I have no communication errors on my boiler. I had this running for 6 days with b7 and b8 without any issue.

Today I reactivated my HC2 and reconnect my RC20RF, after a few seconds I had communication errors on my boiler and RC20RF thermostat.

I have now removed my esp gateway (bbqkees hardware powered via service bus) and will monitor during a few days my RC20RF

I have seen a similar behavior with A22 errors when testing some 1.9.5 beta versions

Any idea which part of the EMS software can disturb the communication with the RFM20 & RC20RF ?

proddy, 2020-07-30:
interesting how you go to Tx mode 40! Did you try the modes 1 and 2?

Also to be clear, did you get any errors using version 1.9? And which mode were you using in v1.9.5?

MichaelDvP, 2020-07-30:
Yes, i see there is a conflict with the roomcontroller simulation in v2, but this was not implemented in v1.9.5. 
I'll make a pr to fix this. 
If you are compiling your own, add in `roomcontrol.ccp` line 68:
```
    // no reply if the temperature is not set
    if (remotetemp[hc_] == EMS_VALUE_SHORT_NOTSET) {
        return;
    }
```

lmdecroos, 2020-07-30:
@proddy : I used mode 1 in b5 but for b6  and b7 I used mode 40 as this was the only mode whereof I had no tx errors.
Version 1.9.5 was running stable since begin of May, no errors. I used mode 1  as this is the EMS1 protocol in the 1.x release

proddy, 2020-07-30:
ok, try b9 when I've merged Michael's changes in and use tx_mode 1. Thanks for testing v2 by the way.

lmdecroos, 2020-07-30:
Thanks for the quick fix
testing b10
tx_mode 1

lmdecroos, 2020-07-31:
works 14 hours without any error on my thermostat and ems gateway
 
EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 40520
  #read requests sent: 5959
  #write requests sent: 10
  #corrupted telegrams: 0 (0%)
  #tx fails (after 3 retries): 0

proddy, 2020-08-01:
@lmdecroos is your system EMS 1.0 or EMS+ ? I can't remember.

closed 2020-08-01 by proddy

lmdecroos, 2020-08-01:
Yes EMS 1.0

commit 931dc6dd (2020-07-30): Fix #441
commit ab8e873a (2020-07-30): Merge pull request #442 from MichaelDvP/v2

Fix #441