emsesp/EMS-ESP #444: (V2) how to set ems-esp/thermostat_cmd_mode for HC2 ?

lmdecroos, 2020-07-30:
In version 1.9.5 we have  ems-esp/thermostat_cmd_mode1 for setting the mode of HC1 and  ems-esp/thermostat_cmd_mode2 for setting the mode of HC2.

In version 2.0  ems-esp/thermostat_cmd_mode is used to set the mode of HC1.
Is there a new mqtt syntax for setting the mode on HC2 ? 

MQTT message : ems-esp/thermostat_cmd_mode = day
000+00:11:30.893 I 11: [thermostat] Setting thermostat mode to day for heating circuit 1

MichaelDvP, 2020-07-31:
`thermostat_cmd_mode` works on the first active hc. For other hc use the `thermostat_cmd` with
`{"hc2":{"mode":"day"}}` or `{"hc":2, "mode":"day"}` or `{"hc":2, "cmd":"mode", "data":"day"}`

proddy, 2020-07-31:
spot on, thanks Michael. I will need to start working on updating the Wiki (https://emsesp.github.io/docs/#/MQTT) for v2

lmdecroos, 2020-07-31:
Thanks for providing the new syntax.
It works from the mqtt explorer application
I'm waiting  until bbqkees has updated his domoticz plug-in for V2

closed 2020-08-03 by proddy