emsesp/EMS-ESP #453: (v2) ESP32 WEMOS D1 32 mini compiler define question

SpaceTeddy, 2020-08-10:
Hi all,

I would like to start with compiling and test the ESP32 version and I own an [WeMos D1 32 mini](https://www.amazon.de/TTGO-Wemos-MINI-D1-ESP32-WiFi-Bluetooth-Wemos-D1-Mini-ESP8266-ESP-32-Module-Bluetooth-Internet-Entwicklung-funktionsf%C3%A4hig/dp/B07HJHYPXM) which I would like to use with the EMS-Gateway. 
Well, I've found a compiler define in the code called *WEMOS_D1_32* which switched some GPIO's for ... hopefully, my ESP32 version. Currently I can not identify how to enable this define. May someone help me? 
thanks,
chris

MichaelDvP, 2020-08-10:
edit the `pio_local.ini` and put in
```
[platformio]
default_envs = esp32

[env:esp32]
board = wemos_d1_mini32
build_flags = ${common.build_flags} ${common.debug_flags} -D WEMOS_D1_32 -D FT_NTP=1
```

SpaceTeddy, 2020-08-10:
Hi Michael,

ah Ok, missing build flags. 
will try it.

many thanks for your quick reply,
Chris

SpaceTeddy, 2020-08-10:
Works! 

thanks a lot

closed 2020-08-10 by SpaceTeddy