emsesp/EMS-ESP #454: (v2) RC20RF Error A01

SpaceTeddy, 2020-08-10:
Dear all,
I'm perfectly happy with EMS-ESP V1 version, but I'm interested to play around with the new V2 and ESP32. 
During my quick test with V2 today I recognized, that V2 behaves different to V1 with my system configuration.
I found, that after some time (10min.), my RC20RF room thermostat shows Error *A01* on its display. I'm sure, that the room  temperature was shown correctly on the display and telnet console, but after a while, the RC20RF shows that error. 

With V1 I have following devices on the bus:  
```
These 5 were recognized by EMS-ESP:
 Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID: 0x08, ProductID: 64, Version: 03.06)
 Controller: BC10/RFM20 Receiver (DeviceID: 0x09, ProductID: 68, Version: 02.03)
 Gateway: RC Remote Device (DeviceID: 0x18, ProductID: 94, Version: 02.01)
 Thermostat: RC20RF (DeviceID: 0x18, ProductID: 93, Version: 02.00)
 Thermostat: Sieger ES73 (DeviceID: 0x10, ProductID: 76, Version: 02.08) [master]
```

V2 show following devices:
```
ems-esp-2:/$ show devices
These EMS devices are currently active:

Boiler: BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:03.06)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 
 Subscribed MQTT topics: ems-esp-2/boiler_cmd ems-esp-2/boiler_cmd_wwactivated ems-esp-2/boiler_cmd_wwonetime ems-esp-2/boiler_cmd_wwcirculation ems-esp-2/boiler_cmd_wwtemp 

Thermostat: ES73 (DeviceID:0x10, ProductID:76, Version:02.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37 
 Subscribed MQTT topics: ems-esp-2/thermostat_cmd ems-esp-2/thermostat_cmd_temp ems-esp-2/thermostat_cmd_mode 

Gateway Module: RC20RF (DeviceID:0x18, ProductID:93, Version:02.00)
Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)
```

shows for me, that V2 finds  the room thermostat RC20RF (DeviceID: 0x18, ProductID: 93, Version: 02.00) as  Gateway Module and and does not find the ProductID:94

Can you advice some tests I can do that the V2 find all devices correctly?

btw: The esp8266 version behaves the same.

thanks a lot

proddy, 2020-08-10:
The RC20RF is strange. You will need to modify the platformio.ini file and compile with -DEMSESP_DEBUG so we can see which telegrams and coming in and what is being sent. Then I can add these to the test simulation and try and reproduce the bug.

the A01 error, I don't know. Let's fix the device allocation first.

SpaceTeddy, 2020-08-10:
ok, I will do so tomorrow. 
well, there were some changes regarding the RC20 and heating circuits in the past probably one of these has to do with this. 
I use the RC20RF as remote to the es73 and there should be no need to read or write to this device ID. Anyhow, hopefully I can flash and log it tomorrow. 
... to which logging level I have to set the system?

MichaelDvP, 2020-08-10:
It's a bit strange that also in v1 gateway and RC20RF has thes ame device_id 0x18. When ems-esp requests the version information from 0x18, which device replies? Both devices are in the device lib, but v2 now matches the device to gateway, but reads  the version and device description from the thermostat. 

In the log there is `000+00:17:00.018 D 30: [emsdevice] Fetching values for device ID 0x18` but sadly the log stops before the requests to 0x18 are sent. @SpaceTeddy can you do a longer log and also use `watch raw 18` to see the telegrams from/to the device. Also while logging/watching do a `scan devices` than we see what comes back to the version requests.

@proddy A01 can have different reasons, but for remote controll surly it's a ems-communication error (error code 816). 

(I suppose that the thermostat is also recognized and the RCTime fetch-request to the remote causes the error)

proddy, 2020-08-11:
> ... to which logging level I have to set the system?
@SpaceTeddy In theory you set `log debug` and use the command `scan devices` from the console but I realize this doesn't show you very verbose information so I'll need to improve that. I'm away at the moment but when I'm back I'll push a new build which also contains Michael's last PR with nice fixes.

SpaceTeddy, 2020-08-11:
Hi folks,

I can provide some log files today. Please not, that I have made two logs. a) switch on Heater and EMS-ESP at the same time; b) heater is running and EMS-ESP is reset by button.
```
a) 
[ems-esp-v2_log_working_till_error_heater_and_esp32_on_same_time.txt](https://github.com/proddy/EMS-ESP/files/5055789/ems-esp-v2_log_working_till_error_heater_and_esp32_on_same_time.txt)

b)
[ems-esp-v2_log_working_till_error_heater_on_esp32_reset.txt](https://github.com/proddy/EMS-ESP/files/5055790/ems-esp-v2_log_working_till_error_heater_on_esp32_reset.txt)
```

with my little knowledge I can see that, depends on the startup, the found devices are different. But both result in the error of the RC20RF remote. Logs are made till the Error occoures + some seconds.

Hope that helps.

Error of the RF20M shows: EMS-Bus Error:
[image]

MichaelDvP, 2020-08-11:
> ┌──────────────────────────────────────────┐
│ EMS-ESP version 2.0.0b7                  │

There are more than 100 commits since that version. Please use the actual version (or wait for proddys next commit) 
It's possible that the A01-bus-errors are caused by the old uart and not by the gateway/thermostat mismatch. 

Without `watch` it's difficult to check the logs, but here  the gateway is detected:
`000+00:00:10.647 D 41: [emsesp] Adding new device with device ID 0x18 with product ID 94 and version 02.01`
and here the thermostat only updates the version-info of the gateway instead of registering a thermostat-device (the update loop in `EMSESP::add_device`)
`000+00:01:04.042 D 303: [emsesp] Updating details to already existing device ID 0x18`

SpaceTeddy, 2020-08-11:
oh dear,
I've updated to the latest V2 commit and since 1h no errors and also stable device detection. I still watching to the RC20RF but I guess no action on your side is needed. If it runs for 24h, I will close this issue.

sry guys,
Chris

closed 2020-08-14 by SpaceTeddy