emsesp/EMS-ESP #456: Hometop Ht3 adapter ht_pitiny with EMS-Esp?

URBANsUNITED, 2020-08-14:
Hi!

Currently I am using the hometop HT3 software for monitoring my Junkers heating system. Now i‘ve discovered this system and want to implement it with my HA setup too.

Has anybody managed to get the ht3 ht_pitiny adapter with EMS-ESP working? If yes, how please?

Link to my current system

https://github.com/norberts1/hometop_HT3

Many thanks in advance

URBANSUNITED

proddy, 2020-08-14:
I think Norbert can best answer that question. EMS-ESP, as the name implies uses, the ESP8266 or ESP32 microcontroller to do all the hard work. You still need a circuit to convert the HT3 messages from the TTL line into serial data, for both reading and writing. bbqkees sells cheap boards (see link on the wiki). He also has one for the RPi I believe but this software is not compatible.

norberts1, 2020-08-14:
@URBANsUNITED  @proddy: both hardware can be connected in **parallel** to the same heater-Bus (EMSx or heatronic). They are using different device-ID's, so no traffic-collision is on the heater-bus.
EMS-ESP is using fixed device-ID: 0x0B  (service-key).
ht_pitiny is using default device-ID: 0x0D (modem) but can be configured also to other once's if required.
I have checked this on my heater-system with EMS-ESP rev. 1.9.5 and ht_pitiny connected in parallel to the heater-bus.
The communication of them is done with MQTT for receiving decoded data and sending commands to the heater-system.
For the HA bindings on EMS-ESP see the wiki.
For that HT3/EMS pi_tiny-Adapter HA bindings are not yet available in the project, but I know some forum-members has already created that HA binding.
So, if you like you can run them in parallel on the same heater-bus.

proddy, 2020-09-05:
the EMS-ESP can use a selection of deviceIDs. I'll close this now but feel free to re-open and share some feedback and how far you got.

closed 2020-09-05 by proddy