emsesp/EMS-ESP #464: (v2) HA ems-esp-sensor1: unknown

TimoFriedri, 2020-08-26:
Hi,
I have some issues with the mqtt interaction with Home Assistant:
First some basic info:
- esp8266
- v2b13 flashed
- mqtt switched to Home Assistant
- HOme Assistant 0.114.3 with official MQTT addon, discovery enabled

It creates 2 entities:
- sensor.ems_esp_sensor1
- climate.hc1

Now, for some reason I get error on ems-esp and HA side related to this:
**HA side:**
Log:
```
2020-08-26 15:37:59 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from
```
HA MQTT:
```
Message 4 received on homeassistant/climate/ems-esp/hc1/state at 15:41:
{
    "hc1": {
        "seltemp": 0,
        "daytemp": 21,
        "nighttemp": 18,
        "holidaytemp": 16,
        "nofrosttemp": 2.5,
        "heatingtype": 1,
        "targetflowtemp": 5,
        "offsettemp": 0,
        "designtemp": 75,
        "summertemp": 19,
        "mode": "auto",
        "modetype": "summer"
    }
}
QoS: 0 - Retain: false
Message 3 received on homeassistant/climate/ems-esp/hc1/state at 15:41:
{
    "hc1": {
        "seltemp": 0,
        "daytemp": 21,
        "nighttemp": 18,
        "holidaytemp": 16,
        "nofrosttemp": 2.5,
        "heatingtype": 1,
        "targetflowtemp": 5,
        "offsettemp": 0,
        "designtemp": 75,
        "summertemp": 19,
        "mode": "auto",
        "modetype": "summer"
    }
}
QoS: 0 - Retain: false
Message 2 received on homeassistant/climate/ems-esp/hc1/state at 15:41:
{
    "hc1": {
        "seltemp": 0,
        "daytemp": 21,
        "nighttemp": 18,
        "holidaytemp": 16,
        "nofrosttemp": 2.5,
        "heatingtype": 1,
        "targetflowtemp": 5,
        "offsettemp": 0,
        "designtemp": 75,
        "summertemp": 19,
        "mode": "auto",
        "modetype": "summer"
    }
}
QoS: 0 - Retain: false
Message 1 received on homeassistant/climate/ems-esp/hc1/config at 15:41:
{
    "name": "hc1",
    "uniq_id": "hc1",
    "~": "homeassistant/climate/ems-esp/hc1",
    "mode_cmd_t": "~/cmd_mode",
    "mode_stat_t": "~/state",
    "temp_cmd_t": "~/cmd_temp",
    "temp_stat_t": "~/state",
    "curr_temp_t": "~/state",
    "mode_stat_tpl": "{{value_json.hc1.mode}}",
    "temp_stat_tpl": "{{value_json.hc1.seltemp}}",
    "curr_temp_tpl": "{{value_json.hc1.currtemp}}",
    "min_temp": "5",
    "max_temp": "40",
    "temp_step": "0.5",
    "modes": [
        "off",
        "heat",
        "auto"
    ]
}
QoS: 0 - Retain: true
Message 0 received on homeassistant/sensor/ems-esp/external/config at 15:41:
{
    "dev_cla": "temperature",
    "name": "ems-esp-sensor1",
    "uniq_id": "ems-esp-sensor1",
    "~": "homeassistant/sensor/ems-esp/external",
    "stat_t": "~/state",
    "unit_of_meas": "°C",
    "val_tpl": "{{value_json.sensor1.temp}}"
}
QoS: 0 - Retain: true
```

**ems-esp side:**
WebUI: MQTT Publish Errors: [high number]

telnet show:
```
Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:03.06)                                                                                                                                                                      
  Hot tap water: off                                                                                                                                                                                                                     
  Central heating: off                                                                                                                                                                                                                   
  Warm Water activated: on                                                                                                                                                                                                               
  Warm Water circulation pump available: on                                                                                                                                                                                              
  Warm Water circulation pump freq: 2x3min                                                                                                                                                                                               
  Warm Water circulation active: off                                                                                                                                                                                                     
  Warm Water comfort setting: Hot                                                                                                                                                                                                        
  Warm Water disinfection temperature: 70°C                                                                                                                                                                                              
  Warm Water selected temperature: 55°C                                                                                                                                                                                                  
  Warm Water set temperature: 55°C                                                                                                                                                                                                       
  Warm Water current temperature (intern): 50.3°C                                                                                                                                                                                        
  Warm Water current temperature (extern): 50.3°C
  Warm water storage temperature (extern): 50.3°C
  Warm Water current tap water flow: 0.0l/min
  Warm Water # starts: 7742
  Warm Water active time: 115 days 5 hours 40 minutes
  Warm Water charging: off
  Warm Water disinfecting: off
  Selected flow temperature: 7°C
  Current flow temperature: 39.9°C
  Max boiler temperature: 45.4°C
  Return temperature: 43.9°C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Burner selected max power: 0%
  Burner current power: 0%
  Flame current: 0.0uA
  System pressure: 1.4bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 90°C
  Boiler circuit pump modulation max power: 91%
  Boiler circuit pump modulation min power: 30%
  Boiler circuit pump delay time: 5min
  Boiler temp hysteresis on: -6°C
  Boiler temp hysteresis off: 6°C
  Boiler burner min period: 10min
  Boiler burner min power: 0%
  Boiler burner max power: 100%
  Outside temperature: 22.0°C
  Pump modulation: 0%
  Burner # starts: 72730
  Total burner operating time: 1976 days 9 hours 5 minutes
  Total heat operating time: 1861 days 3 hours 25 minutes

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.11)
  Clock: 15:48:08 26/08/2020
  Warm Water mode: auto
  Damped Outdoor temperature: 21°C
  Temp sensor 1: 24.4°C
  Temp sensor 2: 25.0°C
  Offset int. temperature: 0.0°C
  Min ext. temperature: -10°C
  Building: medium
  Heating Circuit 1:
    Setpoint room temperature: 0.0°C
    Mode: auto
    Mode Type: day
    Program is set to Summer mode
    Day temperature: 21.0°C
    Night temperature: 18.0°C
    Nofrost temperature: 2.5°C
    Holiday temperature: 16.0°C
    Offset temperature: 0.0°C
    Design temperature: 75°C
    Summer temperature: 19°C
    Target flow temperature: 5°C
```
log debug:
every 10 seconds:
```
000+04:40:42.202 D 66: [mqtt] Publishing topic homeassistant/climate/ems-esp/hc1/state (#3679, attempt #1, pid 1)
```
occasionally:
```
000+04:41:02.152 D 102: [mqtt] Publishing topic ems-esp/boiler_data (#3683, attempt #1, pid 1)
000+04:41:01.549 D 97: [mqtt] Publishing topic ems-esp/heartbeat (#3682, attempt #1, pid 1)
```
show mqtt:
```
MQTT is connected
MQTT Heartbeat is enabled
MQTT publish fails: 38

MQTT subscriptions:
 topic: ems-esp/boiler_cmd, [cmd]: comfort wwactivated wwtapactivated wwonetime wwcirculation flowtemp wwtemp burnmaxpower burnminpower boilhyston boilhystoff burnperiod pumpdelay
 topic: ems-esp/system_cmd, [cmd]: pin send pin send pin send
 topic: ems-esp/thermostat_cmd, [cmd]: wwmode temp mode nighttemp daytemp nofrosttemp remotetemp datetime minexttemp calinttemp building control pause party holiday summertemp designtemp offsettemp holidaytemp
 topic: homeassistant/climate/ems-esp/hc1/cmd_temp
 topic: homeassistant/climate/ems-esp/hc1/cmd_mode

MQTT queue is empty
```
I occasionally saw 3 consecutive failed mqtt publish related logs, but unfortunately they don't show up the last minutes for me to copy. I'll add this as soon as I see it again.

Here is how it looks in HA.
 
[image]

Anyone an idea what's happening? How is it supposed to look like?

proddy, 2020-08-26:
This is the same as issue #438 (See https://github.com/proddy/EMS-ESP/issues/438#issuecomment-665284950)

Basically you don't have the current room temperature either in the console or the payload so the Home Assistant thermostat will not work as it needs both the currtemp and setpoint temps.

What EMS devices do you have? Is the current temperature visible in the thermostat? If so can you find that value back in one of the telegrams? Did it work in v1.9 ?

MichaelDvP, 2020-08-26:
@proddy: i think the confusion is also that the dallas sensor 1 configuration is published to HA, but no sensor is attached (and no value published). It's the configuration-publish in `Sensors::start() {`.
I'm not familiar with HA, but is it possible to send the HA-sensor-configuaration after the first scan, before first value publish, than you also know the number of attached sensors.

@TimoFriedri: As seen on gitter you have GB162, BC10 and RC35. Ther RC35 is mounted on the boiler and not as room-thermostat, so it does not have a room temperature.  Thr internal sensors of RC35 are read as `Temp sensor 1: 24.4°C` and `Temp sensor 2: 25.0°C`, but they are not assigned to a hc  room temperature. If you connect the RC35 with a cable you can select in the menu "weather-controlled with room influence" or "roomcontrolled" and than the internal temperature will be shown als roomtemperature.

proddy, 2020-08-26:
@MichaelDvP I'll push a fix to send the MQTT discovery for dallas sensors when the first sensor is detected. Its harmless but may confuse people. I'll also look into the mode naming

MichaelDvP, 2020-08-26:
@proddy The other thing is, as you mentioned, the missing currtemp where HA expects 
`"curr_temp_tpl": "{{value_json.hc1.currtemp}}",`
If HA also accepts a string you can publish the notset-value as `"{hc1":{"currtemp":"-.-"}}` to make clear that there is no measurement and to avoid the error-message from @TimoFriedri : 
`2020-08-26 15:37:59 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from`

proddy, 2020-08-26:
@MichaelDvP that won't work, unfortunately. That particular card in Home Assistant (https://www.home-assistant.io/lovelace/thermostat/) expects a numerical value for the current temperature. The solution is to use a different HA card or build a custom one which is quite easy. There are many freely available on github and I recommended one in #438.

TimoFriedri, 2020-08-26:
Hi,
thanks for all the feedback. I am not sure if I shall test something now!?
You are right, the RC35 is mounted in the basement next to the heating unit, so I am only weather-controlled, no roomtemperature feedback. Even though I thought about mounting the RC35 on the 1. floor.

Isn't this a big issue for many people. I think this control setup is quite common here (Germany). I think this should be working with the default  thermostat card.

proddy, 2020-08-26:
I think the solution is to create a custom HA card. The default specification for climate in Home Assistant is to have a current temperature and setpoint temperature (how else would a smart thermostat know when to adjust the temp?). What are you trying to control with HA? Perhaps start with the use case and then we can work on building the feature. In any case, it's not a bug in EMS-ESP.

proddy, 2020-08-27:
I'm modifying the code to only send the HA config when an external dallas sensor is detected (which is the title of this issue). The home assistant card is not something EMS-ESP can control as it just sends out data when it has it. We can continue that discussion in #438

closed 2020-08-27 by proddy