emsesp/EMS-ESP #466: (v2) add GPIOs to WebUI as configurable options (LED, Rx, Tx, Dallas sensor)

proddy, 2020-08-28:
To support a new line of boards that BBQKees is working on (variants of ESP32 for example) make all the GPIOs configurable in the Web Settings page.

proddy, 2020-08-29:
[image]

MichaelDvP, 2020-08-29:
Can you also add the led polarity, it is different for some boards. Maybe as checkbox near `Hide LED` or combine it with hide as a select-box with `high/low/hide`. Also i think it's better to change the position dallas an LED, than led-gpio is next to hide-LED.

proddy, 2020-08-29:
Yup. I'll merge into Hide

MichaelDvP, 2020-08-29:
Ups, i see the hide logic simply inverses the polarity, than it's no need to add an extra setting.  My fault.

proddy, 2020-08-29:
I left the logic in `system.h`:

```c
// internal LED
#ifndef EMSESP_NO_LED
#if defined(ESP8266)
    static constexpr uint8_t LED_ON = LOW;
#elif defined(ESP32)
#ifdef WEMOS_D1_32
    static constexpr uint8_t LED_ON = HIGH;
#else
    static constexpr uint8_t LED_ON = LOW;
#endif
#endif
#else
    static constexpr uint8_t LED_ON = 0;
#endif
```

was your idea to move this to the Web and have a new parameter called led_mode which has values HIGH, LOW, OFF ?

MichaelDvP, 2020-08-29:
This was the first idea, but the `Hide LED` inverts the LED logic and for the `WEMOS_D1_32`  it's only nessesary to check `Hide LED`to activate it. Maybe the naming is confusing in this case, `Invert LED` instead of `Hide LED` meets it better in this case.

MichaelDvP, 2020-08-29:
Reduced the above to
```
// internal LED
    static constexpr uint8_t LED_ON = LOW;
```
this is the tested and working setting for `WEMOS_D1_32` with acitvated LED (only renamed hide option, no compile flag needed):
[image]

proddy, 2020-08-30:
The feature request was intended to disable the LED entirely when there is a good connection to the EMS bus and WiFi. So only flash when there is something wrong. Perhaps we need another word for it since it does many things now?

MichaelDvP, 2020-08-30:
Yes, and with the configurable gpio you have all options, no need to change anything in the logic.
1. gpio 0: led always off
2. gpio set to valid led: led flashing if something is wrong, goes to on or off with good connection, depending on led polarity
3. Hide LED (Invert LED) checked: same as 2., flashing on error, but other state on good connection

But on D1_mini_32  with inverted led it looks a bit odd to check `Hide LED` for activating the led on good connection.

proddy, 2020-08-30:
ok lets rename hide_led to invert_led everywhere in the code then to be consistent. Shall I do it?

MichaelDvP, 2020-08-30:
[image]

No code change needed ;-)

proddy, 2020-08-30:
beautiful!

closed 2020-08-31 by proddy