emsesp/EMS-ESP #467: Dallas 18B20

frind844, 2020-08-29:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://github.com/proddy/EMS-ESP/wiki/Troubleshooting)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session using the `system` command and `info`.*

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2020-08-30:
Yes , the DS18S20 uses a different number format, you have to multiply the value by 8 to get the temperature. 
If you can compile insert in `sensor.cpp` line 186:
```
    if (addr[0] == TYPE_DS18S20) {
        raw_value = (raw_value << 3) + 12 - scratchpad[SCRATCHPAD_CNT_REM];
    } else {
        // Adjust based on device resolution
```
close the `}` after the resolution calculation and set `static constexpr size_t SCRATCHPAD_CNT_REM  = 6;` to the header.

frind844, 2020-08-30:
thank you for the quick answer.

proddy, 2020-08-30:
@MichaelDvP wanna add this to 2.0.0b1 in the dev branch?

MichaelDvP, 2020-08-30:
When working on sensors, what do you think about this:
- The resolution is 1/16 degree, i.e. 0.0625°C and you display it with two decimals, but i found that the last decimal is more or less a random number. I think it's better to round the values to one decimal. 
- the values on the dashboard are centered and not in line if different length, also not in line with the boiler/thermostat values. I set all values right aligned and sensors `.toFixed(1)`.

[image]

Should i add this too?

proddy, 2020-08-30:
yes, these are great additions.

I also noticed the Brand column is empty. I'll create a new issue for this.

commit d6c6c718 (2020-08-30): Merge pull request #473 from MichaelDvP/dev

Fix #467 and master thermostat check
closed 2020-08-30 by proddy