emsesp/EMS-ESP #469: (V2) set master settings are not preserved

lmdecroos, 2020-08-30:
I have a RC35 and a RC20 thermostat.
The RC35 is my master thermostat, until 2.0b10 the gateway software was always using the RC35 as master.

When using the 2.0 release version the gateway chooses my RC20 as master.
The "thermostat" "set master" setting is working, but the settings are not preserved

**at reboot :** 
ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 

**Thermostat: RFM20 Remote (DeviceID:0x18, ProductID:94, Version:02.01) ** master device ****
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 

**setting the master to my RC35 ( device id 10)**

 ems-esp:/$ thermostat
 ems-esp:/thermostat$ su
 Password: 
 000+00:01:13.492 N 0: [shell] Admin session opened on console
 ems-esp:/thermostat# set master 10
 Master Thermostat Device ID = 10

**Checking the changes**

 ems-esp:/thermostat# exit
 ems-esp:/# show devices
 These EMS devices are currently active:

 Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
  This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 

 Thermostat: RFM20 Remote (DeviceID:0x18, ProductID:94, Version:02.01)
  This Thermostat will respond to telegram type IDs: 0xA3 0x06 

 Thermostat: RC20RF (DeviceID:0x19, ProductID:93, Version:02.00)
  This Thermostat will respond to telegram type IDs: 0xAF 

 Thermostat: RFM20 Remote (DeviceID:0x1A, ProductID:94, Version:02.01)

 **Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:21.08) ** master device **
  This Thermostat will respond to telegram type IDs: 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37 

The set master are taken in account for the current session but are are not stored and lost at reboot

**when checking again** 
 ems-esp:/thermostat# set master 
Master Thermostat Device ID = auto
ems-esp:/thermostat#

MichaelDvP, 2020-08-30:
The command `set master` without parameter sets it to default (auto), to check the setting use the command `set`

lmdecroos, 2020-08-30:
Ok thanks, this makes sense for the behaviour I have seen

ems-esp:/$ thermostat
ems-esp:/thermostat$ set
Master Thermostat Device ID = 10

Now it sticks

proddy, 2020-08-30:
@MichaelDvP I think we should standardize on the `set` commands, so `set` shows the settings in that context and `set <something>` with no value reports an error, to avoid confusion. I'm pretty sure I used all variations in the code

MichaelDvP, 2020-08-30:
@lmdecroos Are the thermostats  `RFM20 Remote` with DeviceIDs `0x18` and `0x1A` real thermostats or only unused channels on the RF-Gateway? The RC20RF is clearly the remote for hc2 and type-id 0xAF is correct, do you have other hc activated?

@proddy : Type-ids 0x06 and 0xA3 should always be mapped to the master thermostat, i'll check this.
As far as i see it's the only `set <something>` command with optional parameter, all others are mandatory.

proddy, 2020-08-30:
@MichaelDvP I'll make the change in the dev branch then.

lmdecroos, 2020-08-30:
@MichaelDvP  The RC35 is the master thermostat and used for HC1, the RC20RF is the thermostat for my HC2

The RFM20 is the controller who is connected to the EMS bus and communicates wireless with the RC20RF thermostat

Thermostat: RFM20 Remote (DeviceID:0x1A, ProductID:94, Version:02.01)

Thermostat: RC20RF (DeviceID:0x19, ProductID:93, Version:02.00)

MichaelDvP, 2020-08-30:
@lmdecroos Thanks for the information.
Can you please check with the latest dev-built the `show devices` command for `set master 0` and `set master 10` (both also after esp reboot). The type-ids should be none for the RFM20, 0xAF for the RC20RF and RC35 with all other ids including 0x06.

lmdecroos, 2020-08-30:
@MichaelDvP

Upgraded to 2.0.1b1

ems-esp:/$ thermostat
ems-esp:/thermostat$ su
Password: 
000+00:05:25.232 N 0: [shell] Admin session opened on console
ems-esp:/thermostat# set master 
Not enough arguments for command
ems-esp:/thermostat# set 
Master Thermostat Device ID = 10

ems-esp:/thermostat#

ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 

Thermostat: RFM20 Remote (DeviceID:0x18, ProductID:94, Version:02.01)

Thermostat: RC20RF (DeviceID:0x19, ProductID:93, Version:02.00)
 This Thermostat will respond to telegram type IDs: 0xAF 

Thermostat: RFM20 Remote (DeviceID:0x1A, ProductID:94, Version:02.01)

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:21.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37 

Mixing Module: MM10 (DeviceID:0x21, ProductID:69, Version:02.01)
 This Mixing Module will respond to telegram type IDs: 0xAA 0xAB 0xAC 

Switching Module: WM10 (DeviceID:0x11, ProductID:71, Version:02.00)

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)

Changed set master to 0

ems-esp:/thermostat# set master 0
Master Thermostat Device ID = auto
ems-esp:/thermostat# set
Master Thermostat Device ID = auto

ems-esp:/thermostat# 

ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9 

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:21.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37 

Thermostat: RFM20 Remote (DeviceID:0x18, ProductID:94, Version:02.01)

Thermostat: RC20RF (DeviceID:0x19, ProductID:93, Version:02.00)
 This Thermostat will respond to telegram type IDs: 0xAF 

Thermostat: RFM20 Remote (DeviceID:0x1A, ProductID:94, Version:02.01)

Mixing Module: MM10 (DeviceID:0x21, ProductID:69, Version:02.01)
 This Mixing Module will respond to telegram type IDs: 0xAA 0xAB 0xAC 

Switching Module: WM10 (DeviceID:0x11, ProductID:71, Version:02.00)

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)

I did a few reboots in auto mode and the gateways is always recognising the RC35 as master. 
Will this stick or should I set the master to 10 ( RC35) ?

MichaelDvP, 2020-08-30:
> I did a few reboots in auto mode and the gateways is always recognising the RC35 as master.

Great. Thanks for the feedback. Also type-Ids are recognized correctly, that was the goal.

> Will this stick or should I set the master to 10 ( RC35) ?

Leave it to auto and if it really does not stick, report here.

proddy, 2020-09-05:
@lmdecroos can we close this?

lmdecroos, 2020-09-07:
@proddy Yes you can close this issue

closed 2020-09-07 by proddy