emsesp/EMS-ESP #470: Displaying the Brand in the webUI

proddy, 2020-08-30:
_Originally posted by @MichaelDvP in https://github.com/proddy/EMS-ESP/issues/467#issuecomment-683399357_

[image]

I noticed the Brand column is empty and would have expected to see a value at least for the Boiler. My Nefit shows up, not sure why your Buderus stuff doesn't. The value is stored in the Version(0x02) telegram, the 10th byte (`decode_brand(telegram->message_data[9])`). Can you see if this comes back from Device ID 0x08? Perhaps we need to explicity request it.

In any case I was thinking in the WebUI dropping the Brand column and pre-fixing the Brand to the Model value. What do you think?

MichaelDvP, 2020-08-30:
Only the boiler give a value in pos 9, but 00, i.e. no brand.  All modules are labled `Buderus`  
```
ems:/# watch raw 02
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x02
ems:/# send telegram "0B 90 02 00 20"
000+03:57:59.447 N 1: [telegram] Rx: 10 0B 02 00 56 01 12 6C
ems:/# send telegram "0B 88 02 00 20"
000+03:58:24.094 N 2: [telegram] Rx: 08 0B 02 00 48 03 00 4B 04 1C 00 00 00 00 D3
ems:/# send telegram "0B 89 02 00 20"
000+03:59:05.607 N 3: [telegram] Rx: 09 0B 02 00 44 02 03 25
ems:/# send telegram "0B A1 02 00 20"
000+03:59:31.003 N 4: [telegram] Rx: 21 0B 02 00 45 02 01 D9

```

proddy, 2020-08-30:
strange. Shall I remove the column from the WebUI then?

MichaelDvP, 2020-08-30:
No, i think with the empty column it's clear that the brand is not set. 
Maybe i have a illegal china clone which is compatible to all brands, that's why all tx-modes working for me ;-)

closed 2020-08-30 by MichaelDvP