emsesp/EMS-ESP #471: (v2) ESP32 upload firmware

sq4bja, 2020-08-30:
Hi,
I`m trying to change esp8266 to ESP32 board. And I don`t know how to upload firmware. I`m trying PlatformIO as shown on readme, but it doesn`t work. I thing some paths are incorrect (ex. XX\). Can You explain, much easier, how to start using ESP32 with EMS-ESP?
Tom

MichaelDvP, 2020-08-30:
If connected to USB you can edit the `pio_local.ini` and add (check your com-port):
```
[env:esp32]
upload_protocol = esptool
upload_port = COM13
upload_flags = 
```
and than use `platformio upload`

proddy, 2020-08-30:
that's it. The command to use is `pio run -t upload`

sq4bja, 2020-08-30:
Thanks, but I can't compile it. I'm not PIO expert, so it looks like wrong path or incorrect library.
I'm getting:

```
Compiling .pio\build\esp32\src\console.cpp.o
In file included from src\emsesp.h:36:0,
                 from src\EMSESPDevicesService.cpp:20:
lib/framework/ESP8266React.h:33:21: fatal error: WWWData.h: No such file or directory

*****************************************************************
* Looking for WWWData.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WWWData.h"
* Web  > https://platformio.org/lib/search?query=header:WWWData.h
*
*****************************************************************

In file included from src\emsesp.h:36:0,
                 from src\EMSESPSettingsService.cpp:20:
lib/framework/ESP8266React.h:33:21: fatal error: WWWData.h: No such file or directory

*****************************************************************
* Looking for WWWData.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WWWData.h"
* Web  > https://platformio.org/lib/search?query=header:WWWData.h
*
*****************************************************************

In file included from src\emsesp.h:36:0,
                 from src\EMSESPStatusService.cpp:20:
lib/framework/ESP8266React.h:33:21: fatal error: WWWData.h: No such file or directory

*****************************************************************
* Looking for WWWData.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WWWData.h"
* Web  > https://platformio.org/lib/search?query=header:WWWData.h
*
*****************************************************************

compilation terminated.
compilation terminated.
compilation terminated.
In file included from src\emsesp.h:36:0,
                 from src\console.cpp:20:
lib/framework/ESP8266React.h:33:21: fatal error: WWWData.h: No such file or directory

*****************************************************************
* Looking for WWWData.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WWWData.h"
* Web  > https://platformio.org/lib/search?query=header:WWWData.h
*
*****************************************************************

compilation terminated.
*** [.pio\build\esp32\src\EMSESPSettingsService.cpp.o] Error 1
*** [.pio\build\esp32\src\EMSESPStatusService.cpp.o] Error 1
*** [.pio\build\esp32\src\EMSESPDevicesService.cpp.o] Error 1
*** [.pio\build\esp32\src\console.cpp.o] Error 1
================================================================================================== [FAILED] Took 11.62 seconds ==================================================================================================
Environment    Status    Duration
-------------  --------  ------------
esp8266        IGNORED
esp32          FAILED    00:00:11.617
============================================================================================= 1 failed, 0 succeeded in 00:00:11.617 =============================================================================================The terminal process "C:\Users\CP24\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

proddy, 2020-08-30:
The missing file is generated by a script which platformio calls. What I'll do is create a script that uploads it to usb

MichaelDvP, 2020-08-30:
@sq4bja the missing file is the webdata which is build by NPM. If NPM is missing you have to install [nodejs](https://nodejs.org/en/download/)

sq4bja, 2020-08-30:
Ok, that was clue. After install nodejs I have no errors and I have ems-esp on ESP32 directly from PlatformIO. So, which pins are recommended to do EMS communication?
Tom

proddy, 2020-08-30:
The defaults should work. Any pin on the ESP32 can be a UART Rx/Tx.

On a Wemos D1-32 Michael uses Rx=23(D7), Tx=5(D8) and on my Lolin D32's I used Rx=17 and Tx=16. But any should work.

sq4bja, 2020-08-30:
Ok, th edafault is GPIO5 for TX and GPIO23 for RX. However, esp32 takes more power then Wemos. Powering from the EMS bus requires adding a capacitor on the step-down input. Now I have a stable and fast boiler controller. Thank you for your help.
Tom

closed 2020-08-31 by proddy