emsesp/EMS-ESP #474: V2 Logamatic TC100

Sonusss, 2020-08-30:
Having upgraded to V2.0.0 I cannot get thermostat data anymore.
EMS-ESP just reports thermostat type like this (command show devices):
Thermostat: Logamatic TC100/Moduline Easy (DeviceID:0x18, ProductID:202, Version:02.21) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 

It was working on 1.9.3

proddy, 2020-08-30:
That's indeed a bug. Will fix.

proddy, 2020-08-30:
fixed in the dev branch.

```
ems-esp:/# show devices
These EMS devices are currently active:

Thermostat: Bosch Logamatic TC100/Moduline Easy (DeviceID:0x18 ProductID:202, Version:02.21) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x0A 
```

Sonusss, 2020-08-31:
Well,

Having upgraded to 2.0.1b1 now there are no more detected devices on the bus...
Tx mode is 1 as before.
Downgrading to 2.0.0 gives  devices but no thermostat data.

proddy, 2020-08-31:
with 2.0.1b1 what did it detect exactly?

Sonusss, 2020-08-31:
Just nothing like if EMS bus isn't connected

proddy, 2020-08-31:
Any reported tx errors?

Sonusss, 2020-09-01:
Hello,
I made some screenshots
[image]
[image]
[image]
[image]

proddy, 2020-09-01:
could you open the telnet console and type `show`. Turn logging on `log all`. If still nothing happens try `scan`

proddy, 2020-09-01:
it could be related to a uart change in 2.0.1. If you're ok building the firmware yourself try replace `src/uart/emsuart_esp8266.cpp` with this one: [emsuart_esp8266.zip](https://github.com/proddy/EMS-ESP/files/5154625/emsuart_esp8266.zip)

Sonusss, 2020-09-02:
Mmmmh, 

Even telnet isn't performing well. It's behavior is erratic, closing connexion suddenly, not geting typed characters ...
I cannot get the connexion for more than 30 seconds.

**This is what show reports:**

ems-esp:/$ show
EMS-ESP version 2.0.1b1

Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.08)

Thermostat: Logamatic TC100/Moduline Easy (DeviceID:0x18, ProductID:202, Version:02.21)

Dallas temperature sensors:
  ID: 28-FF31-7650-1883-2F, Temperature: 20.3°C

**This is log results:**
ems-esp:/$ log all
Log level = all
000+00:00:12.025 D 0: [telegram] Sending read Tx [#0], telegram: 0B 88 07 00 20 A8
000+00:00:12.076 D 1: [emsesp] Last Tx read successful
000+00:00:12.081 D 2: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
000+00:00:12.081 D 3: [telegram] Tx read request to device 0x08 for type ID 0x02
000+00:00:12.081 D 4: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
000+00:00:12.081 D 5: [telegram] Tx read request to device 0x09 for type ID 0x02
000+00:00:12.081 D 6: [emsesp] New EMS device detected with ID 0x18. Requesting version information.
000+00:00:12.081 D 7: [telegram] Tx read request to device 0x18 for type ID 0x02
000+00:00:12.129 D 8: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
000+00:00:12.129 D 9: [telegram] Tx read request to device 0x08 for type ID 0x02
000+00:00:12.129 D 10: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
000+00:00:12.129 D 11: [telegram] Tx read request to device 0x09 for type ID 0x02
000+00:00:12.129 D 12: [emsesp] New EMS device detected with ID 0x18. Requesting version information.
000+00:00:12.129 D 13: [telegram] Tx read request to device 0x18 for type ID 0x02
000+00:00:12.449 D 14: [telegram] Sending read Tx [#1], telegram: 0B 88 02 00 20 BC
000+00:00:12.523 D 15: [emsesp] Last Tx read successful
000+00:00:12.528 D 16: [boiler] Adding new Boiler with device ID 0x08
000+00:00:12.528 D 17: [mqtt] Registering MQTT cmd comfort with topic boiler_cmd
000+00:00:12.528 D 18: [mqtt] Registering MQTT cmd wwactivated with topic boiler_cmd
000+00:00:12.528 D 19: [mqtt] Registering MQTT cmd wwtapactivated with topic boiler_cmd
000+00:00:12.528 D 20: [mqtt] Registering MQTT cmd wwonetime with topic boiler_cmd
000+00:00:12.528 D 21: [mqtt] Registering MQTT cmd wwcirculation with topic boiler_cmd
000+00:00:12.528 D 22: [mqtt] Registering MQTT cmd flowtemp with topic boiler_cmd
000+00:00:12.528 D 23: [mqtt] Registering MQTT cmd wwtemp with topic boiler_cmd
000+00:00:12.528 D 24: [mqtt] Registering MQTT cmd burnmaxpower with topic boiler_cmd
000+00:00:12.528 D 25: [mqtt] Registering MQTT cmd burnminpower with topic boiler_cmd
000+00:00:12.528 D 26: [mqtt] Registering MQTT cmd boilhyston with topic boiler_cmd
000+00:00:12.528 D 27: [mqtt] Registering MQTT cmd boilhystoff with topic boiler_cmd
000+00:00:12.528 D 28: [mqtt] Registering MQTT cmd burnperiod with topic boiler_cmd
000+00:00:12.528 D 29: [mqtt] Registering MQTT cmd pumpdelay with topic boiler_cmd
000+00:00:12.528 D 30: [emsesp] Adding new device with device ID 0x08 with product ID 123 and version 06.08
000+00:00:12.528 D 31: [emsesp] Fetching values for device ID 0x08
000+00:00:12.528 D 32: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:00:12.528 D 33: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:00:12.528 D 34: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:00:12.536 D 35: [telegram] Sending read Tx [#2], telegram: 0B 89 02 00 20 B4
000+00:00:12.536 D 36: [emsesp] Last Tx read successful
000+00:00:12.536 D 37: [emsesp] Adding new device with device ID 0x09 with product ID 125 and version 03.03
000+00:00:12.536 D 38: [emsesp] Fetching values for device ID 0x09
000+00:00:13.155 D 39: [mqtt] Subscribing to topic: ems-esp/boiler_cmd
000+00:00:13.196 D 40: [telegram] Sending read Tx [#3], telegram: 0B 98 02 00 20 3C
000+00:00:13.240 D 41: [emsesp] Last Tx read successful
000+00:00:13.245 D 42: [thermostat] Adding new thermostat with device ID 0x18 (as master)
000+00:00:13.245 D 43: [telegram] Tx read request to device 0x18 for type ID 0x0A
000+00:00:13.245 D 44: [emsesp] Adding new device with device ID 0x18 with product ID 202 and version 02.21
000+00:00:13.245 D 45: [emsesp] Fetching values for device ID 0x18
000+00:00:13.669 D 46: [telegram] Sending read Tx [#4], telegram: 0B 88 02 00 20 BC
000+00:00:13.715 D 47: [emsesp] Last Tx read successful
000+00:00:13.721 D 48: [emsesp] Updating details on already existing device ID 0x08
000+00:00:14.043 D 49: [telegram] Sending read Tx [#5], telegram: 0B 89 02 00 20 B4
000+00:00:14.084 D 50: [emsesp] Last Tx read successful
000+00:00:14.089 D 51: [emsesp] Updating details on already existing device ID 0x09
000+00:00:14.342 D 52: [telegram] Sending read Tx [#6], telegram: 0B 98 02 00 20 3C
000+00:00:14.393 D 53: [emsesp] Last Tx read successful
000+00:00:14.398 D 54: [emsesp] Updating details on already existing device ID 0x18
000+00:00:14.791 D 55: [telegram] Sending read Tx [#7], telegram: 0B 88 19 00 20 D0
000+00:00:14.882 D 56: [emsesp] Last Tx read successful
000+00:00:14.886 D 57: [emsesp] Received UBAMonitorSlow
000+00:00:15.140 D 58: [telegram] Sending read Tx [#8], telegram: 0B 88 33 00 20 78
000+00:00:15.216 D 59: [emsesp] Last Tx read successful
000+00:00:15.220 D 60: [emsesp] Received UBAParameterWW
000+00:00:15.563 D 61: [telegram] Sending read Tx [#9], telegram: 0B 88 16 00 20 EC
000+00:00:15.655 D 62: [emsesp] Last Tx read successful
000+00:00:15.659 D 63: [emsesp] Received UBAParameters
000+00:00:15.937 D 64: [telegram] Sending read Tx [#10], telegram: 0B 98 0A 00 20 1C

Sonusss, 2020-09-02:
Having made several tests with telnet and "log all" it looks that the system keeps on rebooting...
I cannot get log timestamp bigger than 18 seconds

proddy, 2020-09-02:
Strange. What hardware are you using? Is it a home-grown circuit? Is the ESP8266 a clone? Does the console work when its only plugged into the PC/laptop?

Sonusss, 2020-09-02:
It is 8266 based
Wemos D1 mini V3.0

Sonusss, 2020-09-02:
Definitely the problem comes from rebooting and the exact moment I refresh WEB UI from time to time it reports the right devices.
Depending on the uptime it may have scanned the bus or not...

Sonusss, 2020-09-02:
Regarding serial console cannot get anything except the following just after pressing reset button:
[11:33:00:202] sd<break>
[11:33:00:202] d<0x9c><0xdf>|<break>
[11:33:00:202] <0x8c>d<0xe0>|<0x03><0x04><0x04><0x0c><0x84><0x04>l<0xe4><0x04>c|<0x8e><0x83><0x03><0xe4><0x1b><0x93>{<0xdb>c<0x84><0x0c>c<0x84><0xfb>'o<0x9f>dog<0x9c><0xe3><0xec><0x0c>#<0x1c>x<0xec><0x8f>${$sdx<0xf3>o<0xe0><0x18><0x03><0x04><0x0c><0x83><0x04>l<0x04><0xc4><0x9c><0x04><0x04><0x0c>c<0x04>o<0xe3><<0x03><0xec>l<0x8f><0x0c><0xc7>b<0x8c><0xf3>no<0xef><break>
[11:33:00:230] d<0x8c><0xc7>d`<0x02><0x98><0x1b>g'<0x04>l <0x03><0x07><0x03>n{<0x87><0x93><0x9b>g<0x04><0x04>c<0x03>l`<0x03><0x0f>;<0x93><0x9b>o<0x04><0x04>b<0x03>$`<0x03><0x9c>c<0x0c><0x04><0x8f><0x1c><0x03><0x93>ls<0x9b>`<0x02><0xfc><0x82>'<0x9c><0x03>

proddy, 2020-09-02:
Not sure what the problem is. I haven't seen it before. You can try running the ESP plugged into the USB with the EMS bus attached. Note in v2 there is no Serial console anymore, just Telnet. If more people experience the same issue I'll look into further.

As for this issue (TC100 not showing) I'm working on it.

Sonusss, 2020-09-02:
If I put Tx mode to 0 the system is stable with V2.0.1b1 (not rebooting constantly)
I trid to compile with the uart file you sent but I got a bin file too big 9MB instead of 4MB ?!

proddy, 2020-09-02:
wow, so Tx mode is causing the crashes. not good at all! I reverted the UART changes so try with yesterday's build https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build

I also found a fix for the TC100

Sonusss, 2020-09-02:
I loaded the genuine travis-dev-build but still the same.
I first booted without EMS bus connected and settled Tx mode to 0
Connected the EMS bus and still fine - syste is stable
There is one strange thing:
   V1.9.5 reports CPU freq as 80MHz while V2.0.1b1 reports 160MHz

I then enabled logging to rsyslog (network logging server) hoping to get more debug info but il looks the same, see below.
You can easily see that it reboots constantly but the timing is quite variable, uptime is never longer than 18 seconds

Sep  2 16:17:23 ems-esp - ﻿000+00:00:07.619 D 50: [telegram] Tx read request to device 0x18 for type ID 0x0A
Sep  2 16:17:23 ems-esp - ﻿000+00:00:07.619 D 51: [emsesp] Adding new device with device ID 0x18 with product ID 202 and version 02.21
Sep  2 16:17:23 ems-esp - ﻿000+00:00:07.619 D 52: [emsesp] Fetching values for device ID 0x18
Sep  2 16:17:36 ems-esp - ﻿000+00:00:00.000 I 0: [syslog] Log level set to DEBUG
Sep  2 16:17:36 ems-esp - ﻿000+00:00:00.000 D 1: [telegram] Tx read request to device 0x08 for type ID 0x07
Sep  2 16:17:36 ems-esp - ﻿000+00:00:00.000 I 2: [emsesp] EMS Device library loaded with 70 records
Sep  2 16:17:36 ems-esp - ﻿000+00:00:04.746 D 3: [emsesp] WiFi Connected with IP=192.168.1.79, hostname=ems-esp
Sep  2 16:17:38 ems-esp - ﻿000+00:00:07.735 D 4: [emsesp] No telegram type handler found for ID 0x18 (src 0x08)
Sep  2 16:17:38 ems-esp - ﻿000+00:00:08.090 D 5: [emsesp] No telegram type handler found for ID 0x2A (src 0x08)
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.848 D 6: [telegram] Sending read Tx [#0], telegram: 0B 88 07 00 20 A8
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.899 D 7: [emsesp] Last Tx read successful
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.907 D 8: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.907 D 9: [telegram] Tx read request to device 0x08 for type ID 0x02
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.907 D 10: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.907 D 11: [telegram] Tx read request to device 0x09 for type ID 0x02
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.907 D 12: [emsesp] New EMS device detected with ID 0x18. Requesting version information.
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.907 D 13: [telegram] Tx read request to device 0x18 for type ID 0x02
Sep  2 16:17:41 ems-esp - ﻿000+00:00:11.998 D 14: [telegram] Sending read Tx [#1], telegram: 0B 88 02 00 20 BC
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.062 D 15: [emsesp] Last Tx read successful
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 16: [boiler] Adding new Boiler with device ID 0x08
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 17: [mqtt] Registering MQTT cmd comfort with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 18: [mqtt] Registering MQTT cmd wwactivated with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 19: [mqtt] Registering MQTT cmd wwtapactivated with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 20: [mqtt] Registering MQTT cmd wwonetime with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 21: [mqtt] Registering MQTT cmd wwcirculation with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 22: [mqtt] Registering MQTT cmd flowtemp with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 23: [mqtt] Registering MQTT cmd wwtemp with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 24: [mqtt] Registering MQTT cmd burnmaxpower with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 25: [mqtt] Registering MQTT cmd burnminpower with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 26: [mqtt] Registering MQTT cmd boilhyston with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 27: [mqtt] Registering MQTT cmd boilhystoff with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 28: [mqtt] Registering MQTT cmd burnperiod with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 29: [mqtt] Registering MQTT cmd pumpdelay with topic boiler_cmd
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 30: [emsesp] Adding new device with device ID 0x08 with product ID 123 and version 06.08
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 31: [emsesp] Fetching values for device ID 0x08
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 32: [telegram] Tx read request to device 0x08 for type ID 0x19
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 33: [telegram] Tx read request to device 0x08 for type ID 0x33
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.067 D 34: [telegram] Tx read request to device 0x08 for type ID 0x16
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.101 D 35: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.101 D 36: [telegram] Tx read request to device 0x09 for type ID 0x02
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.101 D 37: [emsesp] New EMS device detected with ID 0x18. Requesting version information.
Sep  2 16:17:41 ems-esp - ﻿000+00:00:12.101 D 38: [telegram] Tx read request to device 0x18 for type ID 0x02
Sep  2 16:17:43 ems-esp - ﻿000+00:00:12.496 D 39: [telegram] Sending read Tx [#2], telegram: 0B 89 02 00 20 B4
Sep  2 16:17:43 ems-esp - ﻿000+00:00:12.536 D 40: [emsesp] Last Tx read successful
Sep  2 16:17:43 ems-esp - ﻿000+00:00:12.543 D 41: [emsesp] Adding new device with device ID 0x09 with product ID 125 and version 03.03
[... cut ...]
Sep  2 16:18:52 ems-esp - ﻿000+00:00:00.000 D 1: [telegram] Tx read request to device 0x08 for type ID 0x07
Sep  2 16:18:52 ems-esp - ﻿000+00:00:00.000 I 2: [emsesp] EMS Device library loaded with 70 records
Sep  2 16:18:52 ems-esp - ﻿000+00:00:01.883 D 3: [emsesp] No telegram type handler found for ID 0x18 (src 0x08)
Sep  2 16:18:52 ems-esp - ﻿000+00:00:02.086 D 4: [emsesp] No telegram type handler found for ID 0x2A (src 0x08)
Sep  2 16:18:52 ems-esp - ﻿000+00:00:04.742 D 5: [emsesp] WiFi Connected with IP=192.168.1.79, hostname=ems-esp

proddy, 2020-09-02:
Ok, this is most definitely a bug in the code then. Something in the incoming telegrams is causing the crash because when you set Tx Mode to 0 everything works. I suspect its in the Thermostat code since you said it worked with 2.0.0 (no crashes) but only you were not receiving any thermostat data. I'll look into the code again and try and think of what we can do.

MichaelDvP, 2020-09-02:
@proddy is it possible that the empty `set_typeids = {};` for the TC100 causes this reboot in that moment the thermostat is registered and tries to read the type_ids?

proddy, 2020-09-02:
@MichaelDvP could be. Thing is it works when I run in standalone (i.e. run on Windows or Ubuntu using `make run` and then running one of the tests. I'll need to simulate this on a real ESP8266

proddy, 2020-09-02:
@MichaelDvP you were right (again). Making a fix now

MichaelDvP, 2020-09-02:
@proddy a possible fix is:
`set_typeids = {0x00};` 
and add in `EMSESP::send_read_request(` and `EMSESP::send_write_request(`  
```
    if (type_id == 0) {
        return;
    }
```

proddy, 2020-09-02:
I took a different approach, the error was when creating a heatingcircuit object

Sonusss, 2020-09-02:
HeHeHe,

Looks like my issue is fixed in V2.0.1b2
Will give a try overnight...

I would like to be more effective in bug reporting, please give some clues...

Thanks you all !

proddy, 2020-09-02:
for reporting issues, its always good to state which hardware you're using (esp8266/esp32, bus/usb-powered, which circuit) and also steps to reproduce the error. A copy of `system/show` and `show` from Telnet is handy too.

In your particular case it was a bad bug that caused a crash so thanks for reporting it and thanks for being patient with us.

by the way in version 2.0 we over-clock the ESP8266 from 80 to 160Mhz (well spotted!). It speeds things up ever so slightly but does have some consequences like wear&tear of the chip, heat consumption etc. It'll run fine at 80Mhz too. Maybe I'll change it back.

Sonusss, 2020-09-03:
Working like a charm.

Thx you all

closed 2020-09-03 by Sonusss