emsesp/EMS-ESP #475: v2 boiler_data payload

PhillyGilly, 2020-09-03:
@proddy Great work on the new v2 interface. I have been manually migrating my sensor.yaml file and have compared the payload as shown in MQTT explorer with the 1.9.5 payload and with the new telnet "show values" output . Apart from a few unmatched values as shown in the excel, I can't find UBA uptime (UBAuptime). Can you bring this back again?
[boiler_data.xlsx](https://github.com/proddy/EMS-ESP/files/5168460/boiler_data.xlsx)

MichaelDvP, 2020-09-03:
@proddy the bug is in `telegram.h` `read_value` in line 104: it should be `> message_length - 1` or `>= message_length`

@PhillyGilly Is your solar  module working now?

MichaelDvP, 2020-09-03:
@PhillyGilly some of the unmatched values:
-   "wWCiPuMode" is telnet line 6
- "wWHeat" is telnet line 18
-  "serviceCode" is also telnet line 29, telnet shows `serviceCode (CodeNumber)` 

Just an idea: Would it help to list the mqtt payload in terminal, i.e.
`Warm Water circulation pump freq (wWCiPuMode): 0x3min`
or
`Warm Water circulation pump freq: wWCiPuMode = 0x3min`

PhillyGilly, 2020-09-03:
Hello @MichaelDvP. Yes the Solar Module is working fine. It was changes in the payload descriptions from bottomTemp to tankBottomTemp that made it look like it wasn't publishing. Also splitting the MM200 into two modules produced a bit of payload fun too. But that's sorted now.

proddy, 2020-09-03:
@MichaelDvP thanks for spotting that. I'll test tonight and include it in the test code. I like the idea of including the mqtt name. If the ESP8266 had more memory I would have loved to put all this into a std::map so its defined once and dynamically rendered for Telnet and MQTT.

PhillyGilly, 2020-09-03:
> @PhillyGilly some of the unmatched values:
>
> * "wWCiPuMode" is telnet line 6
> * "wWHeat" is telnet line 18
> * "serviceCode" is also telnet line 29, telnet shows `serviceCode (CodeNumber)`
>
> Just an [...]

I'm still not getting anything on my Service Code entities.
MQTT Explorer shows "serviceCode": "",  "serviceCodeNumber": 203 and Telnet shows only  System service code:  (203) 
In HA I've got nothing in my Logbook for either sensor.service_code or service.sensor_code_number.
YAML  I've got:
#45 sensor.service_code
- platform: mqtt
  state_topic: 'ems-esp/boiler_data'
  name: 'Service Code'
  value_template: '{{ value_json.ServiceCode }}'
  qos: 1
  payload_available: "online"
  payload_not_available: "offline"
#46 sensor.service_code_number
- platform: mqtt
  state_topic: 'ems-esp/boiler_data'
  name: 'Service Code Number'
  unit_of_measurement: ''
  icon: 'mdi:power-cycle'
  value_template: "{% if value_json.ServiceCodeNumber == 200 %} CH active 
    {% elif value_json.ServiceCodeNumber == 201 %} HW active
    {% elif value_json.ServiceCodeNumber == 202 %} CH anti cycle
    {% elif value_json.ServiceCodeNumber == 203 %} Stand by
    {% elif value_json.ServiceCodeNumber == 204 %} CH cooling
    {% elif value_json.ServiceCodeNumber == 208 %} Service test
    {% elif value_json.ServiceCodeNumber == 265 %} Low CH load
    {% elif value_json.ServiceCodeNumber == 268 %} Component test
    {% elif value_json.ServiceCodeNumber == 270 %} Power up
    {% elif value_json.ServiceCodeNumber == 283 %} Burner start
    {% elif value_json.ServiceCodeNumber == 284 %} Ignition
    {% elif value_json.ServiceCodeNumber == 305 %} HW anti cycle
    {% elif value_json.ServiceCodeNumber == 357 %} Air purge
    {% elif value_json.ServiceCodeNumber == 358 %} Valve kick
    {% else %} {{ value_json.ServiceCodeNumber }} {% endif %}"
  qos: 1
  payload_available: "online"
  payload_not_available: "offline"

So I would expect to be shown as "Stand by"
Any ideas?

proddy, 2020-09-03:
> @proddy the bug is in `telegram.h` `read_value` in line 104: it should be `> message_length - 1` or `>= message_length`
> 
> @PhillyGilly Is your solar module working now?

fixed. thanks again Michael. well spotted. Funny it works in the test under Windows but thats probably how the int32 size is calculated

IGastel, 2020-09-03:
In version 2, the mqtt payload is changed for service code and service code
number. It is now spelled with a lower case s. Previous it was spelled with
a capital S. So the home assistant sensor needs to be adjusted

proddy, 2020-09-04:
> In version 2, the mqtt payload is changed for service code and service code number. It is now spelled with a lower case s. Previous it was spelled with a capital S. So the home assistant sensor needs to be adjusted

that's true. It's now "serviceCode" and "serviceCodeNumber" to be consistent with the other settings. I still need to write the documentation explaining all the MQTT changes and update the wiki. @IGastel we're doing some changes to make integration with Home Assistant easier, see https://github.com/proddy/EMS-ESP/issues/288

PhillyGilly, 2020-09-04:
Thanks. I just couldn't see the wood for the trees! Everything OK now apart from UBA uptime. Is that in the latest release?

closed 2020-09-04 by PhillyGilly