emsesp/EMS-ESP #476: Send mqtt not only in intervals, but also event based

hamster65, 2020-09-03:
Hi,
would it be possible to send out mqtt messages for every change of serviceCode and serviceCodeNumer? I'd like to log and monitor the service codes and create mail alarms through node red when something is wrong.

Since ems-esp uses a fixed interval for the mqtt messages, I guess that I will miss many intermittent codes, so an event based approach would be better for this.
Thanks!
hamster

proddy, 2020-09-03:
would being able to change the frequency only for the MQTT boiler publishes be enough, say every few seconds?

We could add the code but its very specific to your needs. I'd rather find a more elegant solution. Perhaps putting a watch on certain values and sending an event change notification via MQTT.

hamster65, 2020-09-03:
Hi Proddy,
I used ems collectord with the php web interface before, and it provided this view:
https://github.com/moosy/ems-php-webinterface/blob/master/screenshots/webif_logs.png
It helped me once to track down an intermittent problem we had with the boiler, so now I am trying to set up something similar with ems-esp, mqtt, note-red and mail notification.

One second would probably enough to catch all codes. I'll test that over the winter :)
I just thought an event driven approach would be better in terms of resources.

Greetings,
hamster

proddy, 2020-09-03:
There was a similar request on Gitter today from PetrDa. How about if we build an MQTT command that is a request to fetch and return the device data? Perhaps even for a specific device value. For example sending a message to a topic `ems-esp/boiler_cmd` with payload like {"cmd": "fetch": "data":"servicecode"} would send the service code data back in a new `boiler_data` topic. You can then build in a scheduled event to request it every few seconds.

hamster65, 2020-09-04:
That would not work. I'd have to poll the two values then very fast to not miss any changes, every every 100ms or so....

proddy, 2020-09-05:
Ok. This needs some more requirements and some more use cases, as its a very specific need which requires a lot of code changes. How do you put a trigger on a device value? What does the event do? How often? How many events are allowed? etc etc

Since it's open-source software I'd also be happy if someone experiments coding this in a feature branch too.

MichaelDvP, 2020-09-05:
I'm testing now if it is possible to publish on change or if it is flooding mqtt. 
My approach is to make `read_value` bool, returning true if the value has changed and call the function as `changed_ |= telegram->read_value(`
In `updated_values()` the flag is cleared and triggers the mqtt-publish. 
First test is positive: 
```
ems:/$ 000+00:03:02.878 D 146: [emsesp] Received MMStatusMessage
ems:/$ 000+00:03:07.044 D 147: [emsesp] Received RCOutdoorTemp
ems:/$ 000+00:03:07.555 D 148: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:07.834 D 149: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:08.240 D 150: [emsesp] Received UBAMonitorWW
ems:/$ 000+00:03:13.732 D 151: [mqtt] Publishing topic ems/heartbeat (#40, attempt #1, pid 41)
ems:/$ 000+00:03:17.636 D 152: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:17.918 D 153: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:18.324 D 154: [emsesp] Received UBAMonitorWW
ems:/$ 000+00:03:18.634 D 155: [emsesp] Received UBAMonitorSlow
ems:/$ 000+00:03:18.759 D 156: [mqtt] Publishing topic ems/boiler_data (#41, attempt #1, pid 42)
ems:/$ 000+00:03:19.009 D 157: [emsesp] Received UBATotalUptime
ems:/$ 000+00:03:19.162 D 158: [mqtt] Publishing topic ems/boiler_data (#42, attempt #1, pid 43)
ems:/$ 000+00:03:27.620 D 159: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:27.901 D 160: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:28.307 D 161: [emsesp] Received UBAMonitorWW
ems:/$ 000+00:03:32.336 D 162: [emsesp] Received MMStatusMessage
ems:/$ 000+00:03:32.778 D 163: [emsesp] Received RCTime
ems:/$ 000+00:03:32.872 D 164: [mqtt] Publishing topic ems/thermostat_data (#43, attempt #1, pid 44)
ems:/$ 000+00:03:33.073 D 165: [mqtt] Publishing topic ems/thermostat_data2 (#44, attempt #1, pid 45)
ems:/$ 000+00:03:37.704 D 166: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:37.983 D 167: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:38.390 D 168: [emsesp] Received UBAMonitorWW
ems:/$ 000+00:03:47.587 D 169: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:47.867 D 170: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:48.274 D 171: [emsesp] Received UBAMonitorWW
ems:/$ 000+00:03:52.297 D 172: [emsesp] Received RC35Monitor
ems:/$ 000+00:03:57.571 D 173: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:57.851 D 174: [emsesp] Received UBAMonitorFast
ems:/$ 000+00:03:58.257 D 175: [emsesp] Received UBAMonitorWW
```
But the boiler is in summermode now and the values do not change rapid.
I added a checkbox to mqtt-settings `Publish on change` to enable it apart from publish_time.

proddy, 2020-09-05:
great! Back in 1.9 it used to work this way too. I would create a hash from the payload and use it see if any of the values had changed (when publish_time was set to 0 for automatic). But then when all the MQTT issues started appearing I went back to schedule based publishes.

MichaelDvP, 2020-09-05:
On the way, i also added #408 to make everyone happy.
It's a bit more combinations to test, but looks good so far.
Or should we got with setting 0 for automatic? Then it's possible to set one device to automatic and another to timed. But it's not possible to have automatic and redundancy by large publish time. 

[image]

proddy, 2020-09-05:
wow, you've been busy! That would defintely make a few people very happy indeed.

I would go for 0=Automatic and rename a few things 
- Publish Time = Publish Frequency
- Automatic = On Change or Or Update

MichaelDvP, 2020-09-09:
@hamster65 have you tested the v2.0.1b4? Is it ok? Can we close this issue?

hamster65, 2020-09-09:
I flashed just now., looks fine so far, boiler_data is updated on every change, which is quiet often because values jump around for 0.1°C.  What happens when nothing changes? Will there be no update at all? Would it be better to have 2 separate settings, one for an update interval in seconds, an another to update on change in addition? 
I'll reload warm water later and see how it behaves.

MichaelDvP, 2020-09-09:
Separate settings was also my first approach, but proddy votes for 0=on change, and, after thinking about it, i know he's right. 
Thermostat has a clock changing every minute, boiler updates `UBAuptime` every minute, solar counts working minutes and energy when working, mixer changes valve position very fast when working, only in standby you will get no messages from solar/mixing, but than really nothing happens with these modules.

hamster65, 2020-09-10:
I flashed yesterday. Looks fine so far, boiler_data is updated on every change, which is quiet often because values jump around for 0.1°C.  
What happens when nothing changes? Will there be no update at all? Would it be better to have 2 separate settings, one for an update interval in seconds, an another to update on change in addition? 

I got 5 changes of the status code over night when warm water was reheated and it looks as if every update was recorded :)

proddy, 2020-09-12:
fantastic that it works. Shall we close this? Then I can push 2.0.1 out as the next stable main version

proddy, 2020-09-13:
Closing as this is going into the main 2.0.1. We can reopen if new issues appear

closed 2020-09-13 by proddy

proddy, 2020-09-17:
@MichaelDvP just an explanation of some Web UI changes I made earlier this week. I removed the "Settings" from the headers as there was just much repetition of the word Settings in the page (8 times). I also renamed 'Publish Period' too 'Publish Interval' as I think its more appropriate. "Period" has lots of different annotations in the English language ;-)