emsesp/EMS-ESP #479: mixing_data payload

PhillyGilly, 2020-09-04:
My MM200 is configured to manage two hot water circuits wW1 and wW2.
mixing_data9 = {"type":"wwc","flowTemp":52.1,"pumpMod":6,"status":1}
mixing_data10 = {"type":"wwc","flowTemp":55,"pumpMod":0,"status":0}
This is pretty much where we were at the start of closed issue #299 which @MichaelDvP helped with.
Can we revisit split between hc1/hc2 an ww1/ww2?

proddy, 2020-09-04:
Not sure I follow. What is broken exactly? Is the data shown in Telnet and Web correct? @MichaelDvP knows more about the mixing units that I do.

MichaelDvP, 2020-09-04:
I see, data are correct, but the names are not adjusted. I'll do a PR.

proddy, 2020-09-12:
@PhillyGilly all good now?

PhillyGilly, 2020-09-12:
> @PhillyGilly all good now?

Yes. Thanks. As far as I can tell right now.

closed 2020-09-12 by PhillyGilly