emsesp/EMS-ESP #485: Point name changes

glitter-ball, 2020-09-09:
Took the plunge and upgraded to 2.0.1b4.

Fixed the MQTT base but found some points still broken.

Is MQTT point name case sensitive?

In 1.9.5 I had ServiceCodeNumbet work fine in HA. Now it’s broken. MQTT Explorer now shows it with lower case initial ‘s’ - serviceCodeNumber. Would this change be enough to break it or do I need to look elsewhere? Thanks.

proddy, 2020-09-09:
the names in the MQTT payload are case-sensitive and a few have been changed, either to conform to the standard (starting with lower-case) or fixing spelling mistakes.

What is broken exactly?

glitter-ball, 2020-09-09:
I’ll get a list together
s/ServiceCodeNumber was the first one I found and thought I’d better check after a lot of head scratching.

proddy, 2020-09-09:
in MQTT only the servicecode ones and 'desinfection' have changed in the boiler_data payload. In the heartbeat it also has some new tokens. The rest should be the same. I will get around to documenting the changes at some point.

MichaelDvP, 2020-09-09:
Today also in mixing `pumpMod`changed to `pumpStatus`. 
And boiler, solar, mixing uses camelCase, thermostat payload is lower case. To make i consistent maybe some more changes will come.

proddy, 2020-09-09:
Right, let's create a section in the README where we can capture this. To be honest I'm loosing track of all the changes too!

glitter-ball, 2020-09-09:
OK - have managed to fix most stuff. Self-inflicted because I haven't kept things up to date with changes, probably! A brief summary of what I found...

**Pump Status**
Was using heatPmp. Now changed to heatPump. Working.

**Return Temp**
Was using sensor1.temp. Now changed to sensor_28-BE2A-7791-0402-65/temp. Working.

**Boiler Status**
Was using ServiceCodeNumber. Now using serviceCodeNumber. Working.

**Firmware Version**
No longer published - deleted.

**Room Temperature, Set Temperature, Mode - sensor.yaml and climate.yaml**
Was using ems-esp/thermostat_data/.. Convert to ems-esp/thermostat_data1/.. Working.

Anyway, enough working again for now until I have a complete revamp. The firmware upgrade was a white knuckle ride! I read bbqkees page too late, by which time I'd used the web firmware update page and bricked the ESP8266 by loading ESP32 firmware! Fortunately, I managed to get esptool up and running and load the right firmware with USB. At which point it migrated the 1.9.5 settings and came back to life. Phew...

[HA.pdf](https://github.com/proddy/EMS-ESP/files/5197538/HA.pdf)

proddy, 2020-09-12:
thanks @glitter-ball for collecting this. I will include in the MQTT documentation. A few quick observations:
* the topic format of the external dallas temperatures can be changed by adjusting the MQTT format
* the firmware is published in the `start` topic, e.g. `{"version":"1.9.5","IP":"10.10.10.140"}`
* ems-esp/thermostat_data is the correct topic name

You may be interested in https://github.com/proddy/EMS-ESP/issues/288 which when I get around to it will automate the EMS-ESP with HA so you don't have to worry about renaming aanything. Part of that would be creating a custom component/card to display the data within home assistant.

closed 2020-09-12 by proddy