emsesp/EMS-ESP #488: Telnet command to restart

glitter-ball, 2020-09-09:
**Question**
Is there a text mode reboot/restart command under 2.0.1b4? Tried obvious things but didn't work. Would be handy when the web interface struggles. I tried to su with the password quoted in the upgrade notes but it didn't like that, alas. Thanks..

glitter-ball, 2020-09-09:
OK. Found the restart command in the full command listing further down the New Features page.

But... the password `ems-esp-neo` doesn't work when I try and `su`.
```

┌──────────────────────────────────────────┐
│ EMS-ESP version 2.0.1b4                  │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ su
Password:
su: incorrect password
000+00:10:37.356 N 0: [shell] Invalid admin password on console
ems-esp:/$
```

proddy, 2020-09-09:
@glitter-ball if you upgraded from v1.9.5 this would have been the admin password.

glitter-ball, 2020-09-10:
OK. Got it. All good. Thanks.

closed 2020-09-10 by glitter-ball