emsesp/EMS-ESP #490: Dallas sensors not working

JewelZB, 2020-09-10:
In the latest 2.0.1 development build the dallas sensors seem to be not functioning.
The web interface states "No connected Dallas temperature sensors detected" and no MQTT data is observed for the sensors.

I have two sensors attached in parasite mode that worked fine in the 1.9.5 build.
In the syslog I  see "[sensors] Bus reset failed" messages.

glitter-ball, 2020-09-10:
I have one sensor in 3-wire mode and it worked fine when I upgraded from 1.9.5 to 2.0.1b4. The MQTT payload changed so I had to tweak definition in HA, but that's not relevant to your issue.

Is the Dallas GPIO pin setting correct (14 on my system)? That stopped my sensor working when I first installed it.

JewelZB, 2020-09-10:
Yes the sensors are connected to pin 14, and the setting is on pin 14.
Rolling back the firmware to 1.9.5 fixes the problem for now.

MichaelDvP, 2020-09-11:
Dallas parasite-mode has to be acivated, but this setting is missing in the actual build.

proddy, 2020-09-11:
we need to add parasite back then?

JewelZB, 2020-09-11:
I checked the code and Parasite mode seems to be missing altogether from the new sensors.cpp.
Yes it would be great to have parasite mode back in 2.0.
I already tried adding it myself but ran into some problems with the way sensors.cpp communicates with the sensors (seems to be designed to support hotplugging?) so need to dig further.

MichaelDvP, 2020-09-11:
I checked how parasite works and the 1.9.5 code, it is not a big deal in the software. I can add the parasite setting, but can not test, i have not such a sensor. @proddy should i make a pr?

MichaelDvP, 2020-09-11:
@JewelZB if you want to try and compile your own, i think all you have to do is in `sensor.cpp`
Line 70: `bus_.write(CMD_CONVERT_TEMP, 1);`
Line 80: `if (time_now - last_activity_ > CONVERSION_MS) {`
(skip the complete-check, it does not work in parasite mode)
than this is fixed setting to parasite on.

JewelZB, 2020-09-11:
@MichaelDvP , thanks for the quick reply! I hope i can find a moment to try it out this weekend!

proddy, 2020-09-11:
I have a few parasite DS18b20's lying around so will also test this weekend

commit fb05c139 (2020-09-11): add dallas parasite setting (fix #490), sort settings page
commit fead546d (2020-09-11): Merge pull request #491 from MichaelDvP/dev

add dallas parasite setting (fix #490), sort settings page
closed 2020-09-13 by proddy