emsesp/EMS-ESP #492: Firmware 2.0.1b4 does not persist data, does not behave as documented for wi-fi setup and crashes

stevedcc, 2020-09-11:
Today I set up my Gateway, which arrived yesterday with 2.0.1b4 already flashed.  I have noticed the following issues

1. Both wi-fi setup and TX-Mode Configuration are not persisted through restart
2. THe wifi-setup connects immediately after providing the information, which does not reflect the documentation to restart in the wiki
3. The gateway crashes within a few minutes of running.

proddy, 2020-09-12:
Thanks for reporting. Have you reached out to BBQKees too as I know he thoroughly tests all gateways before shipping?

To get your EMS-ESP stable I suggest removed the ESP8266 Wemos board, connecting it to a PC via USB, erasing the flash and uploading the latest firmware from the repo. The details are in the README.md file. Unfortunately, I haven't had time to update my Wiki yet.

proddy, 2020-09-12:
tested 1, 2 and 3 and cannot reproduce. I would need more detailed information and steps on how to reproduce. If you can go back to 1.9.5 and recreate the settings (also export both of them as backups to share) and see if it happens again. 

Also which WiKi are you referring too?

stevedcc, 2020-09-12:
I guess I saw dev/beta firmware and jumped the gun. Probably my mistake for posting in the wrong place.

I was referring to the install instructions for the gateway, so yes, not this forum.

Tbh, I suspect this could be a power issue: as far as I remember from my days as an embedded developer failing to persist flash/eeprom is often one of the first signs of not enough power and there is already a controller and a thermostat on the bus. I’ll try supplying external usb power.

[image]

twaymouth, 2020-09-13:
Thought I should share that I am experiencing a variant of the same issue. I am not experiencing any instability however settings are not persisted through a reboot (wifi, MQTT ect). I am using a wemos D1 mini. As an aside the bin file for the stable 2.0 release appears to have been removed?

proddy, 2020-09-13:
@twaymouth which version of the firmware did you use? Did you upgrade from 1.9? If you flash with the latest 2.0.1 does the same thing happen? I'd like to get to the bottom of this as both BBQkees and myself tested the upgrades thoroughly.

I removed the firmware bin file from the 2.0.0 release in GitHub as it was a litle buggy when first released. Take the latest dev branch https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build

When the dallas parasite mode has been tested I'll push 2.0.1 to the main branch and make it the current stable version.

twaymouth, 2020-09-13:
I used firmware 2.0.1b5, I did not upgrade from 1.9 rather erased the flash then loaded 2.0.1b5. 

As a test I loaded 1.9, configured wifi, confirmed it persisted as expected through a reboot then loaded 2.0.1b5 without erasing the flash. After doing so settings persist as expected.

I then erased the flash completely and loaded 2.0.1b5 from scratch which again caused settings not to persist.

Out of interest I noticed the file system used  / total varies greatly when upgrading vs erasing and flashing 2.0.1b5. See attached screenshots of an upgrade / erase the flash. Not sure if this is related to the settings not persisting?

Hopefully this all makes sense.

[image]

[image]

MichaelDvP, 2020-09-13:
> Hopefully this all makes sense.

yes, thank you, it seems that converting from 1.9.5 works, but the filesystem on clean chip is not formated correctly. 
@proddy I think it's in `system.cpp` line 655 after `if (!SPIFFS.begin()) {` has to be something like
```
        l_cfg.setAutoFormat(true);
        LittleFS.setConfig(l_cfg); // auto format to LittleFS
        LittleFS.begin();
```
but i have not tested this yet.

stevedcc, 2020-09-13:
Would explain why my brand new gateway with 2.01b4 did not persist.

Steven Crawford
Laubestr. 17, 60594 Frankfurt am Main
+49 (0) 69 680 99480

proddy, 2020-09-13:
just checked this and confirm its truly broken. On a clean flashed ESP8266 it will keep formatting the LittleFS filesystem on restart. I'm working on a quick fix.

proddy, 2020-09-14:
this should be fixed now. If you're still getting reboots let me know

closed 2020-09-14 by proddy

stevedcc, 2020-09-14:
Sorry it took a while to try and install the updated firmware.  Family, kids.  Everything seems to be working fine now, data is persisted and I have 15 minutes uptime, which was impossible before.  Thanks for fixing it.

twaymouth, 2020-09-15:
Fixed, thanks for your prompt attention.