emsesp/EMS-ESP #493: V2.0 - MQTT- how to set thermostat?

Gert05, 2020-09-11:
Good evening. I have upgraded to v2.0. It's looking great - thanks for taking the effort to further develop this software.

I'm trying to adapt my Domoticz plugin to the new MQTT structure. I have succeeded in reading the values that I need (mostly temp, pressure), but I'm stuggling to find the right command for setting the thermostat temperature.

I have a Nefit boiler (0x08/123/06.07), a BC10 controller (0x09/190/01.04) and a Moduline 400 thermostat (0x10/78/02.20).

Based on the readme.md comments I came up with the following payload in ems-esp/thermostat_cmd:
     "cmd":"temp", "data":19.5, "hc":1
This does not seem to work.

Could you give an example of the topic/syntax that ems-esp would expect to set the temperature?

Thanks.

proddy, 2020-09-12:
could you do a quick test from the console?

* `ems-esp:/$ show mqtt` and see if the temp command is registered, e.g. ` topic: ems-esp/thermostat_cmd, [cmd]: wwmode temp mode`
* `ems-esp:/$ thermostat` to go to the submenu context
* `ems-esp:/thermostat$ su` to go into admin mode
* `ems-esp:/thermostat# log all` to set the logging to show debug statements. Watch out for any ERR/INFO reports here.
* `ems-esp:/thermostat# call temp 19.5` to set the thermostat temp
* `ems-esp:/thermostat# show` to see if it has been set. It may take a few seconds.

Gert05, 2020-09-12:
That works just fine. ems-esp/thermostat_cmd is registered. Bit of a problem with the `su` password though: I set a new admin password through the web interface but `su` still requires `admin` as a password instead of the password that I set.

I'm on v2.0.1b4 by the way.

proddy, 2020-09-12:
oh, the admin thing may be a bug. I'll look into that. So now leave the console open, with logging enabled, go to MQTTExplorer and send the same topic & payload `ems-esp/thermostat_cmd={"cmd":"temp", "data":19.5, "hc":1`}

I just ran a unit test and it worked (in the standlone version) but haven't tested on a live system.

Did you miss the {}'s perhaps?

Gert05, 2020-09-12:
The {}'s did the trick. Thanks for your help.

I have now got a basic Domoticz plugin working (seltemp/currtemp/syspress + changing thermostat temp). I'm willing to share the plugin if anyone's interested.

proddy, 2020-09-12:
cool! Yeah, please share. Perhaps in a new issue and then I'll add it to the repo.

proddy, 2020-09-12:
> set a new admin password through the web interface but `su` still requires `admin` as a password instead of the password that I set.

The password is the JWT Secret in the WebUI. I just tested this and it works. I didn't use the password of the 'admin' user since this username can be removed or deleted.

[image]

Gert05, 2020-09-12:
Understood. Thanks for clarifying.

closed 2020-09-12 by proddy

bbqkees, 2020-09-29:
I already have a fully working plugin for Domoticz.
https://github.com/bbqkees/ems-esp-domoticz-plugin
I only need to change the MQTT topics for the new 2.0.1 API firmware which I will do when I'm back from my holiday.