emsesp/EMS-ESP #495: Worcester Bosch 8000 (UK) - Central Heating Temperature

fiskn, 2020-09-11:
Hi,

I've got a Worcester Bosch 8000 which seems to looks like it has the same control panel as a Condens 5000 and shows up as such in EMS-ESP. There is no EMS thermostat connected, just a boiler on it's own currently controlled via the legacy on/off connections. I've got EMS-ESP (v2.0) hooked up and getting some values back which is awesome, although things like return temperature and heating temperature are missing. I'm not sure if these boilers are just new and there is still ongoing work to decipher the protocol. Happy to supply logs if they would be useful to find new messages/variables.

I'm also trying to work out how I can set the central heating flow temperature (currently set via front screen). The standard boilerflow temp command doesn't seem to do anything. Running in watch mode via telnet, I can see when I change the central heating temp to 70C on the front panel, the following is sent on the EMS bus:

000+07:31:40.874 N 22: [emsesp] Controller(0x09) -> Boiler(0x08), (0xE6), data: 01 46

Although from everything I've looked through, cmd 0xE6 isn't mentioned anywhere, so not sure if this is new and unsupported command?

Also periodically this set value seems to be sent back  via a 0x2D6 type message (Boiler was set to 69C at this point)

000+07:31:34.415 N 12: [emsesp] Boiler(0x08) -> (0x00), (0x2D6), data: 00 00 00 00 00 45 00 4F

I've tried sending:

send telegram "09 08 e6 01 46"

But I get an error back that 0xE6 is unknown command type

Any ideas?

proddy, 2020-09-12:
Hi @fiskn 

I'm not too familiar with the W-B range. You may first want to check out Norbert's mapping at https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf to see if that gives any hints.

The 0xE6 from the controller is interesting. Luckily @Th3M3 figured this out for last year (see https://github.com/Th3M3/buderus_ems-wiki/blob/master/Einstellungen%20des%20Regelger%C3%A4ts%20MC110.md#typ-0xe6-einstellungen-heizung). I'll be interested to see if this is read-only or we'll also able to modify the settings.

When we've figured this all out I'll be happy to extend ESM-ESP to support the 8000 range.

What do mean exactly that `send telegram "09 08 e6 01 46"` gives an error back? Can you provide a screenshot or log? It should just work ;)

MichaelDvP, 2020-09-12:
@fiskn I think it's the right idea, but wrong command syntax. You can ony send on the bus with the id of the ems-esp, which is 0x0B. Also you have to add the offset field. Try the command `send telegram "0B 08 e6 00 01 46"`

glitter-ball, 2020-09-12:
It’s very possible the boiler has no return sensor - my WB doesn’t. The boiler’s control strategy doesn’t need one, so they don’t fit one.

I added a DS sensor to the return pipe and feed in to HA via ems-esp that way. It seems to work pretty well.

glitter-ball, 2020-09-12:
When I just had on/off hardwired connections to my boiler, flow temp was fixed for CH and HW and set on the boiler. I added a WB Sense II (and a few other bits to make it work). CH flow temp is now calculated by the Sense II and sent to the boiler. HW flow temp seems to default to 85C, so HW cylinder recovers quickly.

fiskn, 2020-09-12:
@MichaelDvP  Yes, sending (send telegram "0B 08 e6 00 01 46") worked, the boiler set temp can be changed from EMS-ESP. Thanks for the hint.

I've noticed the boiler will alter both the burner power down to 1:10, but also modulate the built in pump when the return temp appears to get nearer the set temp, which is quite clever. I'm guessing it's trying to keep the return temp in the condensing range, that's assuming there is a return temp and this behaviour isn't based on some other logic. But I noticed  @glitter-ball graph shows pump speed always at 100%, so I wonder if that would suggest my boiler has a return temp to aid in this behaviour.

My goal is to still use the legacy on/off logic by controlling the zone valves with Relay's, but also at the same time issue a set temp command via EMS-ESP to change the boiler temperature depending on heating task. Something like:

- Mon-Sat 55C heat for Hot water with a boost to 70C on Sunday for sterilisation
- Radiators, heat at 80C until return temp gets near 50C, then drop boiler temperature down to 55C
- Similar behaviour to the Radiators for UFH circuit, but potentially lower temperatures

Then hopefully this means I can drive the boiler at it's max to heat up quickly, but keep it in the best efficiency zone at all times.

I will take a look through that link and see if there is anything else I can spot being sent from the boiler and I will also stick a ds18b20 sensor on the return pipe on the boiler and see if I can see a matching temperature being sent anywhere.

Thanks again

glitter-ball, 2020-09-13:
The installation manual will reveal whether the boiler has a return sensor or not.

I changed the pump modulation settings on my boiler from output-compensated to fixed speed because I wasn’t getting enough flow to far-flung parts of the system at low outputs. It does still modulate to a degree for CH and pops on bc and off to save energy once the house is warm.

glitter-ball, 2020-09-13:
In the [default?] output-compensated mode, the pump speed follows the burner output. As the flow temp approaches the setpoint, the burner modulates down and the pump follows. This may be what you’re seeing.

There are a few opportunities for improvement in the logic in my boiler. It starts the pump just before lighting the burner. It would be better if it ran the pump first for 10-15 seconds to establish the flow temp and let it stabilise and then light the burner. This would reduce the anti-cycling at low heating load, I think.

fiskn, 2020-09-13:
@glitter-ball thanks for the info, sounds like I need to have a read through the manual and see if there is anything else I can tune. I've got temp sensors on all the pipes sticking out the cylinder, so I've got a fairly good idea how hot and how much hot water is left. I'm hoping to base some nodered logic around this so make intelligent decision of whether the cylinder needs topping up. Currently the top and the middle of the cylinder never drops below 50C, so there is a case for optimisation there.

glitter-ball, 2020-09-13:
The pocket for my cylinder sensor is close to the return and, with 60C HW setpoint, the return ends up at pretty much bang on 60C when the HW reheat ends so that looks reasonable.

MichaelDvP, 2020-09-13:
@fiskn to implement this in ems-esp, can you give me some logs? In terminal go to boiler and give me the output of:
`read 16`, `read 1a`, `read 33`, `read 35`, `read e6` and `read ea`
also the log from `watch on 08` for more than a minute will help.

fiskn, 2020-09-13:
@MichaelDvP Here you go, I've also included 2d6 as that seems to display now and again

ems-esp:/boiler# read 16
002+05:38:24.824 N 1: [emsesp] Boiler(0x08) -> (0x0B), (0x16), data: <empty>
ems-esp:/boiler# read 1a
002+05:38:34.057 N 2: [emsesp] Boiler(0x08) -> (0x0B), (0x1A), data: <empty>
ems-esp:/boiler# read 33
002+05:38:39.143 N 3: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: <empty>
ems-esp:/boiler# read 35
002+05:38:45.547 N 4: [emsesp] Boiler(0x08) -> (0x0B), (0x35), data: <empty>
ems-esp:/boiler# read e6
002+05:38:48.678 N 5: [emsesp] Boiler(0x08) -> (0x0B), (0xE6), data: 01 44 00 58 64 0A 00 00 06 FA 05 01 02 64 0A 03 00 00 00 1E 01 00 01 00 01 00 41
ems-esp:/boiler# read ea
002+05:38:54.228 N 6: [emsesp] Boiler(0x08) -> (0x0B), (0xEA), data: 00 00 00 00 00 00 3C FB 00 28 00 02 46 00 00 00 3C 3C 28 1E 46 64 00 00 00 00 00
ems-esp:/boiler# read 2d6
002+05:39:00.010 N 7: [emsesp] Boiler(0x08) -> (0x0B), (0x2D6), data: 00 00 00 00 00 44 00 1A

`002+05:39:59.410 N 8: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 16 00 00 00 00 02 20 00 00 80 00 00 00 14 00 02 16 00 FF
002+05:39:59.587 N 9: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:39:59.777 N 10: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 16 00 64 44 00 00 00 00
002+05:40:00.041 N 11: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
002+05:40:09.423 N 12: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 16 00 00 00 00 02 20 00 00 80 00 00 00 14 00 02 16 00 AA
002+05:40:09.600 N 13: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:40:09.790 N 14: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 16 00 64 44 00 00 00 00
002+05:40:09.995 N 15: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
002+05:40:19.435 N 16: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 16 00 00 00 00 02 20 00 00 80 00 00 00 14 00 02 16 00 10
002+05:40:19.613 N 17: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:40:19.804 N 18: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 16 00 64 44 00 00 00 00
002+05:40:20.008 N 19: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
002+05:40:29.392 N 20: [emsesp] Boiler(0x08) -> (0x00), (0x2D6), data: 00 00 00 00 00 44 00 4D
002+05:40:29.575 N 21: [emsesp] Boiler(0x08) -> (0x00), (0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
002+05:40:29.771 N 22: [emsesp] Boiler(0x08) -> (0x00), (0xBF), data: 08 C3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
002+05:40:30.003 N 23: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 15 00 00 00 00 02 20 00 00 80 00 00 00 14 00 02 15 00 67
002+05:40:30.179 N 24: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:40:30.350 N 25: [emsesp] Boiler(0x08) -> (0x00), (0xD1), data: 80 00
002+05:40:30.527 N 26: [emsesp] Boiler(0x08) -> (0x00), (0xD7), data: 01 01
002+05:40:30.717 N 27: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 15 00 64 44 00 00 00 00
002+05:40:30.916 N 28: [emsesp] Boiler(0x08) -> (0x00), (0xE5), data: 00 00 00 00 00 64 00 00 20 00 00 00 7F 00 19 65 00 00 00 00 19 65 00 00 7F 00 7C
002+05:40:31.151 N 29: [emsesp] Boiler(0x08) -> (0x00), (0xE5), data: 00 00 00 00 00 (offset 27)
002+05:40:31.348 N 30: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
002+05:40:39.408 N 31: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 15 00 00 00 00 02 20 00 00 80 00 00 00 14 00 02 15 00 00
002+05:40:39.585 N 32: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:40:39.776 N 33: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 15 00 64 44 00 00 00 00
002+05:40:39.980 N 34: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
002+05:40:49.419 N 35: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 15 00 00 00 00 02 20 00 00 80 00 00 00 14 00 02 15 00 10
002+05:40:49.595 N 36: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:40:49.787 N 37: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 15 00 64 44 00 00 00 00
002+05:40:49.991 N 38: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
002+05:40:59.429 N 39: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 10 20 2D 2D 00 CB 00 02 15 00 00 00 00 02 1F 00 00 80 00 00 00 14 00 02 15 00 10
002+05:40:59.605 N 40: [emsesp] Boiler(0x08) -> (0x00), (0xE4), data: 00 09 60 00 FE 00 00 (offset 35)
002+05:40:59.797 N 41: [emsesp] Boiler(0x08) -> (0x00), (0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 15 00 64 44 00 00 00 00
002+05:41:00.000 N 42: [emsesp] Boiler(0x08) -> (0x00), (0xE9), data: 05 80 00 80 00 00 00 00 00 46 3C 00 00 02 00 00 00 00 00 00 00 00 00 05 02 00
[... cut ...]
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water charging: off
  Warm Water disinfecting: off
  Selected flow temperature: 0°C
  Current flow temperature: 53.2°C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Burner selected max power: 0%
  Burner current power: 0%
  Flame current: 0.0uA
  Pump modulation: 0%
  Pump modulation2: 0%
  Burner # starts: 127
  Total burner operating time: 4 days 12 hours 21 minutes
  Total heat operating time: 4 days 12 hours 21 minutes
`

MichaelDvP, 2020-09-14:
Good, the 0x16 and 0x33 (the other boiler equivalent to 0xE6/0xEA) are empty, this give no confusion reading the values from both telegram-sets. I add the kown values for these telegrams.

MichaelDvP, 2020-09-14:
@fiskn you can check the latest dev 2.0.2b0 if all works with your boiler.

fiskn, 2020-09-14:
Yep, perfect. Shows the current heating temp and can alter via MQTT.

closed 2020-09-26 by proddy