emsesp/EMS-ESP #496: SysLog improvements

proddy, 2020-09-12:
From a conversation in https://github.com/proddy/EMS-ESP/pull/494#issuecomment-691450178_

SysLog needs a bit tidying up. I suggest
 - moving the settings to its own page in the WebUI next to NTP and MQTT
 - renaming it to Logging
 - Add a enable/disable boolean, so it matches the other services
 - removing OFF as a log level option, as this could be confusing to some people

proddy, 2020-09-12:
@MichaelDvP any comments?

MichaelDvP, 2020-09-12:
In #494 you say log off in combination with mark on is a legal option, so why remove loglevel off?
For the discussed issue it's sufficient to remove the <fragment> in the settings page, a own menu item is more consistent, but not high priority.

proddy, 2020-09-12:
remove the OFF in the settings page?

MichaelDvP, 2020-09-12:
Ups, remove the Fragment-Tag, (i wrote it in brackets)

proddy, 2020-09-12:
ok, so leave SysLog where it is, leave all the levels as they are, but just add the enable/disable?

MichaelDvP, 2020-09-12:
An extra enable/disable is redundant. server=0 disables all, loglevel:off and mark:off disables also, all other combinations enables level, mark or both. Confusing was only that the mark settings hides, but stay active, without hiding the settings are clear to me.

proddy, 2020-09-12:
ok then. Empty host means no syslog.

proddy, 2020-09-13:
@MichaelDvP if you want to allow the SysLog IP to be empty (to fake disabling it) then you also need to modify the Web settings page. Default settings has an empty IP and it will show an error when a user tries to save the settings. Just remove the restriction.

[image]

MichaelDvP, 2020-09-14:
I tested with setting to `0` but did not realize that default is empty. I change it to allow empty host too.

proddy, 2020-11-08:
> An extra enable/disable is redundant. server=0 disables all, loglevel:off and mark:off disables also, all other combinations enables level, mark or both. Confusing was only that the mark settings hides, but stay active, without hiding the settings are clear to me.

@MichaelDvP I've added the Syslog enabled back, for two reasons. 1) the syslog code was running and causing some memory glitches on the esp32 even if it wasn't being used and 2) it's consistent with the other services like mqtt/adp etc

MichaelDvP, 2020-11-09:
I agree that it is consistant with the other settings. 
The memory leak is imho due to loglevel. If server is 0 but loglevel is set, the messages are queued, but not sent.  
But you should allow the settings:
- mark on, loglevel off
- loglevel on, mark off

Also it is better to allow changes of  the settings without reboot.
```
    if (!syslog_enabled_) {
        syslog_.log_level((uuid::log::Level)-1);
        syslog_.mark_interval(0);
        syslog_.destination(0);
   }
```
This should also clear the queue.

proddy, 2020-11-09:
@MichaelDvP ok, agree.

MichaelDvP, 2020-11-10:
I've pushed the changes as i suggested, please check if there are memory glitches. 
I've only tested on esp8266 and it looks good to me.

proddy, 2020-11-10:
Thanks for picking that up Michael. Looks fine, I did make a few comments in the push. 

I've been spending the last days trying to figure out why both the ESP32 and ESP8266 crash when certain log messages are sent. I think you ran into the same problem with the Rx code where the solution was to remove the DEBUG_LOG messages.

proddy, 2020-11-14:
think we can close this now.

proddy, 2020-11-16:
@MichaelDvP , what was your thoughts behind this piece of code in syslog.cpp:158

```c
    if (!syslog_enabled_) {
        syslog_.log_level((uuid::log::Level)-1);
        syslog_.mark_interval(0);
        syslog_.destination((IPAddress)((uint32_t)0));
        return;
    }
```

I don't think its necessary. If syslog service is disabled, it won't start

MichaelDvP, 2020-11-16:
On cold start it's not necessary, but if you change to syslog off while running, the service was started and i think logs are queued (but not send)  if not set to off. These settings ensure that the service do nothing.

proddy, 2020-11-16:
ok. thanks. I'm going to make some minor changes.

MichaelDvP, 2020-11-17:
What i've meant is documented in `syslog.h`, line 123: 
```
 * To disable sending messages, set the host to `0.0.0.0` and the
 * log level to uuid::log::Level::OFF (otherwise they will be
 * queued but not sent).
```

proddy, 2020-11-17:
Ok, then we need to add it back. It's commented out for now. Strange that there is no syslog_.stop() to disable the service. The reason I removed it originally was because @susisstrolch noticed the syslog host showed up as <notset> when the disabled, so overwriting the settings. He'll comment on his findings in this issue.

susisstrolch, 2020-11-17:
### syslog settings don't apply
No single syslog info shows up on syslog server. Also, in telnet console syslog is shown as disabled, even the server and debug level is set in web interface.

Configuration: Standalone board, w/o connection to EMSBus
```
{
  "System": {
    "version": "2.1.1b3",
    "uptime": "000+00:00:09.580",
    "freemem": 57,
    "fragmem": 1
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 0,
    "publish_time_thermostat": 60,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "disconnected"
  },
  "Devices": []
}
```

System configuration:
```
ems-esp-dev:/system$ show
Uptime:        000+00:01:54.893
Chip ID:       0x00f3bf2f
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  31
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x001640C8 (4194304 bytes)
Reset reason:  Exception
Reset info:    Fatal exception:28 flag:2 (Exception) epc1:0x40291d8d epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000039 depc:0x00000000
Sketch size:   962032 bytes (2183168 bytes free)
Free heap:                15456 bytes
Free mem:                 59  %
Heap fragmentation:       3 %
Maximum free block size:  15032 bytes
Free continuations stack: 1232 bytes

WiFi: Connected
SSID: xxxxxxxxxxxx
BSSID: 38:10:D5:B2:22:1E
RSSI: -66 dBm (68 %)
MAC address: 18:FE:34:F3:BF:2F
Hostname: ems-esp-dev
IPv4 address: 192.168.254.120/255.255.255.0
IPv4 gateway: 192.168.254.1
IPv4 nameserver: 192.168.254.1

Syslog: disabled
```
[image]

proddy, 2020-11-17:
@susisstrolch you're using an old version of the web. Rebuild it using npm and try again. You'll find that Syslog is disabled.

susisstrolch, 2020-11-17:
Argh... must build the env:espXXX-local...
Maybe you want to change the naming to env:espXXX-ems+npm?

proddy, 2020-11-17:
yes, I'll rename the env's esp8266 and esp32 to esp8266-release and esp32-release as these are used by the CI Github automated build scripts

susisstrolch, 2020-11-17:
@proddy syslog messages are arriving now.
but... there seems to be a binary artefact in the syslog message:
```
 Nov 17 10:16:46 192.168.254.120 1 - ems-esp-dev - - - - <feff>000+00:01:47.426 I 0: [syslog] Log level set to DEBUG
 Nov 17 10:16:46 192.168.254.120 1 - ems-esp-dev - - - - <feff>000+00:01:47.426 I 1: [emsesp] Syslog started
 Nov 17 10:16:53 192.168.254.81 1 - ems-esp - - - - <feff>002+15:21:20.150 D 543599: [emsesp] Received UBAMonitorFast
 Nov 17 10:16:54 192.168.254.81 1 - ems-esp - - - - <feff>002+15:21:20.430 D 543600: [emsesp] Received UBAMonitorFast
```
Hexdump:
```
00001900: 5d20 2d2d 204d 4152 4b20 2d2d 0a4e 6f76  ] -- MARK --.Nov
00001910: 2031 3720 3130 3a33 393a 3236 2031 3932   17 10:39:26 192
00001920: 2e31 3638 2e32 3534 2e31 3230 2031 202d  .168.254.120 1 -
00001930: 2065 6d73 2d65 7370 2d64 6576 202d 202d   ems-esp-dev - -
00001940: 202d 202d 20**ef bbbf** 3030 302b 3030 3a30   - - ...000+00:0
00001950: 353a 3031 2e31 3731 2044 2031 373a 205b  5:01.171 D 17: [
00001960: 6d71 7474 5d20 5075 626c 6973 6869 6e67  mqtt] Publishing
00001970: 2074 6f70 6963 2065 6d73 2d65 7370 2d64   topic ems-esp-d
00001980: 6576 2f68 6561 7274 6265 6174 2028 2330  ev/heartbeat (#0
00001990: 382c 2072 6574 6169 6e3d 302c 2074 7279  8, retain=0, try
000019a0: 2331 2c20 7369 7a65 2031 3434 2c20 7069  #1, size 144, pi
000019b0: 6420 3129 0a                             d 1).

```
Note the **\<feff\>** - as hexdump it's a '**ef bb bf**'...

proddy, 2020-11-17:
It's by design.

`'\xef\xbb\xbf'` is the UTF8 encoded version of the unicode ZERO WIDTH NO-BREAK SPACE U+FEFF. It is used as a Byte Order Mark (BOM) at the beginning of unicode text files, like syslogs. And part of the [RFC5424]( https://tools.ietf.org/html/rfc5424) protocol which the code is based on.

https://github.com/proddy/EMS-ESP/blob/f45a208ce4ebe25b36ca5f48e945e1d9c0ab554c/lib/uuid-syslog/src/syslog.cpp#L423

closed 2020-11-18 by proddy