emsesp/EMS-ESP #497: Add `write` command

proddy, 2020-09-12:
In the console we have a `read` command in each of the device menus, e.g. `ems-esp:/boiler# read 18`. There is also a generic `send` command to send commands to the EMS bus but this won't work for write commands because the have to follow an EMS poll request from the master/UBA.

So was thinking why not add a `write` command, per device context, such as `write <offset> <data>` which would just add the Tx telegram to the Tx queue.

MichaelDvP, 2020-09-12:
I don't see much benefit to the `send telegram` command. 
If i want to set for example `wwtemp` by raw command it's: `send telegram "0B 08 33 02 2E"` and a possible write command would only skip `<src>` and `<dst>`, e.g. `write 33 02 "2E"` but both working the same. 

**maybe a bit OT:**
I'm actually testing some similar enhancements.  What do you think about this (forget it? open an enhancement-discussion? make a test-fork?):

- `read <device_id> <type_id>` in main context, that works also for mixer, switch, etc without own context (i'm thinking if it is possible to make it global and have the device_id optional in the contexts.). 
- on a `send telegram` with reading: give values back (without setting a watch, like read-command):
- on a `send telegram` with write: verify the write and give values back (without setting a watch). example:
```
ems:/# send telegram "0B 08 33 02 2E"
000+01:47:02.342 N 1: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 2E FB FF 28 FF 07 46 00 00
ems:/# send telegram "0B 08 33 02 30"
000+01:47:54.255 N 2: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 30 FB FF 28 FF 07 46 00 00
```
- i can set any value via mqtt with the `send` command, but can't read a value that isn't fetched by ems-esp. I added a mqtt-topic `response` which works on writes and reads in this way:
writing to bus:
 `{"cmd":"send","data":"0B 08 33 02 30"}`
response  `{"src":"08","dest":"0B","type":"33","offset":"00","data":"08 FF 30 FB FF 28 FF 07 46 00 00"}`
reading a complete telegram:
`{"cmd":"send","data":"0B 88 33 00 20"}` 
response `{"src":"08","dest":"0B","type":"33","offset":"00","data":"08 FF 30 FB FF 28 FF 07 46 00 00"}`
reading a single value gives also the value if 1-4 bytes: 
`{"cmd":"send","data":"0B 88 19 02 02"}`
response `{"src":"08","dest":"0B","type":"19","offset":"02","data":"00 FA","value":250}`

For now i'm using 'read_id' for the feedback, so telnet 'send telegram' also publishes on mqtt-response and vice versa.

proddy, 2020-09-12:
I liked all those enhancements. Having `read` context-sensitive depending on which context/sub-menu you are is a nice touch and easy to implement. `read <device_id> <type_id>` can be only for the root. 

I'll rename this Enhancement Issue and we can use that for tracking. I suggest creating a feature branch (feel free to create one) and use that as these will be some substantial changes. I'd also want to have tests in the test framework. I've been using this a lot recently to simulate EMS and MQTT behavior and it's saved me hours of work without having an ESP or EMS bus nearby.

If you want some help or tasks for me to program let me know.

MichaelDvP, 2020-09-12:
yes `read <device_id> <type_id>`in root was my first approach and is implemented/tested in my dev, but thinking about of change to global with `[device_id]` and set it automatic in context, and a error message without context. 

Since the changes are small i'm not sure that an extra branch makes sense, I first pushed it now to my repro as dev.
Also include are the parasite_changes from #494 and what i think is simple solution for #496.

proddy, 2020-09-12:
go ahead and push it to dev. All your changes are awesome.

MichaelDvP, 2020-09-23:
I close this, with #510 we dont have the device contexts anymore and `read <device-id> <type-id>` is the only global command, also a `write` makes less sense, `send telegram` do the same with only the <src> byte as difference.

closed 2020-09-23 by MichaelDvP