emsesp/EMS-ESP #498: Migrate Travis CI to GitHub actions

proddy, 2020-09-12:
GitHub Actions is quite nice and if we move the CI process (automatic code checking, building firmwares) directly to the repo it will save on using external applications and enable all main contributors to tweak the settings. Other workflows could be added too, such as running some of the unit tests and simulating different devices.

proddy, 2020-09-15:
done. On push and pull in the dev branch the firmware builds are checked and created and binaries moved to https://github.com/proddy/EMS-ESP/tree/firmware/firmware

proddy, 2020-10-09:
looking into whether we can create github actions to build releases on tag commits and also nightly builds. Adding binary artifacts to the 'firmware' branch is not optimal.

proddy, 2020-10-29:
Dev builds are created when the file CHANGELOG_DEV.md is updated and the release is at https://github.com/proddy/EMS-ESP/releases/tag/dev

proddy, 2020-10-30:
build is triggered when `CHANGELOG_LATEST.md` is pushed.

Also create a script to build the official stable release from the `main` branch.

@MichaelDvP I'd like to get 2.1 out to everyone. bbqkees needs it for his next batch of gateways. And the mismatch between the documentation and 2.0.1 is confusing people. When you've done the NTP month+1 fix shall we make the next stable release? or would you rather wait to finish up some other issues?

MichaelDvP, 2020-10-30:
I think it's a good time for a new release. With the latest changes i'm not sure about the thermostat 0xBF. I thought it is on RC300 instead of A2 and used the same value, but now it seems both telegrams are present in RC300. Remove 0xBF? I think publishing 2 error-messages from the same device is to much, errors are very rare.

proddy, 2020-10-30:
agree, let's leave 0xBF for now.

MichaelDvP, 2020-10-30:
Ok, i remove it from thermostat.

closed 2020-11-13 by proddy