emsesp/EMS-ESP #500: Backup and restore settings through web GUI

glitter-ball, 2020-09-13:
Although the code does a really good job of preserving settings across updates, as the features and settings grow, would it be handy to be able to backup and restore the settings via the web GUI? Ideally this would be in some form of human-readable text file. Probably not an immediate high priority.

Potential issues:

- Dumping the existing settings out to a file is probably failry straighforward. Reading them back, validating them and handling errors is less fun. Also, how do you handle backups from one version being restored to a later (or earlier!) one?

- As an interim, offer just to dump settings out to a plain text file, as an easy way of capturing the current settings (this may already be possible with telnet). Users could then re-enter settings manually from this in the GUI, which would deal with the validation problem.

- How do you handle sensitive data? Ideally it would be encrypted in the settings file.

proddy, 2020-09-13:
I had this settings backup/restore feature in the 1.9 version, via the web interface. I decided not to port this over to the new Web UI in version 2.

The version is stored in the settings and used in the upgrade code when a newer firmware replaces the old.

I like your 2nd idea as a first quick approach. Perhaps a button in the WebUI that exports all the settings to a file and zip it with a password, because the file will show all the passwords (MQTT, Wifi, JWT secret, Users).

proddy, 2020-09-18:
in 2.0.2b3 a dump from the settings can be obtained via http://ems-esp/api?device=system&cmd=info

glitter-ball, 2020-09-19:
> in 2.0.2b3 a dump from the settings can be obtained via http://ems-esp/api?device=system&cmd=info

That looks promising. With the Settings page data too, that would pretty much nail it.

proddy, 2020-09-20:
it's all there now in 2.0.2b4. Looks very pretty with a chrome extension like https://chrome.google.com/webstore/detail/json-viewer-awesome/iemadiahhbebdklepanmkjenfdebfpfe?hl=en

proddy, 2020-09-23:
Perhaps final thing is to create a button in the Web UI which will export the settings.

proddy, 2020-09-23:
Added the link to the settings page in now what I'm calling 2.1.0b0.

closed 2020-09-23 by proddy

glitter-ball, 2020-09-24:
Hmm... Could I be cheeky and push my luck by asking for a CR/LF between each setting and its value and between "pages"? Would make it easier to digest? Something like...
```
{"WIFI":{
"ssid":"HIDDEN"
"hostname":"ems-esp"
"static_ip_config":true
"local_ip":"192.168.1.6"
"gateway_ip":"192.168.1.254"
"subnet_mask":"255.255.255.0"
"dns_ip_1":"192.168.1.254"}

"AP":{
"provision_mode":1
"ssid":"ems-esp"
"local_ip":"192.168.4.1"
"gateway_ip":"192.168.4.1"
"subnet_mask":"255.255.255.0"}

"MQTT":{
"enabled":true
"host":"192.168.1.5"
"port":1883
"username":"mqtt"
"client_id":"ems-esp"
"keep_alive":60
"clean_session":false
"max_topic_length":128
"system_heartbeat":true
"publish_time_boiler":0
"publish_time_thermostat":0
"publish_time_solar":10
"publish_time_mixing":10
"publish_time_other":0
"publish_time_sensor":0
"mqtt_format":1
"mqtt_qos":0
"mqtt_retain":false}

"NTP":{
"enabled":true
"server":"uk.pool.ntp.org"
"tz_label":"Europe/London"
"tz_format":"GMT0BST,M3.5.0/1,M10.5.0"}

"OTA":{
"enabled":true
"port":8266}

"Settings":{
"tx_mode":3
"ems_bus_id":11
"syslog_level":-1
"syslog_mark_interval":0
"syslog_host":""
"master_thermostat":0
"shower_timer":false
"shower_alert":false
"rx_gpio":13
"tx_gpio":15
"dallas_gpio":14
"dallas_parasite":false
"led_gpio":2
"hide_led":false
"api_enabled":true
"bool_format":1
}
}
```
Thank!

proddy, 2020-09-24:
LOL. It was done intentional. Keep the JSON format. But I highly recommend a chrome/firefox/edge extension like "JSON Viewer Awesome" to show you a pretty version of it.

glitter-ball, 2020-09-25:
Sorry - I should have realised it was in JSON (not something electrical engineers naturally keep up with!) I'll get a viewer...