emsesp/EMS-ESP #501: implement command-switch to turn Boiler central heating on / off

inter-faze, 2020-09-13:
feature request do implement a command to easily turn on/off the boiler central heating

This feature should imlement the same functionality as e.g. the change of the ( RC25 https://www.buderus.lv/files/Planungsunterlage_GB172_T50_EN_LV.pdf side 46 switching from 14 to 15 and vice versa of Fig. 40) 

Hint of Gitter chat: {"cmd":"send", "data":"0B 08 16 00 00"} to switch of - to switch on  {"cmd":"send", "data":"0B 08 16 00 FF"}

proddy, 2020-09-13:
this is relatively easy to add. Any suggestions on what to call the command?

MichaelDvP, 2020-09-14:
I suggest `heatingactivated` since there is already `wwactivated` and `wwtapactivated`. 

But it's not so easy: in #495 there is boiler that does not have 0x16 command, but 0xE6 for the same settings. I'm checking to automatic distinguish which telegram is send and verified. The same for the ww-parameters in 0x33 and 0xEA. For my boiler 0x16, 0x33 are active and 0xE6, 0xEA empty, in the other case 0x16, 0x33 empty and 0xE6, 0xEA active, so i check the fetch-flag for the parameter-telegrams to see what telegram is active. I also add `heatingtemp` to the commands as this is topic of #495.

proddy, 2020-09-14:
I see. It's more complicated than I first thought. One idea to distinguish between the two telegram typeids between ems1.0 and 2.0 is to use a Deviceflag in the device's library like Thermostats do

MichaelDvP, 2020-09-14:
The automatic detection works, i pushed it to my repo if you want to have a look. In thermostat there are a lot of different functions for different types, but boiler are the same settings, only sorted in different telegrams, but the unused telegrams are empty and can ruled out.
For syslog and sensors i pushed changes direct to you.

proddy, 2020-09-14:
@MichaelDvP if it works, I'm happy if you push it straight into dev

MichaelDvP, 2020-09-14:
Ok, i recheck the E6/EA telegrams and and other boiler settings (incl. #503) and than push it.

MichaelDvP, 2020-09-14:
@inter-faze, switch is added to dev 2.0.2b0, try it.

inter-faze, 2020-09-14:
I will be able to check it next week earliest, will let yo uknow

inter-faze, 2020-09-22:
Hello @proddy and @MichaelDvP: today I had time to upgrade from 1.9.5 to 2.0.2b. The heatingactivated and wwactivated work as expected. 
Is use the bbqkees device

alex-sever-h, 2020-09-26:
Hi. Thank you for this patch. I just got the hardware a couple of days ago, and started playing with it, trying to emulate a thermostat for my Buderus GB062, and I was impressed almost everything works out of the box. 

I tested the commands **heatingactivated** and **heatingtemp** to set the parameters, and I have mixed results:
**heatingtemp** - as far as I've seen, works as expected; when heating is on, I can change the temperature I see on the boiler control panel, and after some time, I see the flowtemperature is set to the same value.
**heatingactivated** - when I start and stop the heating (summer mode as in the GB062 documentation), I see the command received correctly (on telegram 0x16). However, if I try to set the value to on or off, nothing happens. The telegram is received properly, but the boiler remains off, and if I read the telegram, I get 00, not 255 as expected.

Any ideas how to debug this behavior ?

PS: Thank you to all involved in this project, it is great.

proddy, 2020-09-26:
@alex-sever-h do you see any traffic on 0xE6 or 0xEA ? There seems to be a difference between EMS1.0 and newer EMS+ boilers. @MichaelDvP knows this better than I do.

alex-sever-h, 2020-09-26:
Thanks for the fast response. As far as I've seen, this boiler is EMS1.0, I tried EMS+ commands but they did not work. Please correct me if I'm wrong. 
Mind you that I've added prints for the heating_activated and heating_temp in the logs below.

Periodically, I see this, so I assume it's EMS1.0:
```
001+14:08:05.417 D 200: [telegram] Sending read Tx [#28], telegram: 0B 88 16 00 20 EC
001+14:08:05.466 D 201: [emsesp] Last Tx read successful
001+14:08:05.494 D 202: [emsesp] Received UBAParameters
001+14:08:05.494 D 203: [boiler] heating_activated_ (0)
001+14:08:05.494 D 204: [boiler] heating_temp_ (0)
```

Then log of the **heatingactivated** commands is here:
```
ems-esp:/# call boiler heatingactivated 1
output: {}

001+14:04:49.741 I 43: [boiler] Setting boiler heating 
001+14:04:49.747 D 44: [telegram] Sending write Tx [#1], telegram: 0B 08 16 00 FF 57
001+14:04:49.779 D 45: [emsesp] Last Tx write successful
001+14:04:49.779 D 46: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
001+14:04:49.829 D 47: [emsesp] Received UBAParameters
001+14:04:49.829 D 48: [boiler] heating_activated_ (0)
001+14:04:49.829 D 49: [boiler] heating_temp_ (0)
001+14:04:49.926 D 50: [mqtt] Publishing topic ems-esp/boiler_data (#16392, attempt #1, pid 1)
001+14:04:50.091 D 51: [telegram] Sending read Tx [#2], telegram: 0B 88 16 00 20 EC
001+14:04:50.141 D 52: [emsesp] Last Tx read successful
001+14:04:50.145 D 53: [emsesp] Received UBAParameters
001+14:04:50.145 D 54: [boiler] heating_activated_ (0)
001+14:04:50.145 D 55: [boiler] heating_temp_ (0)
```

What is interesting that setting **heatingtemp** only works if heating is activated on the boiler manually (on the control panel), otherwise is read as 00. See here:
```
ems-esp:/# call boiler heatingtemp 50
output: {}

001+14:09:09.946 I 245: [boiler] Setting boiler heating temperature to 
001+14:09:09.962 D 246: [telegram] Sending write Tx [#40], telegram: 0B 08 16 01 32 98
001+14:09:09.994 D 247: [emsesp] Last Tx write successful
001+14:09:09.994 D 248: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
001+14:09:10.044 D 249: [emsesp] Received UBAParameters
001+14:09:10.044 D 250: [boiler] heating_activated_ (0)
001+14:09:10.044 D 251: [boiler] heating_temp_ (0)
001+14:09:10.215 D 252: [mqtt] Publishing topic ems-esp/boiler_data (#16424, attempt #1, pid 1)
001+14:09:10.263 D 253: [emsesp] Received UBAParameters
001+14:09:10.263 D 254: [boiler] heating_activated_ (0)
001+14:09:10.263 D 255: [boiler] heating_temp_ (0)
001+14:09:10.494 D 256: [telegram] Sending read Tx [#41], telegram: 0B 88 16 00 20 EC
001+14:09:10.544 D 257: [emsesp] Last Tx read successful
001+14:09:10.548 D 258: [emsesp] Received UBAParameters
001+14:09:10.548 D 259: [boiler] heating_activated_ (0)
001+14:09:10.548 D 260: [boiler] heating_temp_ (0)
```

But after I manually start heating on the boiler control panel, I see as expected:
```
ems-esp:/# call boiler heatingtemp 50
output: {}

001+14:12:55.697 I 497: [boiler] Setting boiler heating temperature to 
001+14:12:55.747 D 498: [telegram] Sending write Tx [#77], telegram: 0B 08 16 01 32 98
001+14:12:56.435 D 499: [telegram] Sending write Tx [#78], telegram: 0B 08 16 01 32 98
001+14:12:56.467 D 500: [emsesp] Last Tx write successful
001+14:12:56.467 D 501: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
001+14:12:56.518 D 502: [emsesp] Received UBAParameters
001+14:12:56.518 D 503: [boiler] heating_activated_ (255)
001+14:12:56.518 D 504: [boiler] heating_temp_ (50)
001+14:12:56.551 D 505: [mqtt] Publishing topic ems-esp/boiler_data (#16457, attempt #1, pid 1)
001+14:12:56.779 D 506: [telegram] Sending read Tx [#79], telegram: 0B 88 16 00 20 EC
001+14:12:56.831 D 507: [emsesp] Last Tx read successful
001+14:12:56.835 D 508: [emsesp] Received UBAParameters
001+14:12:56.835 D 509: [boiler] heating_activated_ (255)
001+14:12:56.835 D 510: [boiler] heating_temp_ (50)
```

MichaelDvP, 2020-09-26:
The `heatingactivated` command does not work on all boilers. My system with MC10 controller has a mechanical poti to set boiler on and temperature (max is auto). If i try to switch heating off, it is overwritten by the MC10. Also temp works only if in auto position. Here is how it is overwritten: 
```
ems:/# watch on 16
Watching incoming telegrams, displayed in decoded format
ems:/# call boiler heatingactivated off
000+01:41:11.885 I 2: [boiler] Setting boiler heating
000+01:41:12.317 N 3: [emsesp] Boiler(0x08) -> (0x0B), (0x16), data: 00 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
000+01:41:22.095 N 4: [emsesp] Boiler(0x08) -> (0x00), (0x16), data: 00 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
000+01:41:22.351 N 5: [emsesp] Controller(0x09) -> Boiler(0x08), (0x16), data: FF
000+01:41:32.378 N 6: [emsesp] Boiler(0x08) -> (0x00), (0x16), data: FF 5A 64 00 0A FA 0F 02 06 64 64 02 08 F8 0F 0F 0F 0F 1E 05 04 09 09 00 28 00 3C
ems:/# watch off
```

alex-sever-h, 2020-09-26:
Thanks for looking into it, @MichaelDvP . My boiler does not have any mechanical switch whatsoever, just a few buttons.
To manually set heating, I would have to press the heating button, then press **^** or **v** buttons to select temperature.
If I send the heatingtemp through EMS bus, I just have to press the heating (radiator) button, and I already see the selected temperature.

Here is trying to set it on when the boiler is off:
```
ems-esp:/# watch on 16
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x16
ems-esp:/# call boiler heatingactivated on
output: {}

001+15:56:38.853 I 129: [boiler] Setting boiler heating 
ems-esp:/# 
001+15:56:39.362 D 130: [telegram] Sending write Tx [#185], telegram: 0B 08 16 00 FF 57
001+15:56:39.394 D 131: [emsesp] Last Tx write successful
001+15:56:39.394 D 132: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
001+15:56:39.475 N 133: [emsesp] Boiler(0x08) -> (0x00), (0x16), data: 00 00 57 00 00 FB 03 01 03 64 0A 02
001+15:56:39.475 D 134: [emsesp] Received UBAParameters
001+15:56:39.475 D 135: [boiler] heating_activated_ (0)
001+15:56:39.475 D 136: [boiler] heating_temp_ (0)
001+15:56:39.506 D 137: [mqtt] Publishing topic ems-esp/boiler_data (#17251, attempt #1, pid 1)
001+15:56:39.705 D 138: [telegram] Sending read Tx [#186], telegram: 0B 88 16 00 20 EC
001+15:56:39.756 D 139: [emsesp] Last Tx read successful
001+15:56:39.760 N 140: [emsesp] Boiler(0x08) -> (0x0B), (0x16), data: 00 00 57 00 00 FB 03 01 03 64 0A 02
001+15:56:39.760 D 141: [emsesp] Received UBAParameters
001+15:56:39.760 D 142: [boiler] heating_activated_ (0)
001+15:56:39.760 D 143: [boiler] heating_temp_ (0)
```
Here is trying to set it off when the boiler is on:
```
ems-esp:/# watch on 16
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x16
ems-esp:/# call boiler heatingactivated off
output: {}

001+16:01:38.012 I 421: [boiler] Setting boiler heating 
ems-esp:/# 
001+16:01:38.072 D 422: [telegram] Sending write Tx [#227], telegram: 0B 08 16 00 00 A8
001+16:01:38.105 D 423: [emsesp] Last Tx write successful
001+16:01:38.105 D 424: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
001+16:01:38.154 N 425: [emsesp] Boiler(0x08) -> (0x00), (0x16), data: FF 23 57 00 00 FB 03 01 03 64 0A 02
001+16:01:38.154 D 426: [emsesp] Received UBAParameters
001+16:01:38.154 D 427: [boiler] heating_activated_ (255)
001+16:01:38.154 D 428: [boiler] heating_temp_ (35)
001+16:01:38.251 D 429: [mqtt] Publishing topic ems-esp/boiler_data (#17292, attempt #1, pid 1)
001+16:01:38.385 D 430: [telegram] Sending read Tx [#228], telegram: 0B 88 16 00 20 EC
001+16:01:38.434 D 431: [emsesp] Last Tx read successful
001+16:01:38.439 N 432: [emsesp] Boiler(0x08) -> (0x0B), (0x16), data: FF 23 57 00 00 FB 03 01 03 64 0A 02
001+16:01:38.439 D 433: [emsesp] Received UBAParameters
001+16:01:38.439 D 434: [boiler] heating_activated_ (255)
001+16:01:38.439 D 435: [boiler] heating_temp_ (35)
```
I see no controller messages whatsoever here.

However, I can see there is a controller if I try to list the devices:
```
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:45.10)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:45.10)
```

But I can not find any documentation on this specific controller.

MichaelDvP, 2020-09-26:
@alex-sever-h The boiler seems to ignore the command. 
Try terminal with `watch on 16` and change the settings for on/off and temperatur on the boiler.

alex-sever-h, 2020-09-26:
@MichaelDvP  That is what I did previously. It's just that I also had logging enabled, so there is more noise in the messages.

MichaelDvP, 2020-09-26:
@alex-sever-h You have logged two parts, one with heating off, one with heating on and the try to change by ems-esp. I am interrested in the middle part where you changed the boiler setting. 
You can also set 'watch on 8` to see all from the boiler and change the settings on the boiler. 

The ems-esp-command is ignored from your boiler, here you sending successful the command "FF" to the first databyte
```
001+15:56:39.362 D 130: [telegram] Sending write Tx [#185], telegram: 0B 08 16 00 FF 57
001+15:56:39.394 D 131: [emsesp] Last Tx write successful
```
here is the readback with first databyte 00
```
001+15:56:39.394 D 132: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
001+15:56:39.475 N 133: [emsesp] Boiler(0x08) -> (0x00), (0x16), data: 00 00 57 00 00 FB 03 01 03 64 0A 02
```
But i see that also the set-temperatur changes to 0 or 35 degree, maybe you have to set the temperatur too. 
Try to switch off by `send telegram "0B 08 16 00 00 00"` or on by `send telegram "0B 08 16 00 FF 23"`

closed 2020-10-16 by MichaelDvP