emsesp/EMS-ESP #502: Feature to call commands from the Web UI

proddy, 2020-09-14:
Now that all commands are unified across the Console and MQTT, having the same ability to call commands from the Web UI may be useful. For example calling 'temp' in the Thermostat box to set a new setpoint temp. Or 'wwonetime' from the Boiler. It could be a simple form with a drop-down of commands (per context) and two fields for `data` and `id` similar to the MQTT format. And only accessible if the user has 'admin' privileges.

An alternative is to open the telnet console to the web. The advantage of this is you'll see the logging.

MichaelDvP, 2020-09-20:
With the additions of #506 heap is ~2k lower, not enough to reload web-page if telnet is open on esp8266.  I think you can cross out the second option (telnet on web).

proddy, 2020-09-20:
I was thinking when you select a device in the EMS Devices web page, it shows the value below and below that show a new form where you can choose a command from a drop-down, a value, and an id. It's a luxury feature and not high-priority but it does kind of round of the Web UI as being a fully functional dashboard & command center.

proddy, 2020-11-18:
@MichaelDvP do you still think its a good idea?

MichaelDvP, 2020-11-18:
With the API it's possible to do a command from web, but not very comfortable in the browser-url.
A input-line in web for a call command would be fine, or like scan-devices a button for a dialogbox to enter a command.

proddy, 2020-11-18:
What I dislike about the current API is that it's insecure (if you enable it in the Settings) so anyone can send write commands to the EMS-ESP. Having these calls from the Web UI offers some protection and only accessible via a user with the Administrator role. I like the idea of a button which pops up a dialog window where you select the target (system, boiler, thermostat etc), command (free text) and data and id (both optional). The only worry I have is that we're close to 92% progmem flash so not much space left for new web code.

The Web API really needs to use the JWT token for added security. Something to add onto the list for later.

closed 2021-05-18 by proddy