emsesp/EMS-ESP #503: new software doesn't see thermostat

cyniu88, 2020-09-14:
I made upgrade to version v2.0.1, but after that my ems-esp cannot start onetime heating boiler by mqtt.  on telnet session i can see trace: "000+00:30:34.996 I 0: [boiler] Setting boiler warm water OneTime loading on"  after  send request  via mqtt, but buderus  didn't start heating.
Previously on version  1.9.5  everything worked fine.

Yes, I upgrade MQTT message to new format

My configuration:
These EMS devices are currently active:

Boiler: Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x18 0x19 0x34 0x1C 0x2A 0x33 0x14 0x35 0x15 0x16 0x1A 0xD1 0xE3 0xE4 0xE5 0xE9

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)
 This Thermostat will respond to telegram type IDs: 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E

Controller: BC25 (DeviceID:0x09, ProductID:125, Version:03.05)

proddy, 2020-09-14:
that's not good. We need to see what is happening.
- First, do you have any Tx errors (check in Web or `show ems` from console)
- If not, go to console, turn logging on (`log debug`) and then `boiler`, `su`, `call wwonetime on`
- Monitor the log to see what telegram is sent and what the response is

cyniu88, 2020-09-14:
**show ems**

> EMS Bus is connected.
> 
> EMS Bus info:
>   Tx mode: 1
>   Bus protocol: Buderus
>   #telegrams received: 2018
>   #read requests sent: 333
>   #write requests sent: 2
>   #incomplete telegrams: 0 (0%)
>   #tx fails (after 3 retries): 0
> 
> Rx Queue is empty
> 
> Tx Queue is empty

**iDom-client/buderus/ems-esp:/boiler# log all**
Log level = all
000+01:02:46.203 D 4: [emsesp] No telegram type handler found for ID 0xA2 (src 0x10)
000+01:02:50.489 D 5: [mqtt] Publishing topic iDom-client/buderus/ems-esp/boiler_data (#827, attempt #1, pid 1)
000+01:02:50.690 D 6: [mqtt] Publishing topic iDom-client/buderus/ems-esp/thermostat_data (#828, attempt #1, pid 1)
000+01:02:53.903 D 7: [emsesp] Received MC10Status
000+01:02:54.148 D 8: [emsesp] Received UBAMonitorWW
000+01:02:54.378 D 9: [emsesp] Received UBAMonitorFast
000+01:03:00.083 D 10: [emsesp] Fetching values for device ID 0x08
000+01:03:00.083 D 11: [telegram] Tx read request to device 0x08 for type ID 0x19
000+01:03:00.083 D 12: [telegram] Tx read request to device 0x08 for type ID 0x33
000+01:03:00.083 D 13: [telegram] Tx read request to device 0x08 for type ID 0x14
000+01:03:00.083 D 14: [telegram] Tx read request to device 0x08 for type ID 0x16
000+01:03:00.083 D 15: [emsesp] Fetching values for device ID 0x09
000+01:03:00.083 D 16: [emsesp] Fetching values for device ID 0x10
000+01:03:00.083 D 17: [telegram] Tx read request to device 0x10 for type ID 0x2A5
000+01:03:00.083 D 18: [telegram] Tx read request to device 0x10 for type ID 0x2B9
000+01:03:00.220 D 19: [telegram] Sending read Tx [#143], telegram: 0B 88 19 00 20 D0
000+01:03:00.288 D 20: [emsesp] Last Tx read successful
000+01:03:00.293 D 21: [emsesp] Received UBAMonitorSlow
000+01:03:00.548 D 22: [mqtt] Publishing topic iDom-client/buderus/ems-esp/boiler_data (#829, attempt #1, pid 1)
000+01:03:00.585 D 23: [telegram] Sending read Tx [#144], telegram: 0B 88 33 00 20 78
000+01:03:00.657 D 24: [emsesp] Last Tx read successful
000+01:03:00.661 D 25: [emsesp] Received UBAParameterWW
000+01:03:00.749 D 26: [mqtt] Publishing topic iDom-client/buderus/ems-esp/thermostat_data (#830, attempt #1, pid 1)
000+01:03:00.917 D 27: [telegram] Sending read Tx [#145], telegram: 0B 88 14 00 20 E4
000+01:03:00.965 D 28: [emsesp] Last Tx read successful
000+01:03:00.969 D 29: [emsesp] Received UBATotalUptime
000+01:03:01.151 D 30: [mqtt] Publishing topic iDom-client/buderus/ems-esp/heartbeat (#831, attempt #1, pid 1)
000+01:03:01.283 D 31: [telegram] Sending read Tx [#146], telegram: 0B 88 16 00 20 EC
000+01:03:01.352 D 32: [emsesp] Last Tx read successful
000+01:03:01.356 D 33: [emsesp] Received UBAParameters
000+01:03:01.615 D 34: [telegram] Sending read Tx [#147], telegram: 0B 90 FF 00 20 01 A5 19
000+01:03:01.744 D 35: [emsesp] Last Tx read successful
000+01:03:01.749 D 36: [emsesp] Received RC300Monitor
000+01:03:02.080 D 37: [telegram] Sending read Tx [#148], telegram: 0B 90 FF 00 20 01 B9 05
000+01:03:02.206 D 38: [emsesp] Last Tx read successful
000+01:03:02.210 D 39: [emsesp] Received RC300Set
**iDom-client/buderus/ems-esp:/boiler# call wwonetime on**
000+01:03:03.653 I 40: [boiler] Setting boiler warm water OneTime loading on
000+01:03:03.707 D 41: [telegram] Sending write Tx [#149], telegram: 0B 08 35 00 22 06
000+01:03:03.741 D 42: [emsesp] Last Tx write successful
000+01:03:03.741 D 43: [telegram] Sending post validate read, type ID 0x18 to dest 0x08
000+01:03:03.907 D 44: [emsesp] Received UBAMonitorFast
000+01:03:03.967 D 45: [mqtt] Publishing topic iDom-client/buderus/ems-esp/boiler_data (#832, attempt #1, pid 1)
000+01:03:04.102 D 46: [emsesp] Received MC10Status
000+01:03:04.343 D 47: [emsesp] Received UBAMonitorWW
000+01:03:04.538 D 48: [telegram] Sending read Tx [#150], telegram: 0B 88 18 00 20 D4
000+01:03:04.623 D 49: [emsesp] Last Tx read successful
000+01:03:04.628 D 50: [emsesp] Received UBAMonitorFast
000+01:03:05.967 D 51: [emsesp] Received RC300Monitor
000+01:03:10.605 D 52: [mqtt] Publishing topic iDom-client/buderus/ems-esp/boiler_data (#833, attempt #1, pid 1)
000+01:03:10.806 D 53: [mqtt] Publishing topic iDom-client/buderus/ems-esp/thermostat_data (#834, attempt #1, pid 1)
[... 83 lines cut ...]
  Burner current power: 0%
  Flame current: 0.0uA
  System service code: 0H (203)
  Heating temperature setting on the boiler: 35°C
  Boiler circuit pump modulation max power: 100%
  Boiler circuit pump modulation min power: 10%
  Boiler circuit pump delay time: 3min
  Boiler temp hysteresis on: -6°C
  Boiler temp hysteresis off: 6°C
  Boiler burner min period: 10min
  Boiler burner min power: 0%
  Boiler burner max power: 71%
  Outside temperature: 24.9°C
  Pump modulation: 0%
  Burner # starts: 568
  Total burner operating time: 56 days 23 hours 16 minutes
  Total heat operating time: 54 days 8 hours 0 minutes
  Total UBA working time: 226 days 23 hours 39 minutes

000+01:03:24.006 D 62: [emsesp] Received MC10Status
000+01:03:24.281 D 63: [emsesp] Received RCTime
000+01:03:24.505 D 64: [emsesp] Received UBAMonitorWW
000+01:03:24.735 D 65: [emsesp] Received UBAMonitorFast
**iDom-client/buderus/ems-esp:/boiler# log info**
Log level = info

MichaelDvP, 2020-09-14:
The command `0B 08 35 00 22 06` is right, the same as in 1.9.5, and sended successfully. The post validate read should be on 0x34  (not 0x18), but that does not harm the function. Also i see Onetime is not listed in show command.

@cyniu88  as far as i know the onetime funktion only starts if ww current temperature is much below setpoint (i'm not sure if it was 5 or 7 or more degree). If you set the setpoint higher (~45), does the onetime works than?

cyniu88, 2020-09-14:
I don't know why, but after the upgrade wWSelTemp changed from 60 to 40 and I missed this change, after setting it to 60 the boiler starts to heat tap weather

proddy, 2020-09-14:
>  Also i see Onetime is not listed in show command.

do you mean in the console? It is there as `wwonetime`

[image]

MichaelDvP, 2020-09-14:
No, i meant the response to the `show` command in terminal, there i added the state for onetime charging with the last commit.
[image]

proddy, 2020-09-15:
what are the next steps for the issue?

cyniu88, 2020-09-15:
IMHO, main thread is solved, and ticket could be closed.

StrokaPatric, 2020-09-28:
hi all, 
sorry for reply, since the update to 2.0.1. i´ve cannot start onetime heating by mqtt.
Maybe someone can show me the MQTT new format, i try a lot.
At this time i´ve got three issues by MQTT Publish Errors but i can´t find anything in the logs.

Thank you all.

Greeting

MichaelDvP, 2020-09-28:
Here is the [mqtt format](https://emsesp.github.io/docs/#/MQTT?id=receiving-topics), the commands are linked there (API section)

StrokaPatric, 2020-09-28:
Thank you for quick Answer,
my Problem is only by onetime heating, maybe someone had the same problem. In 1.9.5 it still work.

proddy, 2020-09-30:
@StrokaPatric update to 2.1, then try from the console. Use https://emsesp.github.io/docs/#/Troubleshooting to help with enabling the logs.

StrokaPatric, 2020-09-30:
You mean 2.0.1 ?

proddy, 2020-09-30:
no the latest dev build 2.1. You can find it https://github.com/proddy/EMS-ESP/tree/firmware 
this is the latest version we've been developing and kept in sync with the Documentation. I'll need to update the README on this GitHub rep so its more clear to everyone that there are in fact two versions.

StrokaPatric, 2020-10-02:
Okay, i've updated to build 2.1. , now i get more information from the log.
But i got an new Problem and it show's like it is comming from the Smarthome software (xSolution). Is there some experience with the payload like {"cmd" : <wwonetime> , "data":<on>}? i can see in the log an MQTT error: payload cmd, error InvalidInput. I have tryed the Informations from the Documentation, but i disregard an issue, likely my own.

proddy, 2020-10-06:
original issue has be solved. @StrokaPatric if this is not working for you can you open a new GitHub issue

closed 2020-10-06 by proddy