emsesp/EMS-ESP #505: setting of  setpoint room temperature not working

cyniu88, 2020-09-15:
setting of  setpoint room temperature not working on version 2.0.1.
logfile in attachment.
[MobaXterm_192.168.13.111_20200915_204322.txt](https://github.com/proddy/EMS-ESP/files/5227121/MobaXterm_192.168.13.111_20200915_204322.txt)

proddy, 2020-09-15:
which thermostat do you have, the RC300? Did it work with 1.9.5 or any previous version? We're looking into it....

cyniu88, 2020-09-15:
yes, it worked on 1.9.5, if it's needed, I can do downgrade(if it is possible) and collect additional logs.
my hardware:
Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0x2A5 0x2B9 0x2A6 0x2BA 0x2A7 0x2BB 0x2A8 0x2BC 0x31D 0x31E

proddy, 2020-09-15:
I think we'll be able to work out the problem with the data you provided. To be clear do you have a Buderus RC300, RC310, Nefir Moduline 3000, Junkers CW400 or a Bosch Sense II? It does make a difference.

proddy, 2020-09-15:
just quickly looking at the log dump, I see your thermostat is in the 'manual' mode but you're trying to set the temperature for 'auto' mode.

cyniu88, 2020-09-16:
I have a Buderus RC300, I can see it too, but I also don't see a specific parameter for the temp command that will allow me to change it in manual or automatic mode.

MichaelDvP, 2020-09-16:
as @proddy already analyzed, the `temp` command sets the wrong value. And there are also some other parameters missing for RC300, Norbert has the telegrams on his list, but some values are not completly clear to me. There are 3 levels of comfort temperature and the monitor-message has in pos 11 (17 in Norberts list) the heating level. But how is it set the level? For now only the value for comfort2 is set and published as `daytemp`.

What we need is:
- on `temp`command set the right value (auto/manual)
- add `ecotemp`, `manualtemp`, `comforttemp` to the commands
- rename `daytemp`, `nighttemp` to `comforttemp`, `ecotemp`
- publish `comfortlevel` in mqtt and terminal

That's easy to add.

But how to manage the levels?  
(a) List all comfort-values (`comfort1temp`,  `comfort2temp`, ..) in mqtt, or (b) only the active `comforttemp` depending on level?
(a) Add commands for `comfort1temp`, `comfort2temp`, .., or (b) only one command `comforttemp` and set only the active level?

Both options can be confusing to people, on (a) they have to know which is the right value, on (b) they can not set a inactive value. Also a combination of both options with `comforttemp`, `comfort1temp`, `comfort2temp`, `comfort3temp` can be confusing (but maybe less).
 
@cyniu88 You have this controller, what is your opinion?

MichaelDvP, 2020-09-16:
In Norberts list i find also for 1B9 (in Norberts numbering):
- 15 xy Outdoor Temperature Threshold
I think this is `summertemp`
- 18 xy Outdoor Temperature Threshold for prevention of ECO
and this seems to be `nofrosttemp`

but in the RC300 manual i did not find any setting for different comfort temperatures. 

@cyniu88 can you send a log with `watch on 2B9` and change summertemp, nofrosttemp, and other values you may find interresting and note for these values the changes: when, from value, to value. 
Please give me also a value from `watch on 2A5` or a `read 2A5` from thermostat context to see if comfort_level is set.
Also `read A5` to check if building type, etc. are on the same telegram as in RC35, than i can add it.

cyniu88, 2020-09-16:
Yes I can do it but only in the evening. I am currently at work.

cyniu88, 2020-09-16:
in version 1.9.5 I just published the desired temperature value on mqtt in the subject "iDom-client / buderus / ems-esp / thermostat_cmd_temp1"

Thermostan in my case always works in manual mode, I just want to

cyniu88, 2020-09-16:
sorry but an cannot see all this parameter in my RC 300 for example comfort level, but i collected some logs
[MobaXterm_192.168.13.111_20200916_205446.txt](https://github.com/proddy/EMS-ESP/files/5234506/MobaXterm_192.168.13.111_20200916_205446.txt)

MichaelDvP, 2020-09-17:
The values i thought it is `summer`and `nofrost` looks odd with 0 and 15. Default values, if never changed, are 17 and 5 degree. 
The value for comfort level are as expected, level is 1, and temperature 19, 18, 23 degree. But if there is no way to select a value i think it's better to stick to a single value (comfort2 at pos2). We can add later if requested. I'll push a change to the first 3 points mentioned.

proddy, 2020-09-17:
thanks MichaelDvP. Also worth looking into why it worked in 1.9.5.

MichaelDvP, 2020-09-17:
I've looked into 1.9.5 and, as expected, the offset is set in dependence of the actual mode. Same logic as now in v2.

MichaelDvP, 2020-09-17:
@proddy There is a check failed on my commit, but i doesn't understand what is wrong, can you take a look. Here it compiles without error.

proddy, 2020-09-17:
Looks like GitHub's actions is offline. https://github.com/proddy/EMS-ESP/runs/1127413777?check_suite_focus=true. The firmware is built but it can't upload the artifacts. Error is:

```
###### End Diagnostic HTTP information ######
{ 'message': 'Github Actions Unavailable. We are working to restore all services as quickly as possible. Please check back soon.', 'ref': 'Ref A: FD211D5C4FD84317815C383C386E7888 Ref B: CO1EDGE0622 Ref C: 2020-09-17T08:14:02Z' }
##[error]Unable to finish uploading artifact firmware to https://pipelines.actions.githubusercontent.com/7eMvrRgeOwlows6fe6KuXjLhWtwlJCWwmZpCgQ97NYghWr0sdt/_apis/pipelines/workflows/258965598/artifacts?api-version=6.0-preview&artifactName=firmware
```

we can manually restart the job in a few hours when the service is restored.

proddy, 2020-09-17:
works now. builds are available in the firmware directory

proddy, 2020-09-20:
@cyniu88 do you want to retry with the latest dev build and see if this bug is now resolved?

cyniu88, 2020-09-22:
yes, I want to check, where can I find binary file ?

proddy, 2020-09-22:
https://emsesp.github.io/docs/#/Uploading-firmware

cyniu88, 2020-09-23:
hi, on dev branch problem seems to be solved :)

proddy, 2020-09-23:
yay!

lsw271, 2020-09-25:
Hello @proddy, I have a problem with setting the thermostat temperature and warm water temperature.
my configuration 
`Boiler: Junkers GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)`
mine is Junkers Cerapur
`Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.04) Clock: 21:25:08 25/09/2020` and CW400

Yesterday I had topic ems-esp/thermostat_data1, today its gone (its normal?). Now I've got another homeassistant/climate/ems-esp/hc1 
[image]

My climate configuration in HA:

**thermostat**
'- platform: mqtt
  name: Thermostat_test2
  modes:
    - "auto"
    - "heat"
    - "off"
  mode_command_topic: "homeassistant/climate/ems-esp/hc1/state"
  temperature_command_topic: "homeassistant/climate/ems-esp/hc1/state"

  mode_state_topic: "homeassistant/climate/ems-esp/hc1/state"
  current_temperature_topic: "homeassistant/climate/ems-esp/hc1/state"
  temperature_state_topic: "homeassistant/climate/ems-esp/hc1/state"

  mode_state_template: "{{ value_json.hc1.mode }}"
  current_temperature_template: "{{ value_json.hc1.currtemp }}"
  temperature_state_template: "{{ value_json.hc1.seltemp }}"

  temp_step: 0.5'

**warm water**
'- platform: mqtt
  name: boiler
  modes:
    - "auto"
    - "off"
  min_temp: 40
  max_temp: 60
  temp_step: 1
  
  current_temperature_topic: "ems-esp/boiler_data"
  temperature_state_topic: "ems-esp/boiler_data"
  mode_state_topic: "ems-esp/boiler_data"

  current_temperature_template: "{{ value_json.wWCurTmp }}"
  temperature_state_template: "{{ value_json.wWSelTemp }}"
  mode_state_template: "{% if value_json.wWActivated == 'off' %} off {% else %} auto {% endif %}"

  temperature_command_topic: "ems-esp/boiler_cmd_wwtemp"
  mode_command_topic: "ems-esp/boiler_cmd_wwactivated"'

when I change something after a few seconds, the previous temperature returns

maybe i need to turn on some configuration to write to devices (thermostat / boiler)? I've got the gateway since December '19 and i focus to read data not write (maybe apart from trying to set burner minimum power [https://github.com/proddy/EMS-ESP/issues/349](url) :)) but i think setting thermostat temperature was working earlyier

lsw271, 2020-09-25:
i tried also change temperature with instruction in this issue https://github.com/proddy/EMS-ESP/issues/439 but i'havent got any logs in telnet after publish the topic

proddy, 2020-09-26:
@lsw271 you're using the setting MQTT Format to `Home Assistant`. This will use HA's MQTT Discovery format when publishing the topics and payloads, which is why the thermostat data looks different. But not to worry, this is the correct way. You can safely delete that thermostat code from `sensors.yaml` file. In HA you'll see a single state called `climate.hc` (check Developer Tools->State) which you can use in your lovelace interface with any of the thermostat cards.

For the Boiler keep the yaml as I'm still working on bringing that over to HA's MQTT way of working.

Now for the error when setting the temperatures please create a new issue and add this information
 - which version of EMS-ESP are you using? 2.0.1 or 2.1.0b
 - what are you changing, the thermostat data or the boiler's warm water temperature
 - what happens when you run the command in the Telnet Console with logging on? (https://emsesp.github.io/docs/#/Troubleshooting)

lsw271, 2020-09-27:
> lsw271 you're using the setting MQTT Format to `Home Assistant`. This will use HA's MQTT Discovery format when publishing the topics and payloads, which is why the thermostat data looks different. [...]

@proddy  Ok, thanks its working with climate.hc1. Sorry :) 
But in HA i've got only automatically mode, it that ok? 
[image]

Themperature was changed (i saw this in telnet) but it returns to 0 (zero) after few second but its only in HA, thermostat got my set temperature, maybe because of summer mode? I've got 3 heating modes in my thermostat - 1. Summer 2. Summer from X degrees outside 3. Always winter. Ems-esp sends Summer, Eco, Comfort but in my case Eco and Comfort its only for warm water mode, not for central heating. Could we do sth with it?

PS. i would like to ask in general as i am going to finally deal with my boiler / thermostat (previously i havent got time for this). Should I report problems in a specific category? or maybe should I collect everything that hurts me in one issue:)

lsw271, 2020-09-27:
few more words about central heating modes in thermostat

(upper level modes)
1. Always summer
2. Summer from X degrees outside
3. Always winter

(lover level modes) 
1. Auto (with hours / days timetable)
  - daytemp
  - nighttemp
2. Manual 
  - constant temp without using timetable

Now thermostat in HA looks like manual mode with setting constant temperature. It is possible to do thermostat which we could set daytemp and nighttemp, modes in the list it will be automatically and manual. And switch to set upper level modes (also with setting X degrees to summer mode)?

proddy, 2020-09-27:
@lsw271 best create a new issues specifically for your setup so we can get it all working correctly. Since you have a Junkers Cerapur and a CW400 it's quite interesting to get this fully supported.

I wouldn't worry too much about HA yet because I'm working on updating that. Let's focus on the settings that are shown in the Console.

MichaelDvP, 2020-09-28:
@lsw271 
> Themperature was changed (i saw this in telnet) but it returns to 0 (zero) after few second but its only in HA, thermostat got my set temperature, maybe because of summer mode?

Yes, that's because of summermode, in summermode the thermostat sets the actual setpoint to zero, but not the stored one. You can send a temp command, but this temp becomes active after thermostat leaves summermode. In my RC35 it's the same behaviour.

commit a8bf42bb (2020-09-17): Fix #505, add RC300 mqtt commands
closed 2020-10-07 by proddy