emsesp/EMS-ESP #506: HTTP read/write

mglatz, 2020-09-16:
I'd like to propose an option to read and write data with systems that do not directly support MQTT, such as Loxone.
The idea is to publish JSON readings directly to a webpage (such as http://ems-esp.local/json-data) so that systems not compatible with MQTT could just read and scrap the information.
For publishing messages, I'd suggest GET (such as http://ems-esp.local?topic=boiler_cmd&cmd=wwtemp&value=20) and possibly at simple authentication (htttp://USERNAME:PASSWORD@ems-esp...) 
this could also work together with https://github.com/proddy/EMS-ESP/issues/502
for displaying the data, I'd also suggest to add optional conversion of strings to values (on/off => 1/0 etc)

proddy, 2020-09-16:
Nice suggestion. The best approach is build an API over HTTP and use Oauth2 for the authentication. Exposing passwords over http post scares the willies out of me. There are some examples on how GitHub do this (https://developer.github.com/v3/), esphome (https://esphome.io/components/api.html#) and Tasmota (https://tasmota.github.io/docs/Commands/). Needless to say a bit more thought needs to go into this. Do you want to take the lead?

Doing a basic http://ems-esp.local/json-data could be a simple stop-gap solution.

Having the option to convert boolean to 0/1,  off/on, false/true is also a nice option. Can you create a separate issue for that?

bbqkees, 2020-09-17:
I already received multiple requests for Loxone support so its something worth looking into.

The boolean conversion is also a good idea, in Domoticz un-nested MQTT payload of 0 and 1 always gives trouble.
on/off or true/false is better in that case because its a string and not a number.

mglatz, 2020-09-17:
I strongly suggest there would at least be an default-off option for no/basic authentication as systems such as Loxone will just not coupe with oauth.

proddy, 2020-09-17:
I'll start with the original suggestion e.g. `http://ems-esp.local?topic=boiler_cmd&cmd=wwtemp&value=20)` and see how far we get.

proddy, 2020-09-18:
2.0.2b3 implements http://ems-esp/api?device=XXX&cmd=XXX&data=XX&id=XX e.g. `http://ems-esp/api?device=boiler&cmd=wwtemp&data=20&id=1`

MichaelDvP, 2020-09-20:
Wow, a lot of work. Tested some commands without any issue (except some info-commands are not yet implemented). 

Since this is a way to send commands without authentification there should be a option to turn it off.

proddy, 2020-09-20:
yeah, busy with adding the other devices and sensors.

Agree there should be a new setting to enable/disable the write commands to the API. I could also add the authentication to the URL like tasmota does it (`http://<ip>/cm?user=<username>&password=<password>&cmnd=Power%20On`) but it's not very secure.

proddy, 2020-09-20:
added sensors, mixing and solar. Can't really test so let me know if there are problems. Syntax is `http://ems-esp/api?device=sensor&cmd=info`

Also renamed sensors to sensor to keep consistent naming with the other devices.

Still some work to do. I'm not 100% happy with the implementation but its a start.

MichaelDvP, 2020-09-20:
Tested api-info-command for boiler, thermostat, mixing, sensor, all working well (i have no ems-solar). Switching off api commands works also. 
In single/custom mqtt the thermostat_datax is not published, please insert in `thermostat.cpp`, line 445: `Mqtt::publish(topic, rootThermostat);`

> Also renamed sensors to sensor to keep consistent naming with the other devices.

Than mqtt publish as 'sensor_data` (nested/HA/custom) would be also more consistent.

proddy, 2020-09-23:
@mglatz pushed these changes into the latest dev build, v2.1.0b0. Please try it out and re-open this issue if you find bugs.

closed 2020-09-23 by proddy

mglatz, 2020-09-24:
thank you, will do