emsesp/EMS-ESP #507: MQTT retaind message

cyniu88, 2020-09-16:
is it possible to return the flag "retained" in MQTT settings as in 1.9.5 ?

proddy, 2020-09-16:
Can do. What's the use case you had in mind?

cyniu88, 2020-09-16:
I create my own application to view data and configure my boiler and after connecting to the mqtt broker I have to wait for the boiler to send a message. And for this short moment (at worst 10 seconds) I only see "0" :) Hope you understand my case when you take the phone and check the tap temperature it takes less than 3 seconds if messages are retained.

cyniu88, 2020-09-16:
I know there are many tricks to fix this but the mqtt retain flag seems to be the best

proddy, 2020-09-16:
Ok. Have you played with the publish interval setting?

cyniu88, 2020-09-16:
yes, but a small interval generates a lot of traffic on the broker, in my opinion it is better to avoid huge traffic.

proddy, 2020-09-18:
in 2.0.2

closed 2020-09-18 by proddy