emsesp/EMS-ESP #508: how does the command wwactivated work?

SpaceTeddy, 2020-09-18:
Dear all,

in my system Sieger BK15, es73, WW-Buffer, I tried to use the command ``wwactivated`` or ``wwtapactivated`` to disable my warm water heating during my holiday time. I noticed that this is not really working and warm water is continuous heated up during that time.  If I send ``wwtapactivated`` the heater send an error message  [5H 268](https://www.ersatzteile-koeln.de/buderus-fehlermeldung-ems-steuerung/stoerung-4/fehlercode-5h-xxx/fehlermeldung-5h-268.php) , but I tried this only once.

Has somebody already tested the wwactivated command?

thank a lot.

Chris

MichaelDvP, 2020-09-18:
I've tried `wwactivated`  and that works for me. Here is the test, it's the second byte in message 33:
```
ems:/# boiler
ems:/boiler# log info
Log level = info
ems:/boiler# read 33
000+00:48:45.270 N 1: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 30 FB FF 28 FF 07 46 00 00
ems:/boiler# call wwactivated off
000+00:48:56.926 I 2: [boiler] Setting boiler warm water off
ems:/boiler# read 33
000+00:49:03.640 N 3: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 00 30 FB FF 28 FF 07 46 00 00
ems:/boiler# call wwactivated on
000+00:50:16.240 I 4: [boiler] Setting boiler warm water on
ems:/boiler# read 33
000+00:50:21.012 N 5: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 30 FB FF 28 FF 07 46 00 00
ems:/boiler# call wwactivated off
000+00:50:58.482 I 6: [boiler] Setting boiler warm water off
ems:/boiler# show
Boiler: GB125/MC10 (DeviceID:0x08, ProductID:72, Version:03.00)
  Warm Water activated: off
  Warm Water type: buffer
  Warm Water charging type: charge pump
```
The tapwater function uses the the testmode to switch burner, valve and pump on. That causes the message you get, but this works only on flow systems. Also i see that there is indeed a bug: `wwtapactivated off` only switches the valve off, not the testmode. Please check the testmode by `read 1D` and if the first byte is not zero do  `send telegram 0B 08 1D 00 00`
Can you check for your boiler with the above commands: `read33`, `call wwactivated`

MichaelDvP, 2020-09-18:
Sorry, i was wrong: there is no bug in `wwtapactivated`, switch off from testmode works (complete message is set to zero).

MichaelDvP, 2020-09-18:
Another correction of my comments: on/off is inverted:
```
ems:/boiler# call wwtapactivated off
000+01:22:40.801 I 13: [boiler] Setting tap warm tap water off
ems:/boiler# read 1D
000+01:22:47.295 N 14: [telegram] Rx: 08 0B 1D 00 5A 00 00 64 FF 00 00 00 00 00 00 E6
000+01:22:47.299 N 15: [emsesp] Boiler(0x08) -> (0x0B), (0x1D), data: 5A 00 00 64 FF 00 00 00 00 00 00
ems:/boiler# call wwtapactivated on
000+01:23:07.525 I 16: [boiler] Setting tap warm tap water on
ems:/boiler# read 1D
000+01:23:13.751 N 17: [telegram] Rx: 08 0B 1D 00 00 00 00 00 00 00 00 00 00 00 00 1E
000+01:23:13.754 N 18: [emsesp] Boiler(0x08) -> (0x0B), (0x1D), data: 00 00 00 00 00 00 00 00 00 00 00
```

SpaceTeddy, 2020-09-18:
Hi Michael,

I made the following tests:
a) turn the WW knob at the heater to "Auto":
``ems-esp-2:/# read 08 33
000+00:04:19.411 N 4: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 0F 00 02 41 00 FF``
b) turn the WW knob at the heater to "0":
``ems-esp-2:/# read 08 33
000+00:03:45.282 N 3: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 00 3C FB 00 0F 00 02 41 00 FF``
c) turn the WW knob at the heater to "ECO":
``ems-esp-2:/# read 08 33
000+00:06:17.024 N 8: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 0F 00 02 41 D8 FF``

d) send wwactive off via mqtt (knob is in AUTO position):
000+00:12:02.505 I 11: [boiler] Setting boiler warm water off
``ems-esp-2:/# read 08 33
000+00:12:15.482 N 12: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 0F 00 02 41 00 FF``
e) send wwactive on via mqtt (knob is in "AUTO" position):
``000+00:14:20.228 I 14: [boiler] Setting boiler warm water on
ems-esp-2:/# read 08 33
000+00:14:34.611 N 15: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 0F 00 02 41 00 FF``

f) send wwactive on via mqtt (knob is in "0" position):
``000+00:17:30.984 I 20: [boiler] Setting boiler warm water on
ems-esp-2:/# read 08 33
000+00:17:45.707 N 21: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 00 3C FB 00 0F 00 02 41 00 FF``
g) send wwactive off via mqtt (knob is in "0" position):
``000+00:19:45.343 I 22: [boiler] Setting boiler warm water off
ems-esp-2:/# read 08 33
000+00:19:53.941 N 23: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 00 3C FB 00 0F 00 02 41 00 FF``

h) send wwactive on via mqtt (knob is in "ECO" position):
``000+00:21:12.181 I 25: [boiler] Setting boiler warm water off
ems-esp-2:/# read 08 33
000+00:21:25.459 N 26: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 0F 00 02 41 D8 FF``
i) send wwactive off via mqtt (knob is in "ECO" position):
``000+00:21:51.323 I 27: [boiler] Setting boiler warm water on
ems-esp-2:/# read 08 33
000+00:21:59.215 N 28: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 FF 3C FB 00 0F 00 02 41 D8 FF``

for telnet call commands the same:
``000+00:25:58.619 N 5: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 00 3C FB 00 0F 00 02 41 00 FF
ems-esp-2:/boiler# call wwactivated on
ems-esp-2:/boiler# read 33
000+00:26:41.703 N 6: [emsesp] Boiler(0x08) -> (0x0B), (0x33), data: 08 00 3C FB 00 0F 00 02 41 00 FF``

log DEBUG:

ems-esp-2:/boiler# call wwactivated on
000+00:28:22.656 I 59: [boiler] Setting boiler warm water on
000+00:28:22.827 D 60: [telegram] Sending write Tx [#44], telegram: 0B 08 33 01 FF C1
000+00:28:22.852 D 61: [emsesp] Last Tx write successful
000+00:28:22.852 D 62: [telegram] Sending post validate read, type ID 0x33 to dest 0x08
000+00:28:22.895 D 63: [emsesp] Received UBAParameterWW
000+00:28:23.090 D 64: [mqtt] Publishing topic ems-esp-2/boiler_data (#408, attempt #1, pid 1)
000+00:28:23.268 D 65: [emsesp] Received UBAParameterWW
000+00:28:23.523 D 66: [telegram] Sending read Tx [#45], telegram: 0B 88 33 00 20 78

ems-esp-2:/boiler# call wwactivated off
000+00:29:43.454 I 148: [boiler] Setting boiler warm water off
000+00:29:43.524 D 149: [telegram] Sending write Tx [#54], telegram: 0B 08 33 01 00 3E
000+00:29:43.549 D 150: [emsesp] Last Tx write successful
000+00:29:43.549 D 151: [telegram] Sending post validate read, type ID 0x33 to dest 0x08
000+00:29:43.592 D 152: [emsesp] Received UBAParameterWW
000+00:29:43.621 D 153: [mqtt] Publishing topic ems-esp-2/boiler_data (#426, attempt #1, pid 1)
000+00:29:44.046 D 154: [telegram] Sending read Tx [#55], telegram: 0B 88 33 00 20 78

Does it sends the correct data to the boiler?

thx,
Chris

MichaelDvP, 2020-09-18:
Yes, i can reproduce this. The controller overwrites the setting after next publish (with watch 33):
```
000+03:39:52.976 I 4: [boiler] Setting boiler warm water off
000+03:40:03.104 N 7: [emsesp] Boiler(0x08) -> (0x00), (0x33), data: 08 00 30 FB FF 28 FF 07 46 00 00
000+03:40:03.359 N 8: [emsesp] Controller(0x09) -> Boiler(0x08), (0x33), data: FF (offset 1)
000+03:40:13.386 N 9: [emsesp] Boiler(0x08) -> (0x00), (0x33), data: 08 FF 30 FB FF 28 FF 07 46 00 00
```
You can check if the thermostat command `call wwmode off` works. That seems to be stable, but boiler ww keeps activated and wwontime on thermostat should be possible.
```
ems:/thermostat# call wwmode off
000+03:48:36.120 I 19: [thermostat] Setting thermostat warm water mode to off
000+03:48:36.698 N 20: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: 00 00 00 00 00 01 01 00 3C FF
...
000+03:54:03.560 N 28: [emsesp] Thermostat(0x10) -> (0x0B), (0x37), data: 00 00 00 00 00 01 01 00 3C FF
```

SpaceTeddy, 2020-09-18:
hi,
I will try the call wwmode off/Auto/on command and give feedback tomorrow, but the LED setting on the es73 looks promising. :)
good hint to use this command 👍

SpaceTeddy, 2020-09-18:
Hi Michael,
it works with the thermostat wwmode command. I've tested it with my programmed WW time-schedule. 
So, thanks again for your help.

closed 2020-09-23 by proddy