emsesp/EMS-ESP #509: Option to choose how booleans are rendered in Console, Web and MQTT

proddy, 2020-09-19:
Currently booleans are on and off. Add settings in Web to allow either
 on / off
 1 / 0
 true / false

which will be used for rendering values in the Web and Console UIs as well as the MQTT payloads.

MichaelDvP, 2020-09-24:
Working good, only one exception is `/api?device=system&cmd=info` which always shows true/false without quotation marks.
Is there a special reason for this? (Like a plan to import the settings as this json?)

proddy, 2020-09-24:
the true/false is JSON standard on how booleans are represented. I didn't add this new rendering method to system/info but could. No plans to build an import yet. There's another github issue on backup & restore but it's quite a bit of work.

MichaelDvP, 2020-09-24:
I have seen the other backup & restore issue, that's why i thought you plan to use this json without passwords to also simple restore settings. If not, i think it's more consistent to change this also to the selected booleans.  But it doesn't really matter.

I tried with change `void render_boolean()` to `char * render_boolean()` and use in system.cpp:
```
        char       s[6];
        node["enabled"]  = Helpers::render_boolean(s, settings.enabled);
```
Should i push this?

proddy, 2020-09-24:
> Should i push this?

please do. thanks

MichaelDvP, 2020-09-25:
Updated the system-info.

closed 2020-09-25 by MichaelDvP