emsesp/EMS-ESP #510: add console contexts for solar, mixing and the other connected devices

proddy, 2020-09-20:
Currently, there are only context menus in the Console for boiler and thermostat. I'd like to extend this to Solar and Mixing too, but will undoubtedly run into the same problem where there can be multiple solar/mixing devices so the user has to select the 'master device' like it's done in the Thermostat module. It probably makes sense to refactor this code to make it a standard part of the EMSdevice class. So 'set master' is a common command.

proddy, 2020-09-21:
Slept on it and now I have a better idea; get rid of the context menus for devices and only keep the `system` context. Having these separate context menus for each device was originally to overcome heap memory issues loading the [console commands](https://emsesp.github.io/docs/#/Console?id=console-commands) but since they're now offloaded to the new 'call command' subsystem this is no longer a problem. Also switching to contexts and back is cumbersome. 

Here's a better idea, from the root
- `show devices` lists all the devices, highlighting the device_id a little more clearly as this is now the key.
- `call <cmd>` replaces with `call <device_id> <cmd>`. Based on the device_id we can lookup which device type it is.
- `set master <device_type> <device_id>` to handle the case when there are multiple devices of the same device type (thermostats, solar, mixing).

behind the scenes, we'd need to 
- remove the context_menu code from all the devices. This will free up a lot of memory too which is always nice
- create a vector/map lookup table to store the master device_id for each device_type. This is saved in the settings.
- be able to view/modify the master lookup table in the WebUI so the master device can be modified. We'll still keep 0 for 'auto' meaning the first device it discovers.

@MichaelDvP  thoughts?

MichaelDvP, 2020-09-22:
I think it is a good idea. 
But i'm not sure if master-devices apart from thermostat really nedded. For mixing there is no master, every mixing-device does it's own job on hc or wwc, independent of the others. `call <device-id> ...` and `read <device-id> ...` addresses every module, also no need for a master.  For solar i don't know what multible modules do.

proddy, 2020-09-22:
ok, I'll go ahead and make the change and keep 'set master thermostat' only for the thermostat.

proddy, 2020-09-23:
done in 2.1.0b0

closed 2020-09-23 by proddy