emsesp/EMS-ESP #511: Scanned devices not saved in v2 release

lendy007, 2020-09-22:
**Bug description**
The problem solved in #323 is back after upgrade to 2.0.1 release. I can perform `scan devices deep` so it will find my RC30 but there is no option to save this scan resulting in loss of RC30 after each restart of EMS-ESP...

Another issue - even when I scan for RC30 manually and I can see its parameter in web console - it is not sending information about hc1 in MQTT messages.

**MQTT**
`{"intoffset":0,"minexttemp":-15,"building":"medium","wwmode":"auto"}`

**Web interface**
[image]

Would it be possible to check on this please?

Thank you

proddy, 2020-09-22:
I remember that issue. 

The correct method that all EMS devices follow is to query the EMS Bus Master (which is the boiler) for all its connected devices. In your case, which is very special, the Boiler doesn't want anything to know about the RC30. This could be a fault in the boiler or a configuration in the RC30 or something else in your EMS network. I haven't seen any other cases of this. It's surprising that the RC30 even works thinking about it.

I'll be happy to run some more experiments with you to find out why this is happening, but I will not be adding this feature as it breaks all the design principles of v2.

As for MQTT not showing the values, try setting the master thermostat. If it still fails please create a new issue.

MichaelDvP, 2020-09-22:
@lendy007 
> Another issue - even when I scan for RC30 manually and I can see its parameter in web console - it is not sending information about hc1 in MQTT messages.

What mqtt format have you set? In `single` and `custom` the there are other topics for the hc: `thermostat_data1` for hc1, `thermostat_data2` for hc2, etc.

lendy007, 2020-09-22:
@MichaelDvP - I am using Home Assistant format, but I've tried also the others but no information from thermostat...

I will downgrade to 1.9.5 as v2 is unusable for me with my current boiler :-( Hopefully is downgrade possible and won't brick the gateway...

MichaelDvP, 2020-09-22:
For HA there is also an extra topic. `thermostat_data`  for the thermostat data without hcs like you see it,  and `homeassistant/climate/ems-esp/hc1/state` for hc1.

closed 2020-10-09 by proddy