emsesp/EMS-ESP #512: AM2302/DHT22 Support

Sonusss, 2020-09-22:
Hi,

It looks like the current version only supports DALLAS 1 wire sensors.
Can you please add support for DHT22 which is Temp & RH sensor ?

Thx

[AM2302-ETC.pdf](https://github.com/proddy/EMS-ESP/files/5262566/AM2302-ETC.pdf)

proddy, 2020-09-22:
I'd rather not. It's a lot of work for me to code and extend the core and easier if you just spend the 1.5 euros and purchase a ds18b20. :-)

Sonusss, 2020-09-22:
The point is getting also humidity...
Maybe I'll give a try myself even not being senior programmer.
From my point of view there is a value turning ems/esp into a kind of environmental measuring station.
We have most interesting internal values, we have at least external temp when probe is present and we can extend measurements without the need of extra ESP devices consuming energy...;-)

proddy, 2020-09-22:
I'm just not convinced there are users that want to measure humidity?

I'm done many projects with the DHT22 so know how to code it, the complexity is adding it to EMS-ESP alongside the 1-wire implementation for both ESP8266 and ESP32, making sure it's tested and not conflicting with any other I/O pins and extending the web UI to show even more data.  It'll take some time to get it right for everyone. If DHT22s used 1-wire it would have been simpler.

I don't want EMS-ESP to be another esphome, homie, tasmota, espurna clone that does everything. It's a one-trick pony that reads EMS data. Everything else is just non-critical fancy stuff.

If more users think it's a good idea I can add it...let's see how the market reacts !

Sonusss, 2020-09-22:
Hey Proddy,

Do not over react !
I fully understand your point.
That was just a suggestion and I can fully attest about my complete support to your wonderful job to reverse engineer the EMS world and commitment to it.

I thought this could be simple, if not just forget for now.

However, just because I'm the kind of awful throl in the surrounding, adding support for most popular environmental probes would be a plus for an interface dealing with environment and comfort.

As I said, I will give it a try and if any success will contribute and let you know.

Most respectful regards, :-) not joking !

JFB

closed 2020-09-24 by Sonusss