emsesp/EMS-ESP #513: Firmware upgrade not working properly

glitter-ball, 2020-09-23:
**Bug description**
When a new .bin firmware file is dragged and dropped into the v2 GUI, the update does not proceed as code authors think it should.

**Steps to reproduce**
Drag and drop new .bin file into the upgrade page.

**Expected behaviour**

**Actual behaviour** 
Upload bar quickly moves to 100% then no further feedback from GUI. Navigating off page produces orange “Upload cancelled by user”. Reloading GUI shows updated firmware ID so upload has worked but no progress pop ups seen.

glitter-ball, 2020-09-23:
Anyone else seeing this behaviour too? Did try and video it but can’t upload MP4s and hosting it somewhere else was messy.

proddy, 2020-09-23:
Which browser are you using?

glitter-ball, 2020-09-23:
I get the same behaviour in latest versions of Firefox and Edge, both on Windows 10.

MichaelDvP, 2020-09-23:
I have also firefox (tweaked) and edge (stock) on win10. With both i see the expected behaviour. Here for Edge:
After drop the bin:
[image]
a few secondes later:
[image]
The box also works if the window is small. 
[image]
The green box disapears after one or two seconds, it takes some more time until  esp is up again.

@glitter-ball do you see the green box when save settings, after login, or never?

glitter-ball, 2020-09-24:
I've never seen the "Activating new firmware" popup above.

I get the green popup for login and settings saving in Firefox. There's also something else odd going on. When I attempt to connect in Microsoft Edge, I get:
[image]
and the login page never appears, so I can't check the popups in Edge. Edge on my iPhone does work OK with popups. Connecting with Firefox does bring up the login page but I wait 10-15 seconds with a blank white screen for it to appear.

glitter-ball, 2020-09-24:
Tried shrinking the Firefox window to make sure nothing getting lost behind Windows task bar. No good.

But observed something new. Watched MQTT Explorer. There is a gap in the data stream (in red here) after the FW uploads, but, crucially there is no 'info' telegram and the uptime in the heartbeat telegram hasn't reset:
[image]

When I then try and kick the GUI back into life, there's another hiccup and eventually it recovers and the missing info telegram and update up time in heartbeat:
[image]

So, although the firmware uploads and there is a pause in MQTT traffic, I'm not sure the device actually restarts until I poke it somehow, whereupon it resets the uptime counter and sends the info telegram.

Hopefully, there's a clue in all this!

proddy, 2020-09-30:
I still can't reproduce this error (GUI upload on Win10). Anyone else seeing the same behaviour?

glitter-ball, 2020-09-30:
Maybe I should wipe the device <gulp> and see if it still happens? Very odd...

proddy, 2020-09-30:
lets just wait a while to see if anyone else has the issue. There are so many factors involved (firewalls, browser plugins, wifi strength). The web upload basically uses the same logic as the espota.py so if that is working I would expect the web upload to work too.

glitter-ball, 2020-09-30:
OK. One option would be to wireshark the session and see what traffic’s actually passing between browser and device. That would narrow things down a bit.

proddy, 2020-10-07:
closing this to prepare for 2.1. Will re-open if others have the same issue.

closed 2020-10-07 by proddy