emsesp/EMS-ESP #514: MQTT Nefit easy + Excellent HR30

mupsje, 2020-09-23:
Hi,

With V1.9.5 I received with the settings from BBQKees and Wiki. 
several values from my setup.

Nefit easy -- connected to 
Nefit Easy Adapter and --
wired to the Gateway Premium II bought from BBQKees.

Now after update to 2.0.1  I only receive MQTT- values from ems-esp/Thermostat_data

[image]
[image]
[image]
[image]
[image]

proddy, 2020-09-23:
Hi Remco, thanks for logging the issue.

Two problems I see here, first is the Boiler (on device 0x08) is not recognized as an actual Boiler. Can you tell me which Nefit boiler you have?

Second issue it looks like the MQTT is not being sent. From the MQTTExplorer screenshot it's showing all the old retained messages from version 1.9.5. In version 2 the MQTT topics are no longer prefixed with `home/`. So delete all of them using MQTTExplorer (trashcan icon top right), reboot EMS-ESP and see if anything turns up. If not then MQTT is likely broken and you'll need to look into the broker's logs to see what is happening. Are you using HA's internal broker or another one like Mosquitto? Perhaps the username/password is incorrect or not handled correctly in EMS-ESP.

It's also worth updating EMS-ESP to the latest build. You can get the firmware from https://github.com/proddy/EMS-ESP/tree/firmware/firmware. Make sure you choose the ESP8266 version.

To get more debugging, go into the Telnet console and type `show mqtt` to see if anything is stuck in the logs.

proddy, 2020-09-23:
oh your boiler is an "Nefit Ecomline Excellent Hr 30". I see now.

mupsje, 2020-09-24:
Hi Proddy,

Thanks allot for the great feedback
I did upload the new Bin and.....  boom Nefit it is.
Also did a show mqtt in telnet for you.
Today I'm going to test more with Node red and Hassio
Thanks .
[image]

proddy, 2020-09-24:
great. Now we need to make sure MQTT is working. In the telnet if you do a `publish` command it will send all the data immediately to MQTT so it's a good test. You should also see things in the MQTT queue if you're quick. Note to use the publish command you need first to be Admin (use `su` and the default password).

mupsje, 2020-09-24:
Denk dat het nu goed moet zijn...
Nu nog kijken hoe ik het allemaal in Hassio ga zetten..

Think its ok now, challenge to get all into Hassio now..

[image]

proddy, 2020-09-24:
dat ziet er inderdaad wat beter uit. 

But I'm still missing a few MQTT topics that EMS-ESP sends, like `info` and `status`. Check my previous screenshot.

mupsje, 2020-09-24:
Denk dat we er nu zijn.
Heb de RX op stand 3 gezet en boem!
Think this is it!
Put the RX on value 3 and after publish command boooom!!
[image]

mupsje, 2020-09-24:
Now, how to Hassio hahahahaha

glitter-ball, 2020-09-24:
This sounds like a path I've trodden recently. #485 has some of the changes I needed to make to get everything back in HA.

At some point, I need to try the HA MQTT option - I'm still using the "Single" format.

closed 2020-10-06 by proddy

TimelessNL, 2020-12-05:
@mupsje Your Ecomline Excellent HR30 uses iRT right?
If my assumption is correct the Bosch Easy Connect simulates the iRT boiler on the EMS bus as an EMS boiler?