emsesp/EMS-ESP #515: MQTT topic "heating_active "

cyniu88, 2020-09-24:
is MQTT topic heating_active should be available on version 2.1.0b0?

proddy, 2020-09-24:
yes. is it broken? 

[image]

MichaelDvP, 2020-09-24:
Depends on the system. The `Central heating` is linked to `tapwater active` and if it a system with ww-buffer the tapwater does not work and also `Central heating`.  In ww-flow systems both topics are published.

proddy, 2020-09-24:
I remember, related to #354 & #465 

We haven't found a solution that works for all setups.

MichaelDvP, 2020-09-24:
Oh, there is a solution, but we have to check some cases for buffer systems.
Central heating is on if:
With 3-way-valve: burner on, 3-way-valve off, flow pump on,
With wwCharge pump: burner on, flow pump on. (wWCharging is possible parallel to central heating, depends on ww priority).
I can check the next days when it's colder outside and wW isn't heated by boiler not solar.
In my case the flow pump is connected to the mixer, i have to check what the UBAMonitor reports for the pump,

glitter-ball, 2020-09-24:
This sounds like a path I've trodden recently. #485 has some of the changes I needed to make to get everything back in HA.

At some point, I need to try the HA MQTT option - I'm still using the "Single" format.

glitter-ball, 2020-09-24:
Hmm... just looked at mine and realised that my HA doesn't really report this right.
[image]
As Michael says, it needs some simple logic - which could go in HA or EMS-ESP?

In my case (stored hot water cylinder, not instantaneous combi boiler), the boiler's actually producing hot water if 3-way valve=HW and Gas=On. The thermostat has set times for hot water (0645-0830 and 1800-2230 in my case) but these don't get picked up - Hot Water activated shows as always on. I should probably also look at why the Heating bar is blank and why the wWcomfort always says "hot".

cyniu88, 2020-09-24:
unfortunately, in my case, after upgrading to version 2.1.0b0, I never received the MQTT theme "Heating_active". The release works fine

proddy, 2020-09-25:
@cyniu88 do you see heating active in the Web or the Console?

MichaelDvP, 2020-09-25:
@cyniu88 the last release always publishes `heating_active` but the value was wrong for ww-buffer-systems. Now it's only published if there is a valid value.

@proddy maybe it's quite simple, i see in the ems-handbook:
```
Service codes BC:
–H 200 Der Kessel befindet sich im Heizbetrieb.             => heating_active
=H 201 Der Kessel befindet sich im Warmwasserbetrieb.       => ww active
```
I'll log this. The first char is a graphics char, not minus/equal, i don't know which charset.

proddy, 2020-09-25:
@MichaelDvP good spot! I'll trace it too and see what happens when water & heating is on & off.

glitter-ball, 2020-09-25:
I’ve decoded ServiceCodeNumber into a Status text and it does correctly report activity in my system:
[image]

Brown is CH active (burner on) and cyan is HW active. Was hoping to provide more detail from the Logbook but my derived status code isn’t available - maybe logbook can only deal with booleans?

lsw271, 2020-09-25:
I think i've got the same issue. 
`Boiler: Junkers GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)
`
mine is Junkers Cerapur 
`Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.04)
  Clock: 21:25:08 25/09/2020
` and CW400
My tapwater_active doesnt change, always changed heating_active even it was heating hot water

proddy, 2020-09-26:
@lsw271 could you do a quick test. Use the EMS-ESP web UI, click on the boiler and see if you see the first parameter which is "Service Code". Note down the value. Then run the hot water for a few seconds and see if the Service Code values change. With mine idle is 0A (305) and =H (201) when hot water tap is running.

FireWizard52, 2020-09-26:
Hi,
@proddy,

As I reported earlier #354, I did the test as well.
Currently I'm running, EMS-ESP software version 2.0.1.
Boiler type: Boiler: BK13/BK15/Smartline/GB1x2

Quiescent situation:

Service Code | 0H (203)
Hot tap water | off
Central heating | off

Hot water request:

Service Code | =H (201)
Hot tap water | off
Central heating | active

I have not been able to test with the Central heating on
However in old log files I do find a sub code 200, so I assume that it indicates

MichaelDvP, 2020-09-27:
I've made some tests for my system.  The ww-system is buffer and charge pump, no ww-priority. The central heating is with mixer and pump in the mixing unit, no pump at the boiler. Boiler is unmodulated an cycles often at moderate outdoor temperatures.
The service-Code-No. is in normal (non error) state always zero, i also can not display it on the MC10.

- central heating: `-H(0)` when burner is started (before short time `0C(0)` and `0L(0)`, after burner stop `0Y(0)`.
- DHW: if heating is completly off (summermode), than i get `=H(0)` when burner is on. The charge pump switch to on later when boilertemp reaches the actual buffer-temp, and the chargepump stays on 5 min after burner stopped. The code changes to `0Y(0)` with burner stop and does not change when pump stops.
- DHW: if heating flow (mixer) is on and DHW charging starts the code first letter is three dashes, char code 0xF0 and then `H(0)`, goes to `0Y(0)` after burner stop.

Working for me is:
```
heating_active_   = (serviceCodeChar[0] == '-' || serviceCodeChar[0] == 0xF0) && serviceCodeChar[1] == 'H';
tap_water_active_ = (serviceCodeChar[0] == '=' || serviceCodeChar[0] == 0xF0) && serviceCodeChar[1] == 'H';
```
checking the service code number would not work.
But for tap_water_active it's a question of definition. It's also possible to define it as on when charge pump is running. 
Also heating_active can be defined as flow pump running. I'm not sure what benefit i get from these publishes.

proddy, 2020-09-27:
That logic works for me too. We just need a few more tests on Junkers/W-Bosch and then we can make the switch for everyone.

The heating active was to simulate the flame icon that shows up on the Thermostat when the heating is on. I use the 'tap water active' to detect if the shower is running. Ironically shower detection and recording the duration of the shower was how EMS-ESP was born, back in August 2017. We've come a long way since :-)

glitter-ball, 2020-09-27:
Maybe it’s better to publish raw data points and then provide sample logic for HA to combine them into something meaningful?

Or (tongue slightly in cheek) you could build a configurable logic sub-system for EMS-ESP where you can combine existing points with Boolean operators to generate new ones and publish them. That would be a massive chunk of new code, though, unless there are cunning libraries out there already.

Nunak, 2020-09-27:
Where I could download this version? I would like to test it on my junkers cerapur modul and CW400.

MichaelDvP, 2020-09-27:
@glitter-ball The raw data are published already in `boiler_data`, but, depending on publish intervall, not actual. The `heating_active` and `tap_water_active` are always published on change.  You can also set `boiler_data` in mqtt to publish on change, but this give a lot more traffic.

@Nunak You have to check the `serviceCode` in `boiler_data` in the actual master or dev version. The actual master shows heating/tapwater topics, but in some cases wrong, the dev shows these topics only in flow systems. We are trying to find a logic that works on all systems, but this is not implemented yet.

proddy, 2020-09-27:
@Nunak grab the dev build from https://github.com/proddy/EMS-ESP/tree/firmware/firmware, load up the web and record the service code changes when the heating is on/off and hot tap water is on/off. That would really help us, thanks

FireWizard52, 2020-09-27:
Hi,
@proddy 

I have upgraded to the latest beta, version 2.1.0b0 and did some testing.
As reported before the quiescent situation (no tapwater, no heating) reported 0H(203) and the tapwater active reported =H(201)
Heating active reports -H(200)
Both tap water active and heating active report on topic ems-esp/heating_active = 1.
The topic ems-esp/tapwater_active = 0 and does not change,

This is what I'm able to contribute.

Regards

proddy, 2020-09-27:
@FireWizard52 excellent, thanks for testing. I think we're close.

Nunak, 2020-09-27:
> @Nunak grab the dev build from https://github.com/proddy/EMS-ESP/tree/firmware/firmware, load up the web and record the service code changes when the heating is on/off and hot tap water is on/off. That would really help us, thanks

So I did upgrade and looks like heating is working - see In mqtt and home assistant "active"

Tap water is still not working, watch on when watter was running and stoped:

000+00:13:22.603 N 408: [emsesp] Thermostat(0x10) -> Boiler(0x08), (0x35), data: 11 01
000+00:13:22.657 N 409: [emsesp] Thermostat(0x10) -> Boiler(0x08), (0x35), data: 37 (offset 3)
000+00:13:25.543 N 410: [emsesp] Boiler(0x08) -> (0x13), (0x05), data: 00 (offset 34)
000+00:13:25.902 N 411: [emsesp] Boiler(0x08) -> (0x00), (0x07), data: 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:13:26.132 N 412: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 19 01 25 59 00 01 23 00 C0 80 00 80 00 01 26 FF FF FF 00 00 00 00 00 00 00
000+00:13:26.346 N 413: [emsesp] Boiler(0x08) -> (0x00), (0x34), data: 32 01 96 02 1D 81 00 00 03 00 03 D6 AC 00 41 ED 00
000+00:13:26.571 N 414: [emsesp] Boiler(0x08) -> (0x00), (0x19), data: 00 84 80 00 80 00 FF FF 00 00 00 A9 DA 0B BE FF 00 00 00 07 E8 53 00 67 ED 80 00
000+00:13:35.557 N 415: [emsesp] Boiler(0x08) -> (0x00), (0x18), data: 19 01 25 59 00 01 23 00 C0 80 00 80 00 01 26 FF FF FF 00 00 00 00 00 00 00
000+00:13:35.779 N 416: [emsesp] Boiler(0x08) -> (0x13), (0x05), data: 00 (offset 34)
000+00:13:36.171 N 417: [emsesp] Boiler(0x08) -> (0x00), (0x34), data: 32 01 96 02 1C 81 00 00 03 00 03 D6 AC 00 41 ED 00
000+00:13:39.908 N 418: [emsesp] Thermostat(0x10) -> (0x00), (0x06), data: 14 09 14 1B 21 1D 06 01 10 FF 00
000+00:13:41.761 N 419: [emsesp] Thermostat(0x10) -> Boiler(0x08), (0x1A), data: 19 64 00
000+00:13:41.795 N 420: [emsesp] Thermostat(0x10) -> (0x00), (0x267), data: 00 00
000+00:13:42.034 N 421: [emsesp] Thermostat(0x10) -> (0x00), (0x2A5), data: 00 DA 21 2B 19 00 2B 28 00 93 03 03 01 00 93 03 2D 00 00 11 01 03 08 85 00
000+00:13:42.410 N 422: [emsesp] Thermostat(0x10) -> (0x00), (0x2A5), data: 08 04 FF E2 00 00 FF 64 37 00 3C 01 FF 01 02 (offset 25)
000+00:13:42.607 N 423: [emsesp] Thermostat(0x10) -> (0x00), (0x31D), data: 00 00 0A 07

proddy, 2020-09-27:
@Nunak could you try the test again (water on/off, heating on/off) and note down what the Service Codes are from the Web interface `http://ems-esp/ems-esp/devices`, like:

[image]

just to see if they match 100% with our thinking.

Nunak, 2020-09-27:
@proddy there is.
[image]

lsw271, 2020-09-27:
> 
> 
> @lsw271 could you do a quick test. Use the EMS-ESP web UI, click on the boiler and see if you see the first parameter which is "Service Code". Note down the value. Then run the hot water for a few seconds and see if the Service Code values change.
> 
> On my Nefit boiler it `-H (200)` when the central heating is on and `=H (201)` when warm water is on

I havent got service code in UI. I checked it from HA. 
200 = central heating is on
201 = warm water is on
203 = no central heating / no warm water

In the manual (junkers) i've got 17 service codes... I need to write some template sensor in HA to decode this ;)

glitter-ball, 2020-09-28:
@lsw271 see #281. @proddy helped me knock up some HA .yaml to decide ServiceCodeNumber which works well.

MichaelDvP, 2020-09-28:
@Nunak 
> Tap water is still not working, watch on when watter was running and stoped:

I can't see that the burner starts. 
Here is a warm water request from thermostat `Thermostat(0x10) -> Boiler(0x08), (0x35), data: 11 01`
with temperature 55°C `Thermostat(0x10) -> Boiler(0x08), (0x35), data: 37 (offset 3)`
but burner stays of and boiler is still in normal heating mode (01 at pos 5): `Boiler(0x08) -> (0x00), (0x18), data: 19 01 25 59 00 01 23 00 C0 80 00 80 00 01 26 FF FF FF 00 00 00 00 00 00 00`
and that's because the temperature of the buffer is 54.1°C, no need for heating (021D in pos3): `Boiler(0x08) -> (0x00), (0x34), data: 32 01 96 02 1D 81 00 00 03 00 03 D6 AC 00 41 ED 00` 

Can you do such a watch when burner starts and stops for heating and for warm water?

@proddy 
There is no statusCode in message 18, i think we need an alternate logic if statusCode is zero. Byte 5 in message 18 (see Norberts list byte9) is a candidate. I check this on my boiler.

proddy, 2020-09-28:
@MichaelDvP yes, alternative logic could be the old-school way of doing it. But I'm 100% sure there must be some status codes somewhere for HT3 as this is what their thermostats use. I worked on the Home Assistant MQTT Discovery yesterday so expect a big push coming in soon. Best not to add any PRs yet.

MichaelDvP, 2020-09-28:
I've tested the byte 5 and it's working good, no need to check the statusCode. 
```
        heating_active_   = ((boilerState_ & 0x09) == 0x09);
        tap_water_active_ = ((boilerState_ & 0x0A) == 0x0A);
```
and in `void Boiler::process_UBAMonitorFast(std::shared_ptr<const Telegram> telegram) {` add
```
    changed_ |= telegram->read_value(boilerState_, 5);
```

or condensed methode for check_active: 
```
void Boiler::check_active() {
    if ((boilerState_ & 0x09) != (last_boilerState & 0x09)) {
        Mqtt::publish(F("heating_active"), (boilerState_ & 0x09) == 0x09);
    }
    if ((boilerState_ & 0x0A) != (last_boilerState & 0x0A)) {
        Mqtt::publish(F("tapwater_active"), (boilerState_ & 0x0A) == 0x0A);
        EMSESP::tap_water_active((boilerState_ & 0x0A) == 0x0A);
    }
    last_boilerState = boilerState_;
}
```

proddy, 2020-09-28:
thanks, I'll include it my next push

FireWizard52, 2020-09-29:
@proddy 

Hi,

Yesterday evening I have installed this new beta as well, v2.1.0b1 and this morning I have been able to do some more testing.

Results:

Currently I'm running, EMS-ESP software version 2.1.0b1.
Boiler type: Boiler: BK13/BK15/Smartline/GB1x2

Quiescent situation:

Service Code | 0H (203)
Hot tap water | off
Central heating | off

Hot tapwater request:

Service Code | =H (201)
Hot tap water | off
Central heating | off

Heating request:

Service Code | -H (201)
Hot tap water | off
Central heating | off

Monitoring with MQTTExplorer indicate the same Services codes and sub codes.
However for the first time ever, I saw that the topic ems-esp/tapwater_actived showed a payload "on", when requested hot tap water. Similar ems-esp/heating_active shows also a payload "on", when heating was requested.

So this works.  Great!!

However, after to upgrade from 2.1.0b0 to 2.1.0b1 the WebUI does not update "Hot tap water" (It did not before either), but also "Central heating:  is not updated any more.

As a consequence the domoticz plug-in from BBQKees is broken as well and does not update any more.
However that plug-in need an update as many sensor/switches do not function as update to v2.0.0, end August/begin September

Regards

proddy, 2020-09-29:
cool, thanks @FireWizard52 for testing this out. Kudos also to @MichaelDvP for figuring out the correct way and programming it.

The "on"/"off" in `tapwater_actived` is the new change to handle the way booleans are rendered. Default is on/off but can be changed to true/false or 1/0 via the Settings page in the Web.

I'll look into why the web isn't updating the hot tap water and central heating.

As for Domoticz I'll create a new GitHub issue and get the migration work started.

MichaelDvP, 2020-09-29:
@FireWizard52 
>Heating request:
Service Code | -H (201)
Hot tap water | off
Central heating | off

is this a typo? central heating should be on. 

@proddy 
now it's setting only shown on mqtt, for web/telnet the variables have to be set. I see this later and edited my post.
`heating_active_ = ((boilerState_ & 0x09) == 0x09);` etc.
i can add this when adding the wwcircpump/wwcircmode commands.

FireWizard52, 2020-09-29:
@MichaelDvP

No, it is not a typo. Indeed, central heating should be on.
It was "on" in the MQTT published message, but not in the WebUI

proddy, 2020-09-29:
already fixed it, I'll do a push in a while. distracted with work stuff ;)

MichaelDvP, 2020-09-29:
Should i wait with a push or do a pr? 
I'd like to update some boiler values, thermostat RC300 summermode and analog enable setting.

proddy, 2020-09-29:
ok, done with my changes. It looks ok but I did rush it so watch out for errors.

MichaelDvP, 2020-09-29:
ok, i've added my changes too.

FireWizard52, 2020-09-29:
Updated from 2.1.0b1 to 2.1.0b2

System continued to work without any issues. Tap water active is still okay.
However the WebUI needs some extra attention, as this has been changed as well.

The two last lines only show (in my case) 50% and 1.5bar

[image]

This is probably caused, as the Service code and the Service code number are on 2 lines now and a line for Warm Water current temperature (extern) has been added.

The text for Pump modulation has disappeared and a text for Pressure has to be added.

This is what I have noted so far.

proddy, 2020-09-29:
@FireWizard52 thanks again for testing. I think I know the problem (Michael probably does too). Not enough memory reserved for the json buffer. We'll fix it.

Nunak, 2020-09-30:
Updated also from 2.1.0b1 to 2.1.0b2 and looks that I am able to see tap water and also heating. Thanks!

[image]

proddy, 2020-09-30:
Just did some testing on Win with chrome, edge, firefox upgrading 2.0.1 to 2.1.0 via the Web and it works. 

On OSX the select file & drag-drop is not working because I think OSX's security is blocking it. Need to look into why.

closed 2020-09-30 by proddy