emsesp/EMS-ESP #516: Building env

Sonusss, 2020-09-24:
Hi,

I'm trying to install my building environment.
Have followed your instructions regarding nodeJS and Platformio but when I "pio run" I get errors about missing LittleFS.h

I googled found this: https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS/src
copied both .h and .cpp files to lib.

Now there is another missing #include which I can't figure: <flash_hal.h>

I guess you should have default libraries not included in you project.

Can you please help ?

proddy, 2020-09-24:
You must be doing something wrong installing pio. Did you follow the guidelines on the platformio site? You shouldn't need to copy files.

The automated build process (travis CI and github actions) follow the same steps from a fresh O/S install so I know it works. It's just 3 lines really.

If you can describe the steps you took with screenshots I can help you. Windows or Linux?

Sonusss, 2020-09-24:
I'm on Linux (Ubuntu 18.04 LTS)
I installed Plaformio from this command : python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"

and installed NodeJS as decompress de tarball to /opt folder

Sonusss, 2020-09-25:
I could go a little further...
However now I'm stuck...
lib/framework/ESP8266React.h:33:21: fatal error: WWWData.h: No such file or directory

which comes from:
#ifdef PROGMEM_WWW
#include <WWWData.h>
#endif

proddy, 2020-09-25:
there's a script in the platformio.ini file that runs npm to build that WWWData.h file. Did you uncomment it by chance?

Sonusss, 2020-09-25:
nodejs was outdated....

closed 2020-09-25 by Sonusss